gnuradio/pybombs #448: uhd/Gnuradio installation via pybombs fails (same as #428)

shawnerz, 2017-07-16:
Hello.
Perhaps this is a problem for Ettus. However, I am trying to install GnuRadio from pybombs. When it tries to install uhd, I get the following error:
`[ 60%] Built target uhd
[ 60%] Built target getopt
[ 60%] Linking C executable tx_samples_c
../lib/libuhd.so.3.11: undefined reference to `boost::match_results<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<boost::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > >::maybe_assign(boost::match_results<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<boost::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > > const&)'
../lib/libuhd.so.3.11: undefined reference to boost::re_detail::perl_matcher<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<boost::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::construct_init(boost::basic_regex<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > > const&, boost::regex_constants::_match_flags)'
../lib/libuhd.so.3.11: undefined reference to `boost::filesystem::path_traits::dispatch(boost::filesystem::directory_entry const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)'
../lib/libuhd.so.3.11: undefined reference to `boost::archive::text_iarchive_impl<boost::archive::text_iarchive>::load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)'
../lib/libuhd.so.3.11: undefined reference to `boost::archive::text_oarchive_impl<boost::archive::text_oarchive>::save(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
examples/CMakeFiles/tx_samples_c.dir/build.make:108: recipe for target 'examples/tx_samples_c' failed
make[2]: *** [examples/tx_samples_c] Error 1
CMakeFiles/Makefile2:417: recipe for target 'examples/CMakeFiles/tx_samples_c.dir/all' failed
make[1]: *** [examples/CMakeFiles/tx_samples_c.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2`

I've reinstalled boost. I've tried using gcc-5, but I'm back at gcc-6 now.
Any ideas in to the failures would be greatly appreciated.
Thank you,
-Shawn

shawnerz, 2017-07-16:
Since I made that post, I've tried installing to UHD 3.8 and 3.9. I get the same failures during the make process.
I then tried removing UHD and Boost.
I then built Boost from source. Then tried both using the pybomb and building UHD from source. In all instances, I get to the same point: ~60% and a failure at tx_samples_c.c.
Any suggestions would be welcomed.
Thanks,
-Shawn

mbr0wn, 2017-09-15:
This is a Boost linker error, but unclear what's causing it. What's your distro?

machsFuel, 2017-09-18:
I found a kludge to get around this problem.  In the .pybombs/recipes/gr-recipes/gnuradio.lwr recipe I commented out the depends on uhd and changed -DENABLE_GR_UHD to =OFF.  This allowed me to build a working gnuradio install.

shawnerz, 2017-09-18:
I apologize for not updating my post. I wanted to reply to mbr0wn, but I got busy and I forgot.
To answer his question: I was using Ubuntu LTS 16.04. I do not know what version of the kernel Ubuntu is written around.
The problem I had was because of wrong versions of different packages. To solve my problem, I started with a fresh 16.04 install. Then run the GnuRadio pybomb and it was successful.

However, I had a separate, independent, project. That installation was failing because of uhd. I put that project off to the side and decided I will come back to it later when I had time.
Thank you to machsFuel for a work around. I will try it.
Thank you again!
-Shawn

mbr0wn, 2017-09-26:
If you're on Ubuntu 16.04, this is most certainly a problem caused by having multiple versions of UHD installed.

febs, 2017-10-05:
Shouldn't pybombs work around this multimple existance of the same tools by isolating in the respective prefixes all the necessary tools?

Thanks

febs, 2017-10-05:
BTW I worked around the issue by manually specifying at system level that the default gcc and g++ compilers should be version 5 rather than 6.

 $ sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-5 60
 $ sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-5 60•
 $ sudo update-alternatives --config gcc
 $ sudo update-alternatives --config g++

machsFuel, 2017-10-05:
Oh - I will have to give this a try thanks Fabio!

febs, 2017-10-09:
@mbr0wn  wouldn't it make sense then to also isolate the compilers and *everything* related? That would really allow to have an independent, **consistent** builiding environment - and thus the very same builds for everybody. Also useful for debugging...

Just wondering.

mbr0wn, 2017-11-02:
Well, stuff in `/usr/` is meant not to be ignored. For full isolation, Docker containers are usually the weapon of choice. By design, if you do nothing else, PyBOMBS will inherit your ld.config.cache, and env vars (such as `LD_LIBRARY_PATH`. It assumes that if something is in `/usr/local`, it can be used. If we wanted to ignore the system ld config cache, and the env vars, we'd be moving into non-standard behaviour (and as noted before, that's when you want a container, or simply something like a chroot).

mbr0wn, 2017-11-02:
This issue seems like a mix of unclear problems -- I'll close it.

closed 2017-11-02 by mbr0wn