gnuradio/pybombs #449: Installation of libosmo-dsp fails on centos-7 during PKG_CHECK_MODULES(FFTW3F, fftw3f >= 3.2.0)'

hjmitsd, 2017-07-24:
Hi
I have successfully installed gnuradio-default using pybombs
As a next step I want to install gr-osmosdr
This fails at the installation of libosmo-dsp
The error could be triggerd using
pybombs install  libosmo-dsp

checking for doxygen... /usr/bin/doxygen
./configure: line 11567: syntax error near unexpected token `FFTW3F,'
./configure: line 11567: `PKG_CHECK_MODULES(FFTW3F, fftw3f >= 3.2.0)'
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package libosmo-dsp:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package libosmo-dsp. Aborting.

The following fftw packages are installed

[root@chianti ~]# rpm -qa | grep fftw
fftw-libs-double-3.3.3-8.el7.i686
fftw-doc-3.3.3-8.el7.noarch
fftw-libs-long-3.3.3-8.el7.x86_64
fftw-devel-3.3.3-8.el7.i686
fftw-libs-3.3.3-8.el7.x86_64
fftw-3.3.3-8.el7.x86_64
fftw-devel-3.3.3-8.el7.x86_64
fftw-libs-double-3.3.3-8.el7.x86_64
fftw-3.3.3-8.el7.i686
fftw-libs-long-3.3.3-8.el7.i686
fftw-static-3.3.3-8.el7.i686
fftw-libs-single-3.3.3-8.el7.x86_64
fftw-libs-single-3.3.3-8.el7.i686
fftw-libs-3.3.3-8.el7.i686
fftw-static-3.3.3-8.el7.x86_64

Any idea how to proceed
regards

Hansjörg

mbr0wn, 2017-09-15:
I think this is an autotools version error. What distro are you running, and which version of autotools do you have?

mbr0wn, 2017-09-15:
Looks like CentOS 7 comes with automake 1.13. I can repro this in a Docker container.