gnuradio/pybombs #454: pybombs failed to install boost

cybercrypt13, 2017-09-02:
I've been searching all night for a solution to this but not sure exactly what my problem is.  When I run:
pybombs install gnuradio 
I get the following result.  I'm running SUSE Tumbleweed Latest version and all packages seem to be installed.  Can you tell me where I might look to discover the problem.

PyBOMBS - INFO - PyBOMBS Version 2.3.1
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:

PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: boost
PyBOMBS.Packager.source - WARNING - Build dir already exists: /home/username/source/sdr/src/boost/build
Installing:  (100%) [============================================================================================================================================================================================]
PyBOMBS.Packager.source - ERROR - Problem occurred while building package boost:
Installation failed
PyBOMBS.install_manager - ERROR - Error installing package boost. Aborting.

Thanks,

cybercrypt13, 2017-09-02:
When running with -v option I get one line that reads: 
PyBOMBS.Requirer - DEBUG - Requirements met.
PyBOMBS.Packager.source - DEBUG - Using build directory: /home/ghancock/source/sdr/src/boost/build
PyBOMBS.Packager.source - WARNING - Build dir already exists: /home/ghancock/source/sdr/src/boost/build
PyBOMBS.Packager.source - DEBUG - Package boost is already configured.
PyBOMBS.Packager.source - DEBUG - Package boost is already built.
PyBOMBS.Packager.source - DEBUG - Installing package boost
PyBOMBS.Packager.source - DEBUG - In cwd - /home/ghancock/source/sdr/src/boost/build
PyBOMBS._process_thread() - DEBUG - Executing command `$ cd .. && ./bjam --threading=multi --layout=tagged link=shared -j4 install --build-dir=build'
Installing:  (100%) [============================================================================================================================================================================================]
PyBOMBS.monitor_process() - DEBUG - Thread signaled termination or returned
PyBOMBS.monitor_process() - DEBUG - Return value: 1
PyBOMBS.Packager.source - ERROR - Problem occurred while building package boost:
Installation failed
PyBOMBS.install_manager - ERROR - Error installing package boost. Aborting.

mbr0wn, 2017-09-06:
Can you do -vv please?

mbr0wn, 2017-09-14:
Also what's  your distro?

mbr0wn, 2017-09-14:
It looks like boost build is failing, but it should rebuild with higher verbosity to show what's going on.

cybercrypt13, 2017-09-14:
I will do that shortly

Thanks

mbr0wn, 2017-09-14:
OK, found part of the problem. It seems like Suse bakes the version number into package names. :confounded: 

Not sure how to solve that.

The other problem is, why didn't it build from source...

mbr0wn, 2017-09-14:
Here's the problem: The package name for Boost devel packages in SUSE contains the version, so it changes all the time. The zypper plugin needs a way to handle this without having to update the Boost (and other recipes all the time).

mbr0wn, 2017-09-14:
As for the source build: A newer Boost works fine, and Tumbleweed has gcc 7 so I guess that's just a mismatch. Also, the Boost recipe does all the heavy lifting during the 'install' command, so that'll hide errors.

cybercrypt13, 2017-09-14:
This may be a stupid question but would it matter if we used yast or is
zipper only way to deal with it?   I am new to SUSE so sorry if that was a
dumb thing to ask

Glenn

mbr0wn, 2017-09-15:
I updated Boost version on gr-recipes, and now at least it compiles.

mbr0wn, 2017-10-07:
@cybercrypt13 Not a dumb question at all, the answer is simply that someone implemented zypper, and no one implemented yast. However, I don't think the package names change with the tool.

I'm closing this for now -- I think it'll just be easier to keep updating the recipes.

closed 2017-10-07 by mbr0wn