gnuradio/pybombs #455: PyBOMBS Quick Start Fails

TehWan, 2017-09-04:
Following step 3 of the Quick Start steps produces errors:
> $ pybombs recipes add-defaults
> PyBOMBS - INFO - PyBOMBS Version 2.3.1
> usage: pybombs recipes [-h] {list-repos,add,list,update,remove} ...
> pybombs recipes: error: argument sub_command: invalid choice: 'add-defaults' (choose from 'list-repos', 'add', 'list', 'update', 'remove')

Step 4 also fails:
>  $ pybombs prefix init ~/prefix -a myprefix -R gnuradio-default
> PyBOMBS - INFO - PyBOMBS Version 2.3.1
> PyBOMBS.prefix - ERROR - Could not find recipe for `gnuradio-default'

TehWan, 2017-09-04:
PyBOMBS was installed on a Raspberry Pi Zero W using `sudo pip install PyBOMBS`

mbr0wn, 2017-09-06:
We already have an issue on this, the readme shows the latest code, but pip install does not install the very latest bleeding edge code.

closed 2017-09-06 by mbr0wn

nakic, 2017-09-13:
Any pointers to other people who stumbled over the same problem? How does someone who just heard of and installed pybombs get gnuradio installed on a centos machine? A sequence of shell commands would be much appreciated. ;-)