gnuradio/pybombs #456: quickstart does not currently work

777arc, 2017-09-07:
Currently if you install pybombs through pip, you get an older version and `pybombs recipes add-defaults` is not a supported command.  This might seem like a non-issue, but I think it's extremely important that gnuradio be easy to install (through pybombs), and that the pybombs quickstart works all the time.  Especially with GRCon coming up.

zhudsonSM, 2017-09-13:
Had this exact problem. Until it is solved follow quickstart from before #447 here: https://github.com/gnuradio/pybombs/commit/4004f8c0d0727e9c64d85cd2f88935301f634e60?short_path=04c6e90#diff-04c6e90faac2675aa89e2176d2eec7d8

777arc, 2017-09-13:
Yeah, but most new people aren't going to be reading this issue, they will just jump to the conclusion that pybombs sucks =(

mbr0wn, 2017-09-13:
OK I just tagged a 2.3.2 release with that feature.

777arc, 2017-09-14:
Nice, quickstart works for me now

closed 2017-09-14 by 777arc