gnuradio/pybombs #457: pybombs recipes list, AttributeError

lryindra, 2017-09-07:
I'm seeing an AttributeError when executing pybombs recipes list command.  I saw this error on Ubuntu 17.04 32-bit and Ubuntu 16.04 64-bit.  The virtual environment is Python 3.5.  Are virtual environments the cause of this error?

(.venv) lryindra@lryindra-VirtualBox:~/Desktop/radio$ pybombs recipes list
PyBOMBS - INFO - PyBOMBS Version 2.3.1
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
Loading package information.....Traceback (most recent call last):
  File "/home/lryindra/Desktop/radio/.venv/bin/pybombs", line 11, in <module>
    sys.exit(main())
  File "/home/lryindra/Desktop/radio/.venv/lib/python3.5/site-packages/pybombs/main.py", line 32, in main
    return dispatch() or 0
  File "/home/lryindra/Desktop/radio/.venv/lib/python3.5/site-packages/pybombs/commands/base.py", line 194, in dispatch
    return get_cmd_dict(cmd_list)[args.command](cmd=args.command, args=args).run()
  File "/home/lryindra/Desktop/radio/.venv/lib/python3.5/site-packages/pybombs/commands/base.py", line 120, in run
    return self.subcommands[self.args.sub_command]['run'](self)()
  File "/home/lryindra/Desktop/radio/.venv/lib/python3.5/site-packages/pybombs/commands/recipes.py", line 218, in run_list
    'installed_by': format_pkg_list(pkgmgr.installed(pkg, return_pkgr_name=True)),
  File "/home/lryindra/Desktop/radio/.venv/lib/python3.5/site-packages/pybombs/package_manager.py", line 174, in installed
    pkg_version = pkgr.installed(r)
AttributeError: 'NoneType' object has no attribute 'installed'

mbr0wn, 2017-09-13:
Hm, can't repro. Can you give me a path to this error? I tried installing PyBOMBS into a clean Docker; didn't cause this error.

mbr0wn, 2017-09-14:
OK: Problem is, when there's no prefix, we don't instantiate a source package manager, but we need it to learn details about OOTs.

mbr0wn, 2017-09-14:
I think cb0f5b91fcc3b97b9f6c3fa6fd3be63901612482 should fix this.

lryindra, 2017-09-14:
Thank you.

I think cb0f5b9<https://github.com/gnuradio/pybombs/commit/cb0f5b91fcc3b97b9f6c3fa6fd3be63901612482> should fix this.

jep, 2017-09-14:
I just installed PyBOMBS version 2.3.2 on a fresh install of Linux Mint 18.2 Cinnamon 64-bit. I get the same error when running `pybombs recipes list`

`Traceback (most recent call last):
  File "/usr/local/bin/pybombs", line 11, in <module>
    load_entry_point('PyBOMBS==2.3.2', 'console_scripts', 'pybombs')()
  File "/usr/local/lib/python2.7/dist-packages/pybombs/main.py", line 32, in main
    return dispatch() or 0
  File "/usr/local/lib/python2.7/dist-packages/pybombs/commands/base.py", line 194, in dispatch
    return get_cmd_dict(cmd_list)[args.command](cmd=args.command, args=args).run()
  File "/usr/local/lib/python2.7/dist-packages/pybombs/commands/base.py", line 120, in run
    return self.subcommands[self.args.sub_command]['run'](self)()
  File "/usr/local/lib/python2.7/dist-packages/pybombs/commands/recipes.py", line 238, in run_list
    'installed_by': format_pkg_list(pkgmgr.installed(pkg, return_pkgr_name=True)),
  File "/usr/local/lib/python2.7/dist-packages/pybombs/package_manager.py", line 174, in installed
    pkg_version = pkgr.installed(r)
AttributeError: 'NoneType' object has no attribute 'installed'`

is there a way to apply the new fix or is it not yet released?

mbr0wn, 2017-09-14:
@jep Check out the manual for how to do a pip install from git, and then you'll have the fix.

mbr0wn, 2017-09-14:
@lryindra @jep Please reopen if you don't think this is fixed on the latest git version.

closed 2017-09-14 by mbr0wn