gnuradio/pybombs #460: Problem occurred while building package Gnuradio: unable to fetch recipe gnuradio

estroy, 2017-09-17:
I am trying to install Gnuradio along with gr-radar and gr-iio on a Ubuntu 17.04 VMWare session. It is a clean install. First I installed 17.04 and then I tried to install Gnuradio. It ran for a long time but then I got the message in the subject line, followed by Error installing package gnuradio. Aborting.
The command that I used was:
pybombs prefix init ~/prefix -a myprefix -R gnuradio-default

What am I doing wrong, or how do I get this all to install correctly?

mbr0wn, 2017-09-26:
We'd need to see the logs. Typically, this means the git clone failed.

argilo, 2017-10-23:
I ran into this problem too (when going through the Quickstart instructions, running on Ubuntu 17.10). It looked like the problem was with git-cache preventing the volk submodule from being fetched. Removing the `git-cache` line from `.pybombs/config.yml` solved the problem for me.

argilo, 2017-10-23:
When I ran with the `-v` option, output was the following:
```
PyBOMBS.install_manager - INFO - Installing package: gnuradio
PyBOMBS.PackageManager - DEBUG - install(gnuradio, install_type=source, static=False)
PyBOMBS.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBOMBS.PackageManager - DEBUG - Checking if package gnuradio is installed...
PyBOMBS.PackageManager - DEBUG - Package gnuradio is not installed.
PyBOMBS.PackageManager - DEBUG - Using packager source
PyBOMBS.Packager.source - DEBUG - State on package gnuradio is 0
PyBOMBS.Fetcher - DEBUG - Fetching source for recipe: gnuradio
PyBOMBS.Fetcher - DEBUG - Switching to src directory: /home/argilo/prefix/src
PyBOMBS.Fetcher - DEBUG - Getting fetcher for https://github.com/gnuradio/gnuradio.git, type git
PyBOMBS.Fetcher - DEBUG - Using fetcher <pybombs.fetchers.git.Git object at 0x7f66093b2d90>
PyBOMBS.Fetcher.git - DEBUG - Requiring packages on host system: ['git']
PyBOMBS.Fetcher.git - DEBUG - Requirements met.
PyBOMBS.Fetcher.git - DEBUG - Using url - https://github.com/gnuradio/gnuradio.git
PyBOMBS.Fetcher.git - DEBUG - Adding remote into git ref
PyBOMBS.GitCacheManager - DEBUG - Adding remote: gnuradio -> https://github.com/gnuradio/gnuradio.git
remote: Counting objects: 130555, done.
remote: Compressing objects: 100% (74/74), done.
remote: Total 130555 (delta 3), reused 74 (delta 3), pack-reused 130478
Receiving objects: 100% (130555/130555), 46.67 MiB | 11.47 MiB/s, done.
Resolving deltas: 100% (100351/100351), done.
From https://github.com/gnuradio/gnuradio
 * [new branch]          android    -> gnuradio/android
 * [new branch]          maint      -> gnuradio/maint
 * [new branch]          master     -> gnuradio/master
 * [new branch]          next       -> gnuradio/next
 * [new branch]          python3    -> gnuradio/python3
 * [new tag]             3.3git      -> 3.3git
 * [new tag]             3.5.1git    -> 3.5.1git
 * [new tag]             3.5.2git    -> 3.5.2git
 * [new tag]             3.5git      -> 3.5git
 * [new tag]             3.6.0git    -> 3.6.0git
 * [new tag]             3.6.1git    -> 3.6.1git
 * [new tag]             3.6git      -> 3.6git
 * [new tag]             3.7.0git    -> 3.7.0git
 * [new tag]             3.7.10git   -> 3.7.10git
 * [new tag]             3.7.12git   -> 3.7.12git
 * [new tag]             3.7.2git    -> 3.7.2git
 * [new tag]             3.7.3git    -> 3.7.3git
 * [new tag]             3.7.6git    -> 3.7.6git
 * [new tag]             3.7.7git    -> 3.7.7git
 * [new tag]             3.7.8git    -> 3.7.8git
 * [new tag]             3.7git      -> 3.7git
 * [new tag]             maint-3.3.0 -> maint-3.3.0
 * [new tag]             maint-3.4.0 -> maint-3.4.0
 * [new tag]             maint-3.4.1 -> maint-3.4.1
 * [new tag]             maint-3.4.2 -> maint-3.4.2
 * [new tag]             maint-3.5.1 -> maint-3.5.1
 * [new tag]             maint-3.5.2 -> maint-3.5.2
 * [new tag]             maint-3.5.3 -> maint-3.5.3
 * [new tag]             v3.3.0      -> v3.3.0
 * [new tag]             v3.3.0-rc0  -> v3.3.0-rc0
 * [new tag]             v3.3.0-rc1  -> v3.3.0-rc1
 * [new tag]             v3.3.0-rc2  -> v3.3.0-rc2
 * [new tag]             v3.3.0-rc3  -> v3.3.0-rc3
 * [new tag]             v3.3.1      -> v3.3.1
 * [new tag]             v3.3.1git   -> v3.3.1git
 * [new tag]             v3.4.0      -> v3.4.0
 * [new tag]             v3.4.0git   -> v3.4.0git
 * [new tag]             v3.4.1      -> v3.4.1
 * [new tag]             v3.4.1git   -> v3.4.1git
 * [new tag]             v3.4.2      -> v3.4.2
 * [new tag]             v3.4.2git   -> v3.4.2git
 * [new tag]             v3.4git     -> v3.4git
 * [new tag]             v3.5.0      -> v3.5.0
 * [new tag]             v3.5.0git   -> v3.5.0git
 * [new tag]             v3.5.0rc0   -> v3.5.0rc0
 * [new tag]             v3.5.1      -> v3.5.1
[... 30 lines cut ...]
 * [new tag]             v3.7.6      -> v3.7.6
 * [new tag]             v3.7.6.1    -> v3.7.6.1
 * [new tag]             v3.7.6.2    -> v3.7.6.2
 * [new tag]             v3.7.7      -> v3.7.7
 * [new tag]             v3.7.7.1    -> v3.7.7.1
 * [new tag]             v3.7.7.2    -> v3.7.7.2
 * [new tag]             v3.7.8      -> v3.7.8
 * [new tag]             v3.7.8.1    -> v3.7.8.1
 * [new tag]             v3.7.8.2    -> v3.7.8.2
 * [new tag]             v3.7.8rc1   -> v3.7.8rc1
 * [new tag]             v3.7.9      -> v3.7.9
 * [new tag]             v3.7.9.1    -> v3.7.9.1
 * [new tag]             v3.7.9.2    -> v3.7.9.2
 * [new tag]             v3.7.9.3    -> v3.7.9.3
 * [new tag]             v3.7.9rc1   -> v3.7.9rc1
PyBOMBS._process_thread() - DEBUG - Executing command `git clone https://github.com/gnuradio/gnuradio.git gnuradio --recursive --reference /home/argilo/.pybombs/gitcache -b master'
Cloning:     (100%) [=============================================================================]
PyBOMBS.monitor_process() - DEBUG - Thread signaled termination or returned
PyBOMBS.monitor_process() - DEBUG - Return value: 1
PyBOMBS.Fetcher - DEBUG - That didn't work.
PyBOMBS.Fetcher - DEBUG - Process returned value: 1
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Unable to fetch recipe gnuradio
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.
```

argilo, 2017-10-23:
When I run the `git clone` command on its own, I get:
```
$ git clone https://github.com/gnuradio/gnuradio.git gnuradio --recursive --reference /home/argilo/.pybombs/gitcache -b master
Cloning into 'gnuradio'...
Submodule 'volk' (https://github.com/gnuradio/volk.git) registered for path 'volk'
fatal: submodule 'volk' cannot add alternate: path '/home/argilo/.pybombs/gitcache/modules/volk/' does not exist
Failed to clone 'volk'. Retry scheduled
fatal: submodule 'volk' cannot add alternate: path '/home/argilo/.pybombs/gitcache/modules/volk/' does not exist
Failed to clone 'volk' a second time, aborting
```

argilo, 2017-10-23:
My guess is that the problem occurs because git 2.11 started including submodules when the `--reference` option is used: https://hackernoon.com/whats-new-in-git-2-11-64860aea6c4f#a3a3

The `GitCacheManager` doesn't seem to fetch submodules.

mbr0wn, 2017-11-03:
@argilo Thanks, that was a great hint. It seems like there's not actually a good way to clone submodules into bare repositories. I don't have a solution to this problem yet.

mbr0wn, 2017-11-03:
Looks like `--reference-if-able` is the solution.

mbr0wn, 2017-11-03:
...which was also added in 2.11.

mbr0wn, 2017-11-03:
Adding a version check for git, and then conditionally use `--reference` or `--reference-if-able`.

mbr0wn, 2017-11-03:
5a340b450af1b29ffc4da381b276e96fb2f7b968 enables the use of `--reference-if-able` for newer gits.

closed 2017-11-03 by mbr0wn

piotrva, 2018-05-18:
In order to install GNURadio on Ubuntu 17.10 one have to use following commend to install PyBOMBS:
`[sudo] pip install [--upgrade] git+https://github.com/gnuradio/pybombs.git`

r4d10n, 2018-05-21:
In Ubuntu 18.04, as pointed by @velichkov - manually removing <prefix>/src/gnuradio and doing 
```
pybombs -v fetch gnuradio
```
fetches volk and then retrying the prefix pybombs init command builds gnuradio...