gnuradio/pybombs #462: pymbombs fail on CentOS 7

nakic, 2017-09-19:
Command:
`$ pybombs prefix init ~/prefix -a myprefix -R gnuradio-default`

Output:

[...]
/usr/bin/ld: skipping incompatible /usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../libtinfo.so when searching for -ltinfo
/usr/bin/ld: skipping incompatible /lib/libtinfo.so when searching for -ltinfo
/usr/bin/ld: skipping incompatible /usr/lib/libtinfo.so when searching for -ltinfo
/usr/bin/ld: cannot find -ltinfo
/usr/bin/ld: skipping incompatible /usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../libform.so when searching for -lform
/usr/bin/ld: skipping incompatible /lib/libform.so when searching for -lform
/usr/bin/ld: skipping incompatible /usr/lib/libform.so when searching for -lform
/usr/bin/ld: cannot find -lform
collect2: error: ld returned 1 exit status
make[2]: *** [examples/rx_ascii_art_dft] Error 1
make[1]: *** [examples/CMakeFiles/rx_ascii_art_dft.dir/all] Error 2
make: *** [all] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package uhd:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package uhd. Aborting.

Environment:

$ head -n 1 /etc/centos-release
==> /etc/centos-release <==
CentOS Linux release 7.3.1611 (Core) 

Installed pybombs using pip, with a few hickups caused by what looks similar to issue #455, but seems unrelated to me.

mbr0wn, 2017-09-26:
UHD not finding curses... shouldn't fail like this. Can you pastebin the full log?

mbr0wn, 2017-11-02:
Note: This shouldn't happen because if UHD doesn't find curses, it would not actually try and link against it.

mbr0wn, 2017-11-02:
I'm flagging this unclear until we know this is actually a PyBOMBS bug.