gnuradio/pybombs #463: PyBOMBS fails to install on Fedora 26

cafeclimber, 2017-09-22:
PyBOMBS was install through pip.

After running auto-config, running the following:

    sudo pybombs prefix init /usr/local -a myprefix -R gnuradio-default

produces the following after running through a lot of package installation:
```
PyBOMBS.prefix - INFO - Installing default packages for prefix...
PyBOMBS.prefix - INFO - 
  - gnuradio
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Traceback (most recent call last):
  File "/bin/pybombs", line 11, in <module>
    load_entry_point('PyBOMBS==2.3.3a0', 'console_scripts', 'pybombs')()
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/main.py", line 32, in main
    return dispatch() or 0
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/commands/base.py", line 194, in dispatch
    return get_cmd_dict(cmd_list)[args.command](cmd=args.command, args=args).run()
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/commands/base.py", line 120, in run
    return self.subcommands[self.args.sub_command]['run'](self)()
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/commands/prefix.py", line 183, in run_init
    self.args.virtualenv
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/commands/prefix.py", line 337, in _init_prefix
    return install_dependencies(prefix_recipe.depends)
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/commands/prefix.py", line 318, in install_dependencies
    print_tree=True,
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/install_manager.py", line 112, in install
    lambda pkg: _cached_check_if_pkg_goes_into_tree(pkg, _check_if_pkg_goes_into_tree)
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/dep_manager.py", line 57, in make_dep_tree
    dep_trees[pkg] = self.make_tree_recursive(pkg, filter_callback)
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/dep_manager.py", line 77, in make_tree_recursive
    deps_to_install = set([dep for dep in all_deps if filter_callback(dep)])
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/install_manager.py", line 112, in <lambda>
    lambda pkg: _cached_check_if_pkg_goes_into_tree(pkg, _check_if_pkg_goes_into_tree)
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/install_manager.py", line 97, in _cached_check_if_pkg_goes_into_tree
    ret_val = check_callback(pkg)
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/install_manager.py", line 63, in _check_if_pkg_goes_into_tree
    if not self.pm.exists(pkg):
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/package_manager.py", line 127, in exists
    if self.check_package_flag(name, 'forceinstalled'):
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/package_manager.py", line 88, in check_package_flag
    recipe.get_recipe(pkgname).category
AttributeError: 'Recipe' object has no attribute 'category'
```

mbr0wn, 2017-11-02:
Can you please post the entire log output? I have a hunch the recipe loading is failing further up.

mbr0wn, 2018-06-17:
Running 938df531983dd3e95adb858701b9f96f895718a7, the canonical test for Fedora passes:

    $ ./run-tests.sh --container=fedora26 --skip-pylint

Closing this. If you have some more insight, please reopen or open a new issue.

closed 2018-06-17 by mbr0wn