gnuradio/pybombs #465: PyBombs fails while building apache-thrift while installing gnuradio

satin404, 2017-09-30:
While installing gnuradio, Pybombs fails stating the requirement of libcrypto library. 

PyBOMBS - INFO - PyBOMBS Version 2.3.3a
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Install tree:
|
\- gnuradio
   |
   \- apache-thrift
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: apache-thrift
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
Configuring: (100%) [=================================================================================================================]
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
configure.ac:90: installing './config.guess'
configure.ac:90: installing './config.sub'
configure.ac:27: installing './install-sh'
configure.ac:27: installing './missing'
compiler/cpp/Makefile.am: installing './depcomp'
configure.ac: installing './ylwrap'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supp ...
.
.
.
checking for pthread_create in -lpthread... yes
checking for clock_gettime in -lrt... yes
checking for setsockopt in -lsocket... no
checking for BN_init in -lcrypto... no
configure: error: "Error: libcrypto required."
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package apache-thrift:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package apache-thrift. Aborting.

Please Help?

mbr0wn, 2017-10-02:
We need more details. Distro, selected version, ...

willanth, 2017-10-31:
I have the same issue.  Distro is current Debian 9.2.1 ("squeeze") and version of GNURadio is current release.  Running pybombs straight out of the instructions.

draeath, 2017-11-26:
I believe you need the `libssl-dev` package, it provides [this library](https://packages.debian.org/search?searchon=contents&keywords=libcrypto.so&mode=exactfilename&suite=stable&arch=any).

draeath, 2017-11-26:
This issue should be fixed [upstream](https://issues.apache.org/jira/browse/THRIFT-3207) - is pybombs requiring thrift older than 0.11.0?

draeath, 2017-12-05:
Yes, looks like we're pulling in an ancient version of thrift ([0.9.3](https://github.com/gnuradio/gr-recipes/blob/03b560f7741bcec5502642d76194bd2a11bdbe3d/apache-thrift.lwr#L40) - it would be wonderful if work can happen to update that.

rayninth, 2017-12-10:
I have similar issue, trying to compile gnuradio. can you help on how to resolve this.

SirCosty, 2018-01-14:
Yeah would be nice if some cane update the Thrift version from 0.9.3 
Think the last Thrift version is 0.10.0 so far.
The easiest solution for us so far they too busy to fix might be to make a branch and change the https://github.com/gnuradio/gr-recipes/blob/master/apache-thrift.lwr
Hope will have time to test it as soon as possible and bring feedback

chris007de, 2018-03-09:
Same here, when following the Quick Start manual from https://github.com/gnuradio/pybombs, compilation of Apache Thrift fails, either with libssl-dev and libssl1.0-dev (see comment by @draeath ).
Manually bumping the version in the Thrift recipe to 0.10.0 works for me, using the most recent version, 0.11.0, does not work due to a linker error (TSimpleServer not found) during gnuradio build.

We could easily create a pull request for bumping the Thrift version to 0.10.0, but are there any recipes (besides gnuradio) which could break by doing so?

My system: Debian 9.3, Linux version 4.9.0-6-amd64

Chiggins, 2018-05-22:
Just wanted to post an update that nearly two months after the last comment, I'm still running into this error. Ubuntu 18.04 using [this](http://fieldxp.com/Chapter-2-Installing-the-Software.pdf) install guide.

catalinalb, 2019-11-02:
> Just wanted to post an update that nearly two months after the last comment, I'm still running into this error. Ubuntu 18.04 using [this](http://fieldxp.com/Chapter-2-Installing-the-Software.pdf) install guide.

Hello Chiggins,
I did follow the install guide on ubuntu 18.04 and pybombs is installing gnuradio but the version is 3.8 and gr-iqbal is not compatible with 3.8. How can I trick it to install 3.7 under the gnuradio name. 
Thank you

ghostop14, 2019-11-03:
I have a 3.8 migrated version of gr-iqbal here.  https://github.com/ghostop14/gr-iqbal

The osmocom maintainers haven’t been in a hurry to upgrade their modules to 3.8 compatible OOTs.  You can edit the lwr in the pybombs recipe directory for gr-iqbal to point here then run as normal if you want to go through the full pybombs install till its merged or they upgrade it.