gnuradio/pybombs #467: Problem occurred while building package pyqt4

drom, 2017-10-04:
I have trouble building gnuradio with pybombs.
```sh
sudo pip install --upgrade git+https://github.com/gnuradio/pybombs.git
pybombs -vv recipes add-defaults
pybombs -vv prefix init ~/prefix -a myprefix -R gnuradio-default
```
I am getting this error:
```sh
...
PyBOMBS.ReqScanner - OBNOXIOUS - Done parsing requirements string `make && (gcc || clang) && (gcc-c++ || clang)`
PyBOMBS.Packager.zypper - OBNOXIOUS - Calling ev for recursive satisfier rule evaluation
PyBOMBS.Packager.zypper - DEBUG - Package make has version 4.2.1-2.1 in rpm
PyBOMBS.Packager.zypper - DEBUG - Package gcc has version 7-2.4 in rpm
PyBOMBS.Packager.zypper - DEBUG - Package gcc-c++ has version 7-2.4 in rpm
PyBOMBS.install_manager - OBNOXIOUS - Installed, but no update requested. Does not go into tree.
PyBOMBS.install_manager - DEBUG - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.Requirer - DEBUG - Requirements met.
PyBOMBS.Packager.source - DEBUG - Using build directory: /home/drom/prefix/src/pyqt4
PyBOMBS.Packager.source - DEBUG - Configuring recipe pyqt4
PyBOMBS.Packager.source - DEBUG - Using vars - CommentedMap([('config_opt', ''), ('builddir', '/home/drom/prefix/src/pyqt4')])
PyBOMBS.Packager.source - DEBUG - In cwd - /home/drom/prefix/src/pyqt4
PyBOMBS._process_thread() - DEBUG - Executing command `$ python configure.py --confirm-license -b /home/drom/prefix/bin -d /home/drom/prefix/lib/python2.7/site-packages/ -v /home/drom/prefix/share/sip/ --no-designer-plugin CXXFLAGS+=-Wno-narrowing'
Traceback (most recent call last):
  File "configure.py", line 37, in <module>
    import sipconfig
ImportError: No module named sipconfig
PyBOMBS.monitor_process() - DEBUG - Thread signaled termination or returned
PyBOMBS.monitor_process() - DEBUG - Return value: 1
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
PyBOMBS.Packager.source - DEBUG - Configuring recipe pyqt4
PyBOMBS.Packager.source - DEBUG - Using vars - CommentedMap([('config_opt', ''), ('builddir', '/home/drom/prefix/src/pyqt4')])
PyBOMBS.Packager.source - DEBUG - In cwd - /home/drom/prefix/src/pyqt4
PyBOMBS._process_thread() - DEBUG - Executing command `$ python configure.py --confirm-license -b /home/drom/prefix/bin -d /home/drom/prefix/lib/python2.7/site-packages/ -v /home/drom/prefix/share/sip/ --no-designer-plugin CXXFLAGS+=-Wno-narrowing'
Traceback (most recent call last):
  File "configure.py", line 37, in <module>
    import sipconfig
ImportError: No module named sipconfig
PyBOMBS.monitor_process() - DEBUG - Thread signaled termination or returned
PyBOMBS.monitor_process() - DEBUG - Return value: 1
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package pyqt4:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package pyqt4. Aborting.
```

drom, 2017-10-04:
if I install `python2-qt4` and `python3-qt4` at system level, I am getting different error:

```sh
make[1]: Entering directory '/home/drom/prefix/src/pyqt4/qpy'
make[2]: Entering directory '/home/drom/prefix/src/pyqt4/qpy/QtCore'
g++ -c -pipe -fno-strict-aliasing -O2 -fPIC -Wall -W -D_REENTRANT -DQT_WEBKIT -DQT_NO_DEBUG -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/default -I. -I/usr/include/QtCore -I/usr/include/QtGui -I/usr/include -I/usr/include/python2.7 -I../../QtCore -I. -I. -o qpycore_chimera.o qpycore_chimera.cpp
qpycore_chimera.cpp: In member function ‘void Chimera::set_flag()’:
qpycore_chimera.cpp:376:22: error: ‘pyqt4ClassTypeDef’ was not declared in this scope
         _is_flag = ((pyqt4ClassTypeDef *)_type)->qt4_flags & 0x01;
                      ^~~~~~~~~~~~~~~~~
qpycore_chimera.cpp:376:22: note: suggested alternative: ‘sipClassTypeDef’
         _is_flag = ((pyqt4ClassTypeDef *)_type)->qt4_flags & 0x01;
                      ^~~~~~~~~~~~~~~~~
                      sipClassTypeDef
qpycore_chimera.cpp:376:41: error: expected primary-expression before ‘)’ token
         _is_flag = ((pyqt4ClassTypeDef *)_type)->qt4_flags & 0x01;
                                         ^
qpycore_chimera.cpp:376:42: error: expected ‘)’ before ‘_type’
         _is_flag = ((pyqt4ClassTypeDef *)_type)->qt4_flags & 0x01;
                                          ^~~~~
make[2]: *** [Makefile:274: qpycore_chimera.o] Error 1
make[2]: Leaving directory '/home/drom/prefix/src/pyqt4/qpy/QtCore'
make[1]: *** [Makefile:3: all] Error 2
make[1]: Leaving directory '/home/drom/prefix/src/pyqt4/qpy'
make: *** [Makefile:3: all] Error 2
PyBOMBS.monitor_process() - DEBUG - Thread signaled termination or returned
PyBOMBS.monitor_process() - DEBUG - Return value: 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package pyqt4:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package pyqt4. Aborting.
```

mbr0wn, 2017-10-05:
Can you find out if pyqt4 has a binary package for Tumbleweed?

mbr0wn, 2017-10-05:
...lol you answered that question below.

mbr0wn, 2017-10-05:
Hm, when you hand-installed python*-qt4, it still tried to build pyqt4?

drom, 2017-10-05:
@mbr0wn if I have only `python*-qt4` packages I am getting this error:
```sh
...
yBOMBS.ReqScanner - OBNOXIOUS - Found package combiner &&
PyBOMBS.ReqScanner - OBNOXIOUS - Adding package with name gcc-c++
PyBOMBS.ReqScanner - OBNOXIOUS - Found package combiner ||
PyBOMBS.ReqScanner - OBNOXIOUS - Adding package with name clang
PyBOMBS.ReqScanner - OBNOXIOUS - End of requirements list
PyBOMBS.ReqScanner - OBNOXIOUS - Done parsing requirements string `make && (gcc || clang) && (gcc-c++ || clang)`
PyBOMBS.Packager.zypper - OBNOXIOUS - Calling ev for recursive satisfier rule evaluation
PyBOMBS.Packager.zypper - DEBUG - Package make has version 4.2.1-2.1 in rpm
PyBOMBS.Packager.zypper - DEBUG - Package gcc has version 7-2.4 in rpm
PyBOMBS.Packager.zypper - DEBUG - Package gcc-c++ has version 7-2.4 in rpm
PyBOMBS.install_manager - OBNOXIOUS - Installed, but no update requested. Does not go into tree.
PyBOMBS.install_manager - DEBUG - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.Requirer - DEBUG - Requirements met.
PyBOMBS.Packager.source - DEBUG - Using build directory: /home/drom/prefix/src/pyqt4
PyBOMBS.Inventory - DEBUG - Setting state to `none'
PyBOMBS.Inventory - DEBUG - Saving inventory to file /home/drom/prefix/.pybombs/inventory.yml...
PyBOMBS.Packager.source - ERROR - Problem occurred while building package pyqt4:
There should be a source dir in /home/drom/prefix/src/pyqt4, but there isn't.
PyBOMBS.install_manager - ERROR - Error installing package pyqt4. Aborting.
```
With `python*-qt4-devel` (v4.12.1-1.1) I am getting compile error above.