gnuradio/pybombs #468: Package has no install method: python

6vasia, 2017-10-04:
PyBOMBS.PackageManager - DEBUG - Checking if package python is installable...
PyBOMBS.Packager.portage - DEBUG - Package dev-lang/python has versions 3.4.5, 2.4.4 in portage
PyBOMBS.PackageManager - DEBUG - Package python is not installable.
PyBOMBS.install_manager - ERROR - Package has no install method: python
PyBOMBS - DEBUG - Unresolved install path.

using pybombs: d40b0b04

eix dev-lang/pyhon:
[I] dev-lang/python
     Available versions:  
     (2.7)  2.7.12 ~2.7.13
     (3.4)  3.4.5(3.4/3.4m) ~3.4.6(3.4/3.4m)
     (3.5)  ~3.5.3(3.5/3.5m)
     (3.6)  ~3.6.1-r1(3.6/3.6m)
Installed versions:  2.7.12(2.7) 3.4.5(3.4)

mbr0wn, 2017-10-06:
Distro?

6vasia, 2017-10-08:
Oops, sorry. Gentoo.

noc0lour, 2017-10-08:
@6vasia It looks like the check if python 2.7 is installed failed and python 2.7 then isn't listed as possible installation candidate.

PR #470 (2017-11-02): fix package version resolution in portage for multiple available pack…
This should fix #468 and might prevent further issues from occurring. Currently portage is only listing the "best" match for a specific package. Tweaking the match level parameter of `_xmatch` will list all visible (not masked) versions in portage. This might introduce an issue if multiple versions are available and not yet installed and pybombs would have to select the most appropriate version thus the current `bestmatch-visible` approach. 
I also added a "$" to the package regex otherwise we might match unwanted packages.
closed 2017-11-02 by mbr0wn