gnuradio/pybombs #469: pybombs fail to proceed during libosmocore compilation failure

mluis, 2017-10-07:
**Result:**

```
make[3]: Entering directory '/root/srcs/libosmocore/src'
  CC       conv_acc_sse.lo
In file included from conv_acc_sse.c:59:0:
./conv_acc_sse_impl.h:258:17: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'static'
 __always_inline static void _sse_metrics_k5_n2(const int16_t *val,
                 ^~~~~~
./conv_acc_sse_impl.h:300:17: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'static'
 __always_inline static void _sse_metrics_k5_n4(const int16_t *val,
                 ^~~~~~
./conv_acc_sse_impl.h:340:17: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'static'
 __always_inline static void _sse_metrics_k7_n2(const int16_t *val,
                 ^~~~~~
./conv_acc_sse_impl.h:415:17: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'static'
 __always_inline static void _sse_metrics_k7_n4(const int16_t *val,
                 ^~~~~~
conv_acc_sse.c: In function 'osmo_conv_sse_metrics_k5_n2':
conv_acc_sse.c:84:2: warning: implicit declaration of function '_sse_metrics_k5_n2' [-Wimplicit-function-declaration]
  _sse_metrics_k5_n2(_val, out, sums, paths, norm);
  ^~~~~~~~~~~~~~~~~~
conv_acc_sse.c: In function 'osmo_conv_sse_metrics_k5_n3':
conv_acc_sse.c:93:2: warning: implicit declaration of function '_sse_metrics_k5_n4' [-Wimplicit-function-declaration]
  _sse_metrics_k5_n4(_val, out, sums, paths, norm);
  ^~~~~~~~~~~~~~~~~~
conv_acc_sse.c: In function 'osmo_conv_sse_metrics_k7_n2':
conv_acc_sse.c:111:2: warning: implicit declaration of function '_sse_metrics_k7_n2' [-Wimplicit-function-declaration]
  _sse_metrics_k7_n2(_val, out, sums, paths, norm);
  ^~~~~~~~~~~~~~~~~~
conv_acc_sse.c: In function 'osmo_conv_sse_metrics_k7_n3':
conv_acc_sse.c:120:2: warning: implicit declaration of function '_sse_metrics_k7_n4' [-Wimplicit-function-declaration]
  _sse_metrics_k7_n4(_val, out, sums, paths, norm);
  ^~~~~~~~~~~~~~~~~~
make[3]: *** [Makefile:537: conv_acc_sse.lo] Error 1
make[3]: Leaving directory '/root/srcs/libosmocore/src'
make[2]: *** [Makefile:397: all] Error 2
make[2]: Leaving directory '/root/srcs/libosmocore/src'
make[1]: *** [Makefile:596: all-recursive] Error 1
make[1]: Leaving directory '/root/srcs/libosmocore'
make: *** [Makefile:402: all] Error 2
```

mbr0wn, 2017-11-02:
Is this a PyBOMBS or libosmocore issue? What system does this happen on?

mluis, 2017-11-04:
@mbr0wn pybombs has a fallback for retrying the build. My suggestion for pybombs is to try do it the commit before the one that failed proceeding like the example above.

mbr0wn, 2017-11-06:
The retry mechanism is to drop the number of parallel builds, and turn up verbosity so people can see what failed. Changing the commit under hood would mean changing things from under make's feet. Plus, it makes the assumption that you're using git. I'd rather people hand-roll back a commit if that's what they want to try.
I've been building libosmocore an various platforms successfully, so I'll close this.

closed 2017-11-06 by mbr0wn

mluis, 2017-11-07:
@mbr0wn would you mind trying to build it on an Alpine distro or Stage0 Linux ?

mbr0wn, 2017-11-07:
@mluis can you submit a pr that will instantiate an alpine docker container? Take a look at the other Dockerfiles under tests/ and add one for alpine. Then you can use rum_tests.sh to run the container. When that's merged I might be able to take a peek.

mluis, 2017-11-07:
Thanks. I'm currently with very limited time but it is of my interest do it ASAP.

mbr0wn, 2017-11-07:
That's good to hear. You're the only Alpine requester so far, so I'm really looking forward to your PR.

mbr0wn, 2017-11-07:
Luckily writing the Docker files is not a terribly time-consuming task.

mluis, 2017-11-07:
ahah! So imagine don't having time for that!