gnuradio/pybombs #472: PyBOMBS dependancy chain broke on Alpine Linux.

mluis, 2017-10-10:
Since PyBOMBS doesn't support some Linux Distros, it will grab from a repository and build binaries, although all Alpine Linux libraries are compiled with LibreSSL but pybombs's qt4.lwr is configured to use openssl. This breaks the entire purpose of PyBOMBS.

Suggestion goes for supporting Alpine Linux library checking.

noc0lour, 2017-10-10:
Qt4 is available for Alpine? I don't see apk in the list of available packagers for PyBOMBS. Adding apk to the list of supported packagers would be the way to go for Alpine IMHO.

noc0lour, 2017-10-10:
This should be closed in favour of #471. If someone adds apk for alpine it will get considerably easier to satisfy all dependencies from the alpine repo.

mluis, 2017-10-10:
The issue is the source code compilation flags of pybombs's qt4.lwr.
pybombs should work as expected, in this case by compiling from source, when there's no available configured packager. Since it is compiling with openssl but all Alpine libs are using LibreSSL, PyBOMBS fails.

noc0lour, 2017-10-11:
I'd say this not a PyBOMBS issue but rather something for (dead) QT4 upstream since it should detect the SSL lib with it's buildsystem if anything. If someone adds the alpine packager to resolve this with a binary dependency this issue will be gone. 
The qt4 config_opts are:
```
  config_opt: -opensource --confirm-license -L$prefix/lib -L$prefix/lib64
```
nothing here with OpenSSL vs. LibreSSL

closed 2017-10-15 by mluis