gnuradio/pybombs #473: pybombs failing - unable to find directory

justennis, 2017-10-12:
I am getting an error that I can't get past for installing gnu radio via pybombs.  I'm new to pybombs but would love to get gnu radio up and running with my new Pluto SDR.

any help would be greatly appreciated.

Here is what I see on the screen:

username@ubuntu:~$ pybombs recipes add gr-recipes https://github.com/gnuradio/gr-recipes.git
PyBOMBS - INFO - PyBOMBS Version 2.3.2
error: Unknown subcommand: get-url
usage: git remote [-v | --verbose]
   or: git remote add [-t <branch>] [-m <master>] [-f] [--tags|--no-tags] [--mirror=<fetch|push>] <name> <url>
   or: git remote rename <old> <new>
   or: git remote remove <name>
   or: git remote set-head <name> (-a | --auto | -d | --delete |<branch>)
   or: git remote [-v | --verbose] show [-n] <name>
   or: git remote prune [-n | --dry-run] <name>
   or: git remote [-v | --verbose] update [-p | --prune] [(<group> | <remote>)...]
   or: git remote set-branches [--add] <name> <branch>...
   or: git remote set-url [--push] <name> <newurl> [<oldurl>]
   or: git remote set-url --add <name> <newurl>
   or: git remote set-url --delete <name> <url>

    -v, --verbose         be verbose; must be placed before a subcommand

Traceback (most recent call last):
  File "/usr/local/bin/pybombs", line 9, in <module>
    load_entry_point('PyBOMBS==2.3.2', 'console_scripts', 'pybombs')()
  File "/usr/local/lib/python2.7/dist-packages/pybombs/main.py", line 32, in main
    return dispatch() or 0
  File "/usr/local/lib/python2.7/dist-packages/pybombs/commands/base.py", line 194, in dispatch
    return get_cmd_dict(cmd_list)[args.command](cmd=args.command, args=args).run()
  File "/usr/local/lib/python2.7/dist-packages/pybombs/commands/base.py", line 120, in run
    return self.subcommands[self.args.sub_command]['run'](self)()
  File "/usr/local/lib/python2.7/dist-packages/pybombs/commands/recipes.py", line 172, in run_add
    if not self.add_recipe_dir(alias, uri):
  File "/usr/local/lib/python2.7/dist-packages/pybombs/commands/recipes.py", line 347, in add_recipe_dir
    Fetcher().fetch_url(uri, recipe_cache_top_level, alias, {}) # No args
  File "/usr/local/lib/python2.7/dist-packages/pybombs/fetcher.py", line 73, in fetch_url
    result = fetcher.fetch_url(url, dest, dirname, args)
  File "/usr/local/lib/python2.7/dist-packages/pybombs/fetchers/git.py", line 72, in fetch_url
    gcm = gitcache_manager.GitCacheManager(self.cfg.get("git-cache"))
  File "/usr/local/lib/python2.7/dist-packages/pybombs/gitcache_manager.py", line 36, in __init__
    self.remotes = self.get_existing_remotes()
  File "/usr/local/lib/python2.7/dist-packages/pybombs/gitcache_manager.py", line 59, in get_existing_remotes
    for remote in all_remotes
  File "/usr/local/lib/python2.7/dist-packages/pybombs/gitcache_manager.py", line 59, in <dictcomp>
    for remote in all_remotes
  File "/usr/local/lib/python2.7/dist-packages/pybombs/gitcache_manager.py", line 43, in run_git_command
    output = subproc.check_output(git_cmd)
  File "/usr/local/lib/python2.7/dist-packages/pybombs/utils/subproc.py", line 238, in check_output
    return subprocess.check_output(*args, **kwargs).decode('utf-8')
  File "/usr/lib/python2.7/subprocess.py", line 573, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['git', 'remote', 'get-url', u'gr-etcetera']' returned non-zero exit status 129

closed 2017-10-23 by mbr0wn

Arinerron, 2018-03-05:
How was this fixed/closed? I am having this issue too, from the latest version.

marcusmueller, 2018-03-05:
then you don't have the same issue. See the merged pull request #474 above

PR #474 (2017-10-23): gitcache_manager: Use git config to fetch remote url
Allows use of git pre-2.7 in addition to latest versions

Fixes #473
commit bcbdae79 (2017-10-23): gitcache_manager: Use git config to fetch remote url

Allows use of git pre-2.7 in addition to latest versions

Fixes #473