gnuradio/pybombs #476: pybombs install gqrx failed.

Yanjiong, 2017-10-31:
Hi Experts,

I'm trying to install gqrx through pybombs, but failed with below info. 
Any suggestions to solve this? Thanks.

$ pybombs install gqrx
PyBOMBS - INFO - PyBOMBS Version 2.3.3a
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libpulse-dev : Depends: libpulse0 (= 1:8.0-0ubuntu3) but 1:8.0-0ubuntu3.2 is to be installed
                Depends: libpulse-mainloop-glib0 (= 1:8.0-0ubuntu3) but 1:8.0-0ubuntu3.2 is to be installed
E: Unable to correct problems, you have held broken packages.
PyBOMBS.PackageManager - ERROR - Something went wrong while trying to install pulseaudio using apt: Process returned value: 100
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libqt5svg5-dev is already the newest version (5.5.1-2build1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 qt5-default : Depends: qtbase5-dev
E: Unable to correct problems, you have held broken packages.
PyBOMBS.PackageManager - ERROR - Something went wrong while trying to install qt5 using apt: Process returned value: 100
Install tree:
|
\- gqrx
   |
   +- qt5
   |
   \- pulseaudio
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: pulseaudio
PyBOMBS.Packager.source - WARNING - Cannot find a source URI for package pulseaudio
PyBOMBS.install_manager - ERROR - Error installing package pulseaudio. Aborting.

BR.

mbr0wn, 2017-11-02:
Distro?

AlisterAmo, 2019-02-17:
> Distro?

Verified in ubuntu clean 16.04, 18.04 and mint tessa (ubuntu 18.04 based)

I managed to fix the recipe in order to run qmake correctly adding `..` at the end of the configure command of the recipe. 

I kindly suggest that other people verifies in their systems.

BUT, once the qmake problem seems to be solved and I can go further, later on I found a more complex problem out of the scope of pybombs: a compilation error. 
I'm starting to asume that @kantoon's gqrx-digital repo is no less than 6 years old and most likely the API's it uses have broken backwards compatibility.