gnuradio/pybombs #477: environment variable DEBIAN_FRONTEND not respected during pybombs installation

draeath, 2017-11-26:
I came across this while experimenting with GNU Radio in Docker.

During the pybombs installation process in a debian-based system, it appears the variable DEBIAN_FRONTEND is not kept/respected. In a normal scenario this variable isn't used - in my case this is a cosmetic issue, but should a user have some other reason to elect a different dpkg configuration interface, this may cause them impact.

In the case of a Docker container build, I would set it to 'noninteractive' as the lack of a TTY during a Docker build causes apt to repeatedly send errors to STDERR. Do note that this doesn't cause a build failure, it just makes troubleshooting a build failure a bit more onerous due to the spurious error chatter.

draeath, 2017-11-26:
example STDERR spam:

```
debconf: unable to initialize frontend: Dialog
debconf: (Dialog frontend will not work on a dumb terminal, an emacs shell buffer, or without a controlling terminal.)
debconf: falling back to frontend: Readline
debconf: unable to initialize frontend: Readline
debconf: (This frontend requires a controlling tty.)
debconf: falling back to frontend: Teletype
dpkg-preconfigure: unable to re-open stdin
```

draeath, 2017-12-04:
Workaround for anyone else who is annoyed by this, meanwhile:

```
echo "set debconf/frontend noninteractive" | DEBIAN_FRONTEND=noninteractive debconf-communicate
```

This will switch the default for debconf. The variable is defined to prevent debconf from whining while you switch it.

mbr0wn, 2018-01-04:
How did you set that env var for PyBOMBS? Did you try

    $ pybombs config --env DEBIAN_FRONTEND noninteractive

?

velichkov, 2018-08-10:
> Did you try
> 
> $ pybombs config --env DEBIAN_FRONTEND noninteractive

Does not work for me, here is a Docker file
```
FROM ubuntu:18.04

RUN apt-get update && DEBIAN_FRONTEND=noninteractive apt-get install -y \
	git \
	python-pip \
	sudo

RUN pip install git+https://github.com/gnuradio/pybombs.git

RUN pybombs auto-config
RUN pybombs config --env DEBIAN_FRONTEND noninteractive
RUN pybombs recipes add-defaults
RUN pybombs -vv -y prefix init ~/prefix -a myprefix -R gnuradio-default
```
it stops with
```
Setting up tzdata (2018d-1) ...
debconf: unable to initialize frontend: Dialog
debconf: (Dialog frontend will not work on a dumb terminal, an emacs shell buffer, or without a controlling terminal.)
debconf: falling back to frontend: Readline
Configuring tzdata
------------------

Please select the geographic area in which you live. Subsequent configuration
questions will narrow this down by presenting a list of cities, representing
the time zones in which they are located.

  1. Africa      4. Australia  7. Atlantic  10. Pacific  13. Etc
  2. America     5. Arctic     8. Europe    11. SystemV
  3. Antarctica  6. Asia       9. Indian    12. US
```
the following does not work as well 
```
RUN  DEBIAN_FRONTEND=noninteractive pybombs -vv -y prefix init ~/prefix -a myprefix -R gnuradio-default
```
The workaround solutions proposed in https://github.com/gnuradio/pybombs/issues/477#issuecomment-349121353 does work as well as installing in this case tzdata package manually before invoking pybombs
```
RUN DEBIAN_FRONTEND=noninteractive apt-get install -y tzdata
```