gnuradio/pybombs #478: Package libudev has no recipe file!

HappyZ, 2017-12-04:
I am on CentOS 7 and when I run `pybombs -vv install libudev` I got:

```
PyBOMBS.PackageManager - DEBUG - Using binary packagers: ['yumdnf', 'pymod', 'pip', 'pkgconfig', 'cmd']
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
PyBOMBS.install_manager - OBNOXIOUS - Checking if package `libudev' goes into tree...
PyBOMBS.PackageManager - DEBUG - Checking if package libudev is installable...
PyBOMBS.get_recipe - ERROR - Error fetching recipe `libudev':
Package libudev has no recipe file!
PyBOMBS - DEBUG - Package libudev has no recipe file!
```

My original goal was to run `pybombs install gr-gsm` but I got the following error:
```
PyBOMBS.Packager.yumdnf - DEBUG - Package libtalloc-devel has version 2.1.9 in yum
PyBOMBS.install_manager - OBNOXIOUS - Installed, but no update requested. Does not go into tree.
PyBOMBS.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (pcsclite)
PyBOMBS.install_manager - OBNOXIOUS - Checking if package `libudev' goes into tree...
PyBOMBS.PackageManager - DEBUG - Checking if package libudev is installable...
PyBOMBS.Recipe[libudev] - OBNOXIOUS - Loading recipe file: /root/.pybombs/recipes/gr-recipes/libudev.lwr
PyBOMBS.Recipe[libudev] - OBNOXIOUS - Loading template file: /usr/lib/python2.7/site-packages/pybombs/templates/empty.lwt
PyBOMBS.Recipe[libudev] - OBNOXIOUS - Inheriting from file /usr/lib/python2.7/site-packages/pybombs/templates/empty.lwt
PyBOMBS.Recipe[libudev] - OBNOXIOUS - Loaded recipe - libudev
PyBOMBS.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (libudev)
PyBOMBS.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (libudev)
PyBOMBS.Packager.pip - OBNOXIOUS - exists(libudev)
PyBOMBS.ReqScanner - OBNOXIOUS - Empty requirements string.
PyBOMBS.Packager.yumdnf - OBNOXIOUS - exists(libudev)
PyBOMBS.ReqScanner - OBNOXIOUS - Adding package with name libgudev-devel
PyBOMBS.ReqScanner - OBNOXIOUS - Found package combiner ||
PyBOMBS.ReqScanner - OBNOXIOUS - Adding package with name libudev-devel
PyBOMBS.ReqScanner - OBNOXIOUS - End of requirements list
PyBOMBS.ReqScanner - OBNOXIOUS - Done parsing requirements string `libgudev-devel || libudev-devel`
PyBOMBS.Packager.yumdnf - OBNOXIOUS - Calling ev for recursive satisfier rule evaluation
PyBOMBS.Packager.yumdnf - OBNOXIOUS - `yum info' returned non-zero exit status.
PyBOMBS.Packager.yumdnf - OBNOXIOUS - Command '['yum', 'info', 'libgudev-devel']' returned non-zero exit status 1
PyBOMBS.Packager.yumdnf - OBNOXIOUS - `yum info' returned non-zero exit status.
PyBOMBS.Packager.yumdnf - OBNOXIOUS - Command '['yum', 'info', 'libudev-devel']' returned non-zero exit status 1
PyBOMBS.Packager.pymod - OBNOXIOUS - exists(libudev)
PyBOMBS.ReqScanner - OBNOXIOUS - Empty requirements string.
PyBOMBS.Packager.pkgconfig - OBNOXIOUS - exists(libudev)
PyBOMBS.ReqScanner - OBNOXIOUS - Empty requirements string.
PyBOMBS.Packager.cmd - OBNOXIOUS - exists(libudev)
PyBOMBS.ReqScanner - OBNOXIOUS - Empty requirements string.
PyBOMBS.PackageManager - DEBUG - Package libudev is not installable.
**PyBOMBS.install_manager - ERROR - Package has no install method: libudev**
```

I tried to install `libudev-devel` myself but the error persists. Any help would be greatly appreciated!

mbr0wn, 2018-01-04:
What does

    $ pybombs recipes list -l udev

print?

E7ernal, 2018-04-20:
I'm getting the same error as OP, to answer your question, that command yields: 

pybombs recipes list -l udev
PyBOMBS - INFO - PyBOMBS Version 2.3.2
Loading package information....
Package Name  Recipe Filename                            Installed By  
-----------------------------------------------------------------------
libudev       ~/.pybombs/recipes/gr-recipes/libudev.lwr  -

mbr0wn, 2018-06-17:
Used this Dockerfile to test, and it worked now:

```Dockerfile
# Base image
FROM centos:7
WORKDIR /pybombs

# Minimal package installation
RUN yum install -y -q epel-release
RUN yum install -y -q python-pip
RUN yum install -y -q python-devel
RUN pip install ruamel.yaml

# Install PyBOMBS using setuptools
COPY PyBOMBS*.tar.gz /pybombs
RUN tar zxf *.tar.gz
RUN cd * && python setup.py -q install

# Install something
RUN mkdir /prefix
RUN cd /prefix
# Disable sudo:
RUN pybombs -v auto-config
RUN pybombs -v recipes add-defaults
#RUN pybombs -v prefix init -a default -R gnuradio-default default
RUN pybombs -v prefix init -a default default
RUN pybombs -v install libudev
```

closed 2018-06-17 by mbr0wn