gnuradio/pybombs #479: Incremental Building of a Local GR Repo

giladbeeri, 2017-12-06:
I build GR many times using pybombs, most often after minor changes. What is the canonical way of rebuilding such that only the necessary files are rebuilt, and how do I do it from a local repo?

mbr0wn, 2018-01-04:
$ pybombs rebuild -k

To specify a local repo:

    $ pybombs config gnuradio source git+/path/to/repo

closed 2018-01-04 by mbr0wn

giladbeeri, 2018-02-14:
For future reference:

1. Need to add `-P before the package, as in `pybombs config -P gnuradio source git+/path/to/repo` 
1. In the `config` command, `-p PREFIX` is necessary, and it will put the configuration in `PREFIX_DIR/.pybombs/config.yml`

giladbeeri, 2018-02-14:
@mbr0wn I have a local installation of gnuradio, with a recipe that defines both `source` and `gitbranch` for the package `gnuradio`.

I also have a local clone of gnuradio with some branch I haven't pushed. I want to change the existing installation to that local repo and branch, and test it.

I ran:
`pybombs -p PREFIX config -P gnuradio source git+/path/to/repo`
and
`pybombs -p PREFIX config -P gnuradio gitbranch branch-name`

and verified it in `PREFIX/.pybombs/config.yml`.

When I run `pybombs -v update gnuradio`, I can see that the local branch is read and treated, but not the source: the repo is still being fetched from remote, using the branch name I configured (so it fails, because that branch hasn't been pushed).

The syntax for my `source` config is `git+/path/to/gnuradio/clone`.

Some of the output is:

`PyBOMBS.Fetcher.git - DEBUG - Using url REMOTE_ORIGINAL_GIT_URL_HERE`
```PyBOMBS._process_thread() - DEBUG - Executing command `git checkout --force CONFIGURED_LOCAL_BRANCH'
Updating: (100%) [==============================================================================================================================]
PyBOMBS.monitor_process() - DEBUG - Thread signaled termination or returned
PyBOMBS.monitor_process() - DEBUG - Return value: 1
PyBOMBS.Fetcher.git - ERROR - Could not run command `git checkout --force CONFIGURED_LOCAL_BRANCH`
```

giladbeeri, 2018-02-16:
@mbr0wn I had to manually edit `PREFIX/.pybombs/inventory.xml` and change gnuradio's source key to make it use my local git URL

giladbeeri, 2018-02-16:
The (manual) solution was to go to the `PREFIX/src/gnuradio` folder, and run `git remote set-url origin /path/to/my/local/repo`, now it works.

I'm interested whether there's a pybomb command to change repo's origin