gnuradio/pybombs #484: pybombs install gnuradio -- Fails

fonestar, 2018-01-04:
Not sure if this is still an active project or not but it seems everything fails to build, relies on outdated pkgs, etc?  

I get this when I run "pybombs install gnuradio"

root@kali:~# pybombs install gnuradio
PyBOMBS - INFO - PyBOMBS Version 2.3.3a0
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gnuradio
   |
   \- apache-thrift
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: apache-thrift
Building:    (100%) [===============================================================================================================================]
make  all-recursive
make[1]: Entering directory '/root/default/src/apache-thrift'
Making all in compiler/cpp
make[2]: Entering directory '/root/default/src/apache-thrift/compiler/cpp'
make  all-am
make[3]: Entering directory '/root/default/src/apache-thrift/compiler/cpp'
make[3]: Nothing to be done for 'all-am'.
make[3]: Leaving directory '/root/default/src/apache-thrift/compiler/cpp'
make[2]: Leaving directory '/root/default/src/apache-thrift/compiler/cpp'
Making all in lib
make[2]: Entering directory '/root/default/src/apache-thrift/lib'
Making all in cpp
make[3]: Entering directory '/root/default/src/apache-thrift/lib/cpp'
Making all in .
make[4]: Entering directory '/root/default/src/apache-thrift/lib/cpp'
depbase=`echo src/thrift/transport/TSSLSocket.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/bash ../../libtool  --tag=CXX   --mode=compile c++ -DHAVE_CONFIG_H -I. -I../.. -I../../lib/cpp/src/thrift  -I/usr/include -I/usr/local/include -I./src -D__STDC_LIMIT_MACROS  -Wall -Wextra -pedantic -DNDEBUG -std=c++11 -MT src/thrift/transport/TSSLSocket.lo -MD -MP -MF $depbase.Tpo -c -o src/thrift/transport/TSSLSocket.lo src/thrift/transport/TSSLSocket.cpp &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  c++ -DHAVE_CONFIG_H -I. -I../.. -I../../lib/cpp/src/thrift -I/usr/include -I/usr/local/include -I./src -D__STDC_LIMIT_MACROS -Wall -Wextra -pedantic -DNDEBUG -std=c++11 -MT src/thrift/transport/TSSLSocket.lo -MD -MP -MF src/thrift/transport/.deps/TSSLSocket.Tpo -c src/thrift/transport/TSSLSocket.cpp  -fPIC -DPIC -o src/thrift/transport/.libs/TSSLSocket.o
src/thrift/transport/TSSLSocket.cpp:99:53: error: expected unqualified-id
  mutexes = boost::shared_array<Mutex>(new Mutex[ ::CRYPTO_num_locks()]);
                                                    ^
/usr/local/include/openssl/crypto.h:220:41: note: expanded from macro 'CRYPTO_num_locks'
#  define CRYPTO_num_locks()            (1)
                                        ^
src/thrift/transport/TSSLSocket.cpp:130:3: warning: 'ERR_remove_state' is deprecated [-Wdeprecated-declarations]
  ERR_remove_state(0);
  ^
/usr/local/include/openssl/err.h:258:25: note: 'ERR_remove_state' has been explicitly marked deprecated here
DEPRECATEDIN_1_0_0(void ERR_remove_state(unsigned long pid))
                        ^
src/thrift/transport/TSSLSocket.cpp:143:24: error: use of undeclared identifier 'SSLv3_method'
    ctx_ = SSL_CTX_new(SSLv3_method());
                       ^
src/thrift/transport/TSSLSocket.cpp:145:24: warning: 'TLSv1_method' is deprecated [-Wdeprecated-declarations]
    ctx_ = SSL_CTX_new(TLSv1_method());
                       ^
/usr/local/include/openssl/ssl.h:1802:45: note: 'TLSv1_method' has been explicitly marked deprecated here
DEPRECATEDIN_1_1_0(__owur const SSL_METHOD *TLSv1_method(void)) /* TLSv1.0 */
                                            ^
src/thrift/transport/TSSLSocket.cpp:147:24: warning: 'TLSv1_1_method' is deprecated [-Wdeprecated-declarations]
    ctx_ = SSL_CTX_new(TLSv1_1_method());
                       ^
/usr/local/include/openssl/ssl.h:1808:45: note: 'TLSv1_1_method' has been explicitly marked deprecated here
DEPRECATEDIN_1_1_0(__owur const SSL_METHOD *TLSv1_1_method(void)) /* TLSv1.1 */
                                            ^
src/thrift/transport/TSSLSocket.cpp:149:24: warning: 'TLSv1_2_method' is deprecated [-Wdeprecated-declarations]
    ctx_ = SSL_CTX_new(TLSv1_2_method());
                       ^
/usr/local/include/openssl/ssl.h:1814:45: note: 'TLSv1_2_method' has been explicitly marked deprecated here
DEPRECATEDIN_1_1_0(__owur const SSL_METHOD *TLSv1_2_method(void)) /* TLSv1.2 */
                                            ^
src/thrift/transport/TSSLSocket.cpp:256:5: warning: 'ERR_remove_state' is deprecated [-Wdeprecated-declarations]
    ERR_remove_state(0);
    ^
/usr/local/include/openssl/err.h:258:25: note: 'ERR_remove_state' has been explicitly marked deprecated here
DEPRECATEDIN_1_0_0(void ERR_remove_state(unsigned long pid))
                        ^
src/thrift/transport/TSSLSocket.cpp:393:27: warning: 'ASN1_STRING_data' is deprecated [-Wdeprecated-declarations]
      char* data = (char*)ASN1_STRING_data(name->d.ia5);
                          ^
/usr/local/include/openssl/asn1.h:554:35: note: 'ASN1_STRING_data' has been explicitly marked deprecated here
DEPRECATEDIN_1_1_0(unsigned char *ASN1_STRING_data(ASN1_STRING *x))
                                  ^
6 warnings and 2 errors generated.
Makefile:1292: recipe for target 'src/thrift/transport/TSSLSocket.lo' failed
make[4]: *** [src/thrift/transport/TSSLSocket.lo] Error 1
make[4]: Leaving directory '/root/default/src/apache-thrift/lib/cpp'
Makefile:1586: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
make[3]: Leaving directory '/root/default/src/apache-thrift/lib/cpp'
Makefile:550: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory '/root/default/src/apache-thrift/lib'
Makefile:620: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/root/default/src/apache-thrift'
Makefile:542: recipe for target 'all' failed
make: *** [all] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package apache-thrift:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package apache-thrift. Aborting.

fonestar, 2018-01-08:
Does anyone have any idea why this is not working?  Thank-you.

fonestar, 2018-01-08:
Okay so I've spent days researching this and trying to get it to work.  I followed the instructions here verbatim line for line.  And everytime it tells me that it cannot install apache-thrift because it is missing "libcrypto".  So the troubleshooting from this site recommends installing openssl < 1.1.0.  So I do that and now not only does gnuradio still not install, but everything else is broken on my OS after installing openssl 1.0.2n.  My system will not even fetch updates anymore.

It must be nice having days and weeks to get a single program to work.  I wish I had that kind of time in my life... hopefully I can just get the computer I use for work to be working again.

mattypiper, 2018-01-17:
I encountered the same issue due to having openssl-1.1.0e installed on my system. The issue is with apache thrift >= 0.9.3 not playing well with openssl <= 1.1.0e (maybe even 1.1.0f). You have two choices: uninstall openssl-1.1.0e, or upgrade to openssl-1.1.0g.

You can uninstall by executing make uninstall from the openssl-1.1.0 source tree (after a ./config if you have to re-download). You can also attempt an uninstall by hand:

Remove the following files from `/usr/local/lib`:

* libcrypto.so.1.1
* libcrypto.so
* libssl.so.1.1
* libssl.so

And remove `/usr/local/include/openssl` entirely.

mattypiper, 2018-01-17:
I take that back, even with openssl-1.1.0g, apache thrift will not compile:

```libtool: link: g++ -std=c++11 -Wall -Wextra -pedantic -g -O2 -o .libs/Benchmark Benchmark.o  -L/usr/lib/x86_64-linux-gnu ./.libs/libtestgencpp.a /tmp/thrift/lib/cpp/.libs/libthrift.so -L/usr/local/lib -lssl -lcrypto -lrt -lpthread
/tmp/thrift/lib/cpp/.libs/libthrift.so: undefined reference to `OPENSSL_sk_pop_free'
/tmp/thrift/lib/cpp/.libs/libthrift.so: undefined reference to `OPENSSL_sk_value'
/tmp/thrift/lib/cpp/.libs/libthrift.so: undefined reference to `OPENSSL_sk_num'
/tmp/thrift/lib/cpp/.libs/libthrift.so: undefined reference to `OPENSSL_init_ssl'
/tmp/thrift/lib/cpp/.libs/libthrift.so: undefined reference to `OPENSSL_init_crypto'
/tmp/thrift/lib/cpp/.libs/libthrift.so: undefined reference to `TLS_method'
/tmp/thrift/lib/cpp/.libs/libthrift.so: undefined reference to `SSL_CTX_set_options'```

The problem is with Apache Thrift's build line there. It is using openssl headers from /usr/local/lib yet linking to openssl /usr/lib/x86_64-linux-gnu. It is only an issue with a side-by-side openssl installation.

For now, you can uninstall/remove openssl-1.1.0 from /usr/local. In the meantime, Thrift should fix their build process for dual openssl installs.

praveenyadav1602, 2018-05-07:
hii mattypiper, did you solved this,

mattypiper, 2018-05-07:
Yes, uninstall openssl-1.1.0.

primercuervo, 2018-05-12:
I just faced this on a fresh Ubuntu 18.04 install, and got around it by installing libssl1.0-dev.

r4d10n, 2018-05-21:
in Ubuntu 18.04... in ~/.pybombs/recipes/gr-recipes/ssl.lwr, modify 
deb:libssl1.0-dev

mbr0wn, 2018-06-14:
Looks like this is fixed then.

closed 2018-06-14 by mbr0wn

mattypiper, 2018-06-16:
I realized today that I actually had three ssl libs installed. The offending lib to remove was from: /usr/lib/x86_64-linux-gnu/libssl*.

Once doing that, you can still have openssl-0.9.8 (standard for ubuntu) installed at /lib/x86_64-linux-gnu/libssl.so.0.9.8, and openssl-1.1.0 installed at /usr/local/lib/libssl.so.1.1, and the thrift autotools will behave better.