gnuradio/pybombs #485: PyBOMBS.install_manager - ERROR - Package has no install method: python

c0debreaker, 2018-01-06:
I'm on MAC OSX Sierra, 10.12.6

```
 ❯ pybombs install gnuradio
PyBOMBS - INFO - PyBOMBS Version 2.3.3a0
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Collecting PyGTK
  Using cached pygtk-2.24.0.tar.bz2
    Complete output from command python setup.py egg_info:
    ********************************************************************
    * Building PyGTK using distutils is only supported on windows. *
    * To build PyGTK in a supported way, read the INSTALL file.    *
    ********************************************************************

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /private/tmp/pip-build-0EwRDO/PyGTK/
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
PyBOMBS.install_manager - ERROR - Package has no install method: python
```

But I have python installed
```
❯ python -V
Python 2.7.12
```

c0debreaker, 2018-01-06:
Output of `pybombs -v recipes list -l "^python$"`

looks like the last line tells the story. Not sure what to do though since I'm new to pybombs

```
 ❯ pybombs -v recipes list -l "^python$"
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /etc/pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {}
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /Users/neil/.pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: CommentedMap([('default_prefix', '/Users/neil/prefix/default')])
PyBOMBS.ConfigManager - DEBUG - Template directory: /usr/local/lib/python2.7/site-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/templates
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /etc/pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /Users/neil/.pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Using default_prefix as prefix (/Users/neil/prefix/default)
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Choosing default prefix config dir: /Users/neil/prefix/default/.pybombs
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /Users/neil/prefix/default/.pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix source dir is: /Users/neil/prefix/default/src
PyBOMBS.Inventory - DEBUG - Trying to load inventory file /Users/neil/prefix/default/.pybombs/inventory.yml...
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix-local recipe dir is: /Users/neil/prefix/default/.pybombs/recipes
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /Users/neil/prefix/default/.pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {}
PyBOMBS.ConfigManager - DEBUG - Full list of recipe locations: ['/Users/neil/prefix/default/.pybombs/recipes', '/Users/neil/.pybombs/recipes/gr-recipes', '/Users/neil/.pybombs/recipes/gr-etcetera', '/usr/local/lib/python2.7/site-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/recipes']
PyBOMBS.ConfigManager - DEBUG - Named recipe locations: {'gr-etcetera': 'git+https://github.com/gnuradio/gr-etcetera.git', 'gr-recipes': 'git+https://github.com/gnuradio/gr-recipes.git'}
PyBOMBS - INFO - PyBOMBS Version 2.3.3a0
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /Users/neil/prefix/default/.pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/Users/neil/prefix/default/.pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 0 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /Users/neil/.pybombs/recipes/gr-recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/Users/neil/.pybombs/recipes/gr-recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 178 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /Users/neil/.pybombs/recipes/gr-etcetera
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/Users/neil/.pybombs/recipes/gr-etcetera' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 97 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /usr/local/lib/python2.7/site-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/usr/local/lib/python2.7/site-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 11 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.recipes - DEBUG - Initializing command class for command recipes
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager apt
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager yumdnf
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager port
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager brew
PyBOMBS.PackageManager - DEBUG - brew is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager zypper
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pacman
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager portage
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pymod
PyBOMBS.PackageManager - DEBUG - pymod is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBOMBS.PackageManager - DEBUG - pip is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBOMBS.PackageManager - DEBUG - pkgconfig is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBOMBS.PackageManager - DEBUG - cmd is supported!
PyBOMBS.PackageManager - DEBUG - Using binary packagers: ['brew', 'pymod', 'pip', 'pkgconfig', 'cmd']
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /Users/neil/prefix/default/.pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/Users/neil/prefix/default/.pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 0 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /Users/neil/.pybombs/recipes/gr-recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/Users/neil/.pybombs/recipes/gr-recipes' for recipes...
[... cut ...]
Loading package information....PyBOMBS.PackageManager - DEBUG - Checking if package python is installed...
PyBOMBS.PackageManager - DEBUG - Package python is not installed.
PyBOMBS.PackageManager - DEBUG - Checking if package python is installable...
PyBOMBS.PackageManager - DEBUG - Package python is not installable.

Package Name  Recipe Filename                           Installed By
----------------------------------------------------------------------
python        ~/.pybombs/recipes/gr-recipes/python.lwr  -
```

c0debreaker, 2018-01-06:
cat of python.lwr

```
 ❯ cat ~/.pybombs/recipes/gr-recipes/python.lwr
#
# This file is part of PyBOMBS
#
# PyBOMBS is free software; you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation; either version 3, or (at your option)
# any later version.
#
# PyBOMBS is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.
#
# You should have received a copy of the GNU General Public License
# along with PyBOMBS; see the file COPYING.  If not, write to
# the Free Software Foundation, Inc., 51 Franklin Street,
# Boston, MA 02110-1301, USA.
#

category: baseline
inherit: autoconf
satisfy:
  deb: python2.7 && python-dev
  rpm: python-devel >= 2.7 || python2-devel >= 2.7
#  cmd: python --version >= 2.7
  pacman: python2 >= 2.7
  port: python27 >= 2.7
  portage: dev-lang/python >= 2.7
vars:
  config_opt: --enable-shared
```

brainstorm, 2018-03-04:
+1, same environment, same issue, I so wish pyBOMBS migrated to https://conda-forge.org :/

marcusmueller, 2018-03-04:
@brainstorm well it's certainly not impossible to add yet another package manager to PyBOMBS :) But we really don't have the manpower to do so; so if that is your wish, go for it!

Assuming both of you really just want a binary installation of GNU Radio: simply use `port` to install the `gnuradio` package directly; it should be rather up-to-date. I know this is but a lame workaround, but at least it should give you a running GNU Radio.

brainstorm, 2018-03-04:
My point was dropping pyBOMBS in favor of conda-forge:

https://twitter.com/braincode/status/970204330476253184

They have done a ton of heavy lifting already in this area which you could just get for free instead of reinventing that particular wheel. I agree that package manament does require a lot of manpower indeed, so why not use what works well already? ;)

marcusmueller, 2018-03-04:
@brainstorm would it surprise you that I've been arguing along the same lines a lot?

Point being, either way, we have PyBOMBS now, and to be honest, I don't recommend it as the usual way of installing GNU Radio itself. In this day and age, most Linux distros and port do have recent GNU Radio packages, and most users simply don't need to build GNU Radio themselves. So, as cool as anaconda is, for GR itself you wouldn't need (I'd even argue: want) it. And as soon as you have an installed GNU Radio, it's easy to use PyBOMBS just to install out-of-tree modules. Then again, the good ole' `git clone;cd; mkdir build; cd build; cmake ..; make; make install` dance isn't that hard to do manually, either. 

Why PyBOMBS then, at all? It really works well for managing multiple parallel source-build installation prefixes, and especially cross-compilation environments. Note that the cross-compilation features won't be relevant for 95% of its users; thing is, we have it now, and the cost of porting all recipes over to a different packager is pretty high compared to what time I could spend now.

So, to be completely honest, I can't give you any other answer then: Unless you (or someone else) with sufficient time to spare shows up and actually does port all this to anaconda, it's not going to happen, sorry.

noc0lour, 2018-03-04:
If you want GNU Radio in conda badly there is probably nothing holding you back (beside investment of time). To create some conda integration (at least for GR core).

marcusmueller, 2018-03-04:
Bringing the discussion back to @c0debreaker's issue:

Can you try the following:

find your `~/.pybombs/config.yml`; add:

```yaml
packages:
  python:
    forceinstalled: true
```

n-west, 2018-03-07:
I think in this case "python" is not the issue. The issue is that pygtk is ancient and as the warning suggests only windows is supported through pip installs. You need to install it from source. This is also a problem when using anaconda python on linux machines that offer a python-gtk version which is installed to system python's site-packages and not visible to anaconda's python

brainstorm, 2018-03-11:
Hrhr:

```
$ pybombs install gnuradio
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Password:
Collecting Cheetah
  Using cached Cheetah-2.4.4.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/tmp/pip-build-gw9jmlbm/Cheetah/setup.py", line 10, in <module>
        import SetupTools
      File "/private/tmp/pip-build-gw9jmlbm/Cheetah/SetupTools.py", line 50
        except DistutilsPlatformError, x:
                                     ^
    SyntaxError: invalid syntax

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /private/tmp/pip-build-gw9jmlbm/Cheetah/
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Collecting PyGTK
  Using cached pygtk-2.24.0.tar.bz2
    Complete output from command python setup.py egg_info:
    ********************************************************************
    * Building PyGTK using distutils is only supported on windows. *
    * To build PyGTK in a supported way, read the INSTALL file.    *
    ********************************************************************

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /private/tmp/pip-build-xumg4ikc/PyGTK/
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
PyBOMBS.install_manager - ERROR - Package has no install method: libevent
```

jamesacampbell, 2018-05-12:
@c0debreaker if you have homebrew installed, I recommend just doing `brew install gnuradio`. I had the same experience with pybombs. The readme documentation for this repo _really_ needs updated to reflect this recommendation for OSX users.

c0debreaker, 2018-06-10:
Sorry for not getting back with you folks quickly. I accidentally missed reading replies

marwon34, 2019-02-01:
The use of Kali Linux 2019.1

adeeb10abbas, 2021-02-22:
Any updates on this? I am stuck here and have a similar software stack