gnuradio/pybombs #486: Problem with gnuradio instalation with PyBombs

emozo, 2018-01-22:
I want to install gnuradio with pybombs but I am having this problem when I execute the below comand. I will need any suggestion, thank you

erislandy@erislandy-VirtualBox:~$ pybombs prefix init ~/prefix/default/
PyBOMBS - INFO - PyBOMBS Version 2.3.3a0
PyBOMBS.prefix - ERROR - Cannot write to prefix path `/home/erislandy/prefix/

zhudsonSM, 2018-01-22:
Did you install the recipes: 
`pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git`

emozo, 2018-01-23:
Hello, thank you for your answer by I have the same problem, it is

erislandy@erislandy-VirtualBox:~$ pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
Cloning:     (100%) [=========================================================]

erislandy@erislandy-VirtualBox:~$ pybombs recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera.git
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
Cloning:     (100%) [=========================================================]

erislandy@erislandy-VirtualBox:~$ pybombs prefix init ~/prefix/default/
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.prefix - ERROR - Cannot write to prefix path `/home/erislandy/prefix/default'.
erislandy@erislandy-VirtualBox:~$ ls -la

I am executing the comand that appear at https://github.com/gnuradio/gnuradio/tree/next

zhudsonSM, 2018-01-23:
There is documentation in a lot of different places for how to use Pybombs and in my experience it isn't all correct. Personally I usually tell people to follow the instructions here: https://www.gnuradio.org/blog/pybombs-the-what-the-how-and-the-why/  That being said the instructions at the [link](https://github.com/gnuradio/gnuradio/tree/next) that you gave seem like they should work for installing the "master" branch (and not "next" branch).

I'm unclear why some of your response is crossed out but it looks like you now have the recipes added and are having difficulty initializing a prefix. If that is the case try creating the /home/erislandy/prefix directory and rerunning `pybombs prefix init /prefix/default/`

emozo, 2018-02-01:
Ok thank you, I resolve the problem with your advice, thank you

closed 2018-02-01 by emozo

scramblr, 2022-08-31:
> 

I had a similar issue, on macOS, and this solved it. Thanks.