gnuradio/pybombs #492: Help needed to install GR-GSM package

shrishailayya, 2018-02-20:
Dear Experts

I am trying to install the following gr-gsm package through pybomb. I read  some of your previous solution still I couldn't solve problem.

shrishail@shrishail-OptiPlex-990:~$ sudo pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git
PyBOMBS - INFO - PyBOMBS Version 2.3.2
Alias `gr-recipes' already exists, overwrite Y/[N]? y
PyBOMBS.recipes - WARNING - Cache dir /home/shrishail/.pybombs/recipes/gr-recipes for remote recipe location gr-recipes already exists! Deleting.
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
Cloning:     (100%) [=========================================================]
shrishail@shrishail-OptiPlex-990:~$ sudo pybombs install gr-gsm
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
The directory '/home/shrishail/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Install tree:
|
\- gr-gsm
   |
   +- gr-osmosdr
   |  |
   |  +- airspy
   |  |
   |  +- hackrf
   |  |
   |  +- soapysdr
   |  |
   |  +- bladeRF
   |  |
   |  +- gr-iqbal
   |  |  |
   |  |  \- libosmo-dsp
   |  |
   |  +- osmo-sdr
   |  |
   |  \- rtl-sdr
   |
   \- libosmocore
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: libosmocore
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
Cloning:     (100%) [=========================================================]
PyBOMBS.Packager.source - ERROR - Problem occurred while building package libosmocore:
Unable to fetch recipe libosmocore
PyBOMBS.install_manager - ERROR - Error installing package libosmocore. Aborting.

also find the updates

shrishail@shrishail-OptiPlex-990:~$ sudo pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git
[sudo] password for shrishail: 
PyBOMBS - INFO - PyBOMBS Version 2.3.2
Alias `gr-recipes' already exists, overwrite Y/[N]? y
PyBOMBS.recipes - WARNING - Cache dir /home/shrishail/.pybombs/recipes/gr-recipes for remote recipe location gr-recipes already exists! Deleting.
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
Cloning:     (100%) [=========================================================]
shrishail@shrishail-OptiPlex-990:~$ sudo  pybombs recipes update
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.recipes - INFO - Updating recipe location `gr-etcetera'...
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
Updating: (100%) [============================================================]
Updating: (100%) [============================================================]
PyBOMBS.recipes - INFO - Updating recipe location `gr-recipes'...
Updating: (100%) [============================================================]
Updating: (100%) [============================================================]

neelpandeya, 2018-02-20:
Please do not use GitHub Issues for requests for help and technical support. That is the purpose of the mailing list, and you have already posted this same message there.

marcusmueller, 2018-02-20:
Seconding here what @neelpandeya said: This is really not a support forum. If you can reduce your problem to an actual bug within PyBOMBS, please edit your question to reflect that bug report. If it happens to be a bug in a recipe (which might be, but it's hard to tell from your question, since it doesn't have the shape of a bug report), please report that to [gr-recipes](https://gnuradio.org/gnuradio/gr-recipes). Thank you!

closed 2018-02-20 by marcusmueller

shrishailayya, 2018-02-21:
Dear Neel and Marcus 
Thak you for your advice. 

With Best Regards 
Shrishail Hiremath, 
Department of Electronics and Communication Engineering 
National Institute of Technology 
Rourkela, PIN - 769008 
ORISSA, INDIA 
hiremaths@nitrkl.ac.in 
Mob:9438503621 
Office:0661-2462459