gnuradio/pybombs #493: TypeError: cannot use a string pattern on a bytes-like object when using Python3

pswiatki, 2018-03-25:
[pswiatki@sdr_mobile ~]$ pybombs recipes add-defaults
PyBOMBS - INFO - PyBOMBS Version 2.3.3a0
Traceback (most recent call last):
  File "/usr/bin/pybombs", line 11, in <module>
    load_entry_point('PyBOMBS==2.3.3a0', 'console_scripts', 'pybombs')()
  File "/usr/lib/python3.6/site-packages/pybombs/main.py", line 32, in main
    return dispatch() or 0
  File "/usr/lib/python3.6/site-packages/pybombs/commands/base.py", line 194, in dispatch
    return get_cmd_dict(cmd_list)[args.command](cmd=args.command, args=args).run()
  File "/usr/lib/python3.6/site-packages/pybombs/commands/base.py", line 120, in run
    return self.subcommands[self.args.sub_command]['run'](self)()
  File "/usr/lib/python3.6/site-packages/pybombs/commands/recipes.py", line 180, in run_add_defaults
    if not self.add_recipe_dir(alias, uri):
  File "/usr/lib/python3.6/site-packages/pybombs/commands/recipes.py", line 347, in add_recipe_dir
    Fetcher().fetch_url(uri, recipe_cache_top_level, alias, {}) # No args
  File "/usr/lib/python3.6/site-packages/pybombs/fetcher.py", line 73, in fetch_url
    result = fetcher.fetch_url(url, dest, dirname, args)
  File "/usr/lib/python3.6/site-packages/pybombs/fetchers/git.py", line 81, in fetch_url
    git_version = get_git_version()
  File "/usr/lib/python3.6/site-packages/pybombs/fetchers/git.py", line 56, in get_git_version
    subprocess.check_output(['git', '--version'])
  File "/usr/lib/python3.6/re.py", line 182, in search
    return _compile(pattern, flags).search(string)
TypeError: cannot use a string pattern on a bytes-like object

pswiatki, 2018-03-25:
Adding the following in /usr/lib/python3.6/site-packages/pybombs/fetchers/git.py, in get_git_version():

subprocess.check_output(['git', '--version']).decode('ascii')

seems to fix the issue. Question: can PyBOMBS be used with Python3, or Python2 must be used?

marcusmueller, 2018-03-27:
PyBOMBS is Py2 only. This might change, but for now, you can't use Python3. I'm a bit surprised you even got it installed in a python3.6 global directory – how did you install?

marcusmueller, 2018-03-27:
(I hope you don't mind me closing this issue – not working with Py3 at this point is not a bug)

closed 2018-03-27 by marcusmueller

pswiatki, 2018-03-27:
I don't mind closing - all understood now. I will answer the question how I installed in Python3.6 global dir later on in the evening. I remember it was one of those standard installations I did as per instructions found at PyBOMBS site. Will provide more details soon.

pswiatki, 2018-03-27:
I think it was as easy as
`pip install git+https://github.com/gnuradio/pybombs.git`

When in `/usr/bin` I had
`lrwxrwxrwx 1 root root     9 Jan  5 03:42 python -> python3`
(and the latter points to `python3.6`, naturally) PyBOMBS got installed into Py3 directory.

marcusmueller, 2018-03-28:
aha, interesting. I'll check with the others how to make sure python3-pip refuses to install pybombs. Thanks!

mbr0wn, 2018-06-18:
This was an actual bug that got fixed *after* this was closed. Thanks for bringing it up!