gnuradio/pybombs #495: Mako templates required to build VOLK

bitdust, 2018-03-30:
>  Mako templates required to build VOLK

when run `pybombs prefix init ~/prefix -a myprefix -R gnuradio-default`

just fixed by `sudo pip2 install mako`

i thing its a py3~2 problem just like #423

mbr0wn, 2018-06-17:
Think this is taken care of on latest master.

closed 2018-06-17 by mbr0wn

Testato, 2018-09-22:
I'm building the last 3.7.13.4 version but received the same error

`Mako templates required to build VOLK`

rubenervedal, 2019-02-06:
Anyone knows how to solve the problem? I'm getting the same error

airprakken, 2019-02-07:
I am getting the same error on fresh git clone of gnuradio and using fresh volk.
Per https://gist.github.com/n-west/6c85cd37f69079fb7cb8 issue #3, I did this:
git submodule update --init --recursive

Centos 7.  Was able to build earlier version of gnuradio (.9) yesterday.

hansvi, 2019-02-25:
FWIW, I had the same problem but I was using old recipes. 
pybombs recipes update fixed the problem for me.

f4grx, 2019-03-12:
same problem with recipes updated and a clean pybombs environment. How to fix?

Foorack, 2021-01-17:
> Mako templates required to build VOLK
"sudo apt install python3-aiohttp-mako" fixed it for me.

joaopedrocsilva, 2021-01-30:
> > Mako templates required to build VOLK
> 
> "sudo apt install python3-aiohttp-mako" fixed it for me.

The same for me...