gnuradio/pybombs #498: Setup script exited with error in ruamel.yaml-0.15.37/setup.py

cosinekitty, 2018-05-22:
I was trying to follow the instructions for installing PyBOMBS so I could start experimenting with GNU Radio. I cloned this repo and followed the build instructions. The install failed with an "invalid environment marker" error shown at the bottom of the console listing below.  I have python 2.7.9 and 3.4.2 installed.  I am on Debian Jessie:
```
don@spearmint:~/github/pybombs$ python --version
Python 2.7.9
don@spearmint:~/github/pybombs$ python3 --version
Python 3.4.2
don@spearmint:~/github/pybombs$ uname -a
Linux spearmint 3.16.0-6-amd64 #1 SMP Debian 3.16.56-1+deb8u1 (2018-05-08) x86_64 GNU/Linux
```

Here is my console log.  Please let me know if there is any other diagnostic information I can provide.

```
don@spearmint:~/github$ git clone https://github.com/gnuradio/pybombs.git
Cloning into 'pybombs'...
remote: Counting objects: 5497, done.
remote: Total 5497 (delta 0), reused 0 (delta 0), pack-reused 5497
Receiving objects: 100% (5497/5497), 1.62 MiB | 356.00 KiB/s, done.
Resolving deltas: 100% (3800/3800), done.
Checking connectivity... done.
don@spearmint:~/github$ cd pybombs
don@spearmint:~/github/pybombs$ sudo python setup.py install
[sudo] password for don: 
running install
running bdist_egg
running egg_info
creating PyBOMBS.egg-info
writing requirements to PyBOMBS.egg-info/requires.txt
writing PyBOMBS.egg-info/PKG-INFO
writing top-level names to PyBOMBS.egg-info/top_level.txt
writing dependency_links to PyBOMBS.egg-info/dependency_links.txt
writing entry points to PyBOMBS.egg-info/entry_points.txt
writing manifest file 'PyBOMBS.egg-info/SOURCES.txt'
reading manifest file 'PyBOMBS.egg-info/SOURCES.txt'
writing manifest file 'PyBOMBS.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib.linux-x86_64-2.7
creating build/lib.linux-x86_64-2.7/pybombs
copying pybombs/gitcache_manager.py -> build/lib.linux-x86_64-2.7/pybombs
copying pybombs/fetcher.py -> build/lib.linux-x86_64-2.7/pybombs
copying pybombs/inventory.py -> build/lib.linux-x86_64-2.7/pybombs
copying pybombs/main.py -> build/lib.linux-x86_64-2.7/pybombs
copying pybombs/recipe.py -> build/lib.linux-x86_64-2.7/pybombs
copying pybombs/requirer.py -> build/lib.linux-x86_64-2.7/pybombs
copying pybombs/config_manager.py -> build/lib.linux-x86_64-2.7/pybombs
copying pybombs/install_manager.py -> build/lib.linux-x86_64-2.7/pybombs
copying pybombs/tree_node.py -> build/lib.linux-x86_64-2.7/pybombs
copying pybombs/pb_exception.py -> build/lib.linux-x86_64-2.7/pybombs
copying pybombs/__init__.py -> build/lib.linux-x86_64-2.7/pybombs
copying pybombs/pb_logging.py -> build/lib.linux-x86_64-2.7/pybombs
copying pybombs/config_file.py -> build/lib.linux-x86_64-2.7/pybombs
copying pybombs/package_manager.py -> build/lib.linux-x86_64-2.7/pybombs
copying pybombs/recipe_manager.py -> build/lib.linux-x86_64-2.7/pybombs
copying pybombs/dep_manager.py -> build/lib.linux-x86_64-2.7/pybombs
creating build/lib.linux-x86_64-2.7/pybombs/commands
copying pybombs/commands/base.py -> build/lib.linux-x86_64-2.7/pybombs/commands
copying pybombs/commands/rebuild.py -> build/lib.linux-x86_64-2.7/pybombs/commands
copying pybombs/commands/autoconfig.py -> build/lib.linux-x86_64-2.7/pybombs/commands
copying pybombs/commands/run.py -> build/lib.linux-x86_64-2.7/pybombs/commands
copying pybombs/commands/prefix.py -> build/lib.linux-x86_64-2.7/pybombs/commands
copying pybombs/commands/inv.py -> build/lib.linux-x86_64-2.7/pybombs/commands
copying pybombs/commands/deploy.py -> build/lib.linux-x86_64-2.7/pybombs/commands
copying pybombs/commands/fetch.py -> build/lib.linux-x86_64-2.7/pybombs/commands
copying pybombs/commands/lint.py -> build/lib.linux-x86_64-2.7/pybombs/commands
copying pybombs/commands/git.py -> build/lib.linux-x86_64-2.7/pybombs/commands
copying pybombs/commands/__init__.py -> build/lib.linux-x86_64-2.7/pybombs/commands
copying pybombs/commands/recipes.py -> build/lib.linux-x86_64-2.7/pybombs/commands
[... 225 lines cut ...]
copying PyBOMBS.egg-info/requires.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying PyBOMBS.egg-info/top_level.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
zip_safe flag not set; analyzing archive contents...
pybombs.config_manager: module references __file__
creating dist
creating 'dist/PyBOMBS-2.3.3a0-py2.7.egg' and adding 'build/bdist.linux-x86_64/egg' to it
removing 'build/bdist.linux-x86_64/egg' (and everything under it)
Processing PyBOMBS-2.3.3a0-py2.7.egg
creating /usr/local/lib/python2.7/dist-packages/PyBOMBS-2.3.3a0-py2.7.egg
Extracting PyBOMBS-2.3.3a0-py2.7.egg to /usr/local/lib/python2.7/dist-packages
Adding PyBOMBS 2.3.3a0 to easy-install.pth file
Installing pybombs script to /usr/local/bin

Installed /usr/local/lib/python2.7/dist-packages/PyBOMBS-2.3.3a0-py2.7.egg
Processing dependencies for PyBOMBS==2.3.3a0
Searching for ruamel.yaml
Reading https://pypi.python.org/simple/ruamel.yaml/
Best match: ruamel.yaml 0.15.37
Downloading https://files.pythonhosted.org/packages/c1/ad/9557a78f1b61951623c56225e05dacdd81f80519f13fc39a926a5793edfa/ruamel.yaml-0.15.37.tar.gz#sha256=509842d96fb194f79b57483b76429f8956d8f7ade3cb49d1e5aeb5c5e9ef4918
Processing ruamel.yaml-0.15.37.tar.gz
Writing /tmp/easy_install-geulbw/ruamel.yaml-0.15.37/setup.cfg
Running ruamel.yaml-0.15.37/setup.py -q bdist_egg --dist-dir /tmp/easy_install-geulbw/ruamel.yaml-0.15.37/egg-dist-tmp-wuK8nn
sys.argv ['/tmp/easy_install-geulbw/ruamel.yaml-0.15.37/setup.py', '-q', 'bdist_egg', '--dist-dir', '/tmp/easy_install-geulbw/ruamel.yaml-0.15.37/egg-dist-tmp-wuK8nn']
error: Setup script exited with error in ruamel.yaml setup command: Invalid environment marker: platform_python_implementation=="CPython" and python_version<="2.7"
```

mbr0wn, 2018-06-14:
Not sure what's going on, but you might be able to workaround by installing ruamel.yaml through apt-get.

mbr0wn, 2018-06-14:
Ah, ruamel.yaml is not packaged for Jessie. Hm.

mbr0wn, 2018-06-14:
OK, I could reproduce. A workaround is to

    $ apt install python-dev
    $ pip install ruamel.yaml

Now, really, setup.py should do the exact same thing, so I'm not sure why the workaround works, but regular install does not. I'd appreciate some help here.

mbr0wn, 2018-06-14:
339c21c6f3c232521dab21bf29c92437f55cf31d

cosinekitty, 2018-06-15:
Thanks for providing a workaround, @mbr0wn!  I will give that a try this weekend.

cosinekitty, 2018-06-18:
Well, this is kinda dumb. Since I opened this issue, I forgot about it and went and upgraded to Debian Stretch 9.4.0. I suppose someone else still using Jessie might come here and find this useful though.

mbr0wn, 2018-06-18:
Did it work on Stretch?

cosinekitty, 2018-06-19:
I used the workaround before I thought about it, unfortunately. The pip install worked fine. Probably setup.py would have worked too, now that I’m on Stretch, but I didn’t think to try it. 

Don

deinarson, 2018-08-06:
I am new to pybombs, but for what it its worth, this seems to work.  At least with the docker's debian:jessie image.

If you are not using docker you can just manual running the commands removing the word "RUN"
[Dockerfile gist](https://gist.github.com/deinarson/9733841c48b411d8a32db5c216cdea1d)

So you might be able to close this. :)

argilo, 2020-10-03:
I'll close this off, since Debian Jessie is obsolete and no longer supported by PyBOMBS.

PyBOMBS appears to work well with Debian Buster.

closed 2020-10-03 by argilo