gnuradio/pybombs #500: Fails to install on Ubuntu 18.04

skandragon, 2018-06-03:
When I run:

```
pybombs prefix init -a default gnuradio/default/ -R gnuradio-default
```
I get this in the CMake error log:

```Run Build Command:"/usr/bin/make" "cmTC_83e8d/fast"
/usr/bin/make -f CMakeFiles/cmTC_83e8d.dir/build.make CMakeFiles/cmTC_83e8d.dir/build
make[1]: Entering directory '/home/explorer/gnuradio/default/src/uhd/host/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_83e8d.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_83e8d.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.10/Modules/CheckFunctionExists.c
Linking C executable cmTC_83e8d
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_83e8d.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_83e8d.dir/CheckFunctionExists.c.o  -o cmTC_83e8d -lpthreads 
/usr/bin/x86_64-linux-gnu-ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_83e8d.dir/build.make:97: recipe for target 'cmTC_83e8d' failed
make[1]: *** [cmTC_83e8d] Error 1
make[1]: Leaving directory '/home/explorer/gnuradio/default/src/uhd/host/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_83e8d/fast' failed
make: *** [cmTC_83e8d/fast] Error 2
```

marcusmueller, 2018-06-09:
This is a problem during a build of UHD, not GNU Radio. If I had to take a guess, this looks like either your system does something very strange, or Ubuntu has yet again decided to rename core library packages without appropriate depends.

In either case, it's really unrelated to GNU Radio, and if anything, might be something for the gr-recipe bug tracker (if it's really just a Ubuntu renaming issue).

Might be an issue to raise with the Ettus/USRP community on usrp-users@lists.ettus.com (sign up first). It's pretty responsive.

marcusmueller, 2018-06-09:
ah, my mistake, this is of course OK to keep open here – this is, after all, the Pybombs bug tracker. My bad.

mbr0wn, 2018-06-14:
I'm currently running the Docker-based test for 18.04 in PyBOMBS, and it worked fine (I only just added that). You can confirm by running

    $ ./run-tests.sh --skip-pylint --container=ubuntu_1804

Closing for now -- please reopen if you have other concerns. It's important that we can build on 18.04.

closed 2018-06-14 by mbr0wn