gnuradio/pybombs #502: pybombs prefix init ~/prefix -a myprefix -R gnuradio-default ... fails on Ubuntu 16.04

curtcorum, 2018-06-08:
Dear pybombs team and community,

Had issues with building gnuradio-default on relatively clean but up to date Ubuntu 16.04 LTS. Seems related to https://www.mail-archive.com/discuss-gnuradio@gnu.org/msg66739.html #424 #448

$ pybombs --version
2.3.2

$ gcc --version
gcc (Ubuntu 6.4.0-17ubuntu1~16.04) 6.4.0 20180424

It was necessary to do the following to get the gnuradio-default recipe to build thrift, boost, uhd and gnuradio without errors:

sudo apt-get install g++-6
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-6 60 --slave /usr/bin/g++ g++ /usr/bin/g++-6
sudo update-alternatives --config gcc
      and setting gcc-6 to default

uninstall most of boost 1.58 binary packages (all but non -dev distribution packages) 
adding the following to gnuradio-default.lwr

    # binary boost linking fails *** CAC
    boost:
      forcebuild: True

and modifying config.yml (directly as could not figure out how to escape -std=c++-11 for pybombs config --env)

env:
  CXXFLAGS: -std=c++11 -fext-numeric-literals

and modifying boost.lwr

satisfy:
  deb: libboost-all-dev >= 1.6 || libboost-dev >= 1.6
  rpm: (boost-devel >= 1.6) || (boost_1_61-devel >= 1.6) || (boost_1_64-devel >= 1.6) || (boost_1_65-devel >= 1.6)
  pacman: boost >= 1.6
  port: boost >= 1.6
  portage: dev-libs/boost >= 1.6

Wanted to make sure this got captured as it took several days to resolve and seems to be an ongoing issue.

-Curt

curtcorum, 2018-06-08:
curt@curt--laptop:~$ pybombs inv
PyBOMBS - INFO - PyBOMBS Version 2.3.2
Showing package state:
uhd:                installed
apache-thrift:      installed
boost:              installed
gnuradio:           installed

<<< Welcome to GNU Radio Companion 3.7.12.0 >>>

Block paths:
	/home/curt/.grc_gnuradio
	/home/curt/pybombs_180605/share/gnuradio/grc/blocks
	/home/curt/pybombs_180605/share/gnuradio/grc/blocks

Loading: "/home/curt/src/gnuradio_curt/usrp_rx.grc"

mirmlodro, 2018-06-08:
Hi Curt, 
I got the following error after i tried to install gnuradio using gcc 6. 

> > root@pezp63763:/home/eexmmlo# pybombs --version
> > 2.3.2
> > root@pezp63763:/home/eexmmlo# gcc --version
> > gcc (Ubuntu 6.4.0-17ubuntu1~16.04) 6.4.0 20180424
> > 
> > PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
> > Install tree:
> > |
> > \- gnuradio
> >    |
> >    \- uhd
> > PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
> > PyBOMBS.install_manager - INFO - Installing package: uhd
> > PyBOMBS.PackageManager - WARNING - A source build for package uhd was requested, but binary install was found!
> > Install uhd from source despite binary install available Y/[N]? y
> > Cloning:     (100%) [====================================================]
> > Configuring: (100%) [======================================================================================================================================================================================]
> > Building:    (100%) [======================================================================================================================================================================================]
> > [  2%] Built target uhd_rpclib
> > [ 62%] Built target uhd
> > [ 62%] Linking CXX executable twinrx_freq_hopping
> > ../lib/libuhd.so.3.12: undefined reference to `boost::re_detail::cpp_regex_traits_implementation<char>::transform_primary[abi:cxx11](char const*, char const*) const'
> > ../lib/libuhd.so.3.12: undefined reference to `boost::re_detail::cpp_regex_traits_implementation<char>::transform[abi:cxx11](char const*, char const*) const'
> > collect2: error: ld returned 1 exit status
> > examples/CMakeFiles/twinrx_freq_hopping.dir/build.make:109: recipe for target 'examples/twinrx_freq_hopping' failed
> > make[2]: *** [examples/twinrx_freq_hopping] Error 1
> > CMakeFiles/Makefile2:493: recipe for target 'examples/CMakeFiles/twinrx_freq_hopping.dir/all' failed
> > make[1]: *** [examples/CMakeFiles/twinrx_freq_hopping.dir/all] Error 2
> > Makefile:160: recipe for target 'all' failed
> > make: *** [all] Error 2
> > PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
> > PyBOMBS.Packager.source - ERROR - Problem occurred while building package uhd:
> > Build failed.
> > PyBOMBS.install_manager - ERROR - Error installing package uhd. Aborting.
> > ```
> > 
> > 
```

```

curtcorum, 2018-06-09:
> Hi Curt,
>
> Could you please explain to me how to uninstall most of boost 1.58
> binary packages?
> And also where to find gnuradio-default.lwr / boost.lwr, in my prefix
> installation?
>
> My pybmobs and gcc version:
> $ pybombs --version
> 2.3.2
> $ gcc --version
> gcc (Ubuntu 6.4.0-17ubuntu1~16.04) 6.4.0 20180424
>
> I still have errors during gnuradio building throught pybombs.
> Hope you will help.

I used synaptic to remove extra boost 1.58 binaries. Not sure if this
was strictly necessary.

The recipes are in your ~/.pybombs/gr-recipes/ folder.

hope it helps,

Curt

curtcorum, 2018-06-09:
>         to|boost::re_detail::cpp_regex_traits_implementation::transform[abi:cxx11](char

I removed all the binaries for uhd as well as most of boost before building.

You can do a search for boost and uhd in synaptic to see what is
installed on your system.

curtcorum, 2018-06-09:
Looks like part of this issue is being addressed in:
#https://github.com/gnuradio/gnuradio/pull/1810

marcusmueller, 2018-06-09:
I'm closing this since we merged #1810. (Leaving this discussion, open, however. I think if there's something else blocking building on clean ubuntu 16.04, that'd be worth having a different issue report about)

closed 2018-06-09 by marcusmueller

curtcorum, 2018-06-11:
Here is what may be needed for gcc6:

https://askubuntu.com/questions/746369/how-can-i-install-and-use-gcc-6-on-xenial

mirmlodro, 2018-06-13:
> xenial

Hi curt,
i am again getting this output. Can you please tell how to delete
unnecessary lib and uhd files after searching in synaptic manager. I
perform search in folder but I can't find the file, whereas the synaptic
manage shows some uhd files.
eexmmlo@pezp63763:~/prefix$ pybombs prefix init ~/prefix -a myprefix -R
gnuradio-default
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.prefix - WARNING - There already is a prefix in
`/home/eexmmlo/prefix'.
Continue using this path Y/[N]? y
Alias `myprefix' already exists, overwrite Y/[N]? y
PyBOMBS.ConfigManager - INFO - Creating new config file
/home/eexmmlo/prefix/.pybombs/config.yml
PyBOMBS.prefix - INFO - Installing default packages for prefix...
PyBOMBS.prefix - INFO -
  - gnuradio
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and
installing binary packages:
Install tree:
|
\- gnuradio
   |
   \- uhd
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source
packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: uhd
PyBOMBS.Packager.source - WARNING - Build dir already exists:
/home/eexmmlo/prefix/src/uhd/host/build
Building:    (100%)
[============================================================================================================]
[  2%] Built target uhd_rpclib
[ 62%] Built target uhd
[ 62%] Linking CXX executable twinrx_freq_hopping
../lib/libuhd.so.3.12: undefined reference to
`boost::re_detail::cpp_regex_traits_implementation<char>::transform_primary[abi:cxx11](char
const*, char const*) const'
../lib/libuhd.so.3.12: undefined reference to
`boost::re_detail::cpp_regex_traits_implementation<char>::transform[abi:cxx11](char
const*, char const*) const'
clang: error: linker command failed with exit code 1 (use -v to see
invocation)
examples/CMakeFiles/twinrx_freq_hopping.dir/build.make:109: recipe for
target 'examples/twinrx_freq_hopping' failed
make[2]: *** [examples/twinrx_freq_hopping] Error 1
CMakeFiles/Makefile2:493: recipe for target
'examples/CMakeFiles/twinrx_freq_hopping.dir/all' failed
make[1]: *** [examples/CMakeFiles/twinrx_freq_hopping.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error
messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package
uhd:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package uhd. Aborting.
Mir Lodro