gnuradio/pybombs #503: Linking error on test-gr-blocks step on Arch

txjacob, 2018-06-10:
I'm getting a linking issue when trying to build on Arch, GCC 8.1.1:
```[ 27%] Linking CXX executable test-gr-blocks
/usr/bin/ld: CMakeFiles/test-gr-blocks.dir/qa_set_msg_handler.cc.o: undefined reference to symbol 'pthread_condattr_setclock@@GLIBC_2.3.3'
/usr/lib/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [gr-blocks/lib/CMakeFiles/test-gr-blocks.dir/build.make:242: gr-blocks/lib/test-gr-blocks] Error 1
make[1]: *** [CMakeFiles/Makefile2:2094: gr-blocks/lib/CMakeFiles/test-gr-blocks.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
PyBOMBS.monitor_process() - DEBUG - Thread signaled termination or returned
PyBOMBS.monitor_process() - DEBUG - Return value: 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.
```

mbr0wn, 2018-06-14:
@txjacob Can you help out with this -- a Docker-based test file would be a useful place to start, to create a reproducible failure.

txjacob, 2018-06-17:
Additing -lpthread to the linker list file for test-gr-blocks fixed the problem.

mbr0wn, 2018-06-17:
In that case, it's a GNU Radio issue. Thanks all, for looking into this! I'm not sure if this has already come up on discuss-gnuradio, but that's probably where it should go.

closed 2018-06-17 by mbr0wn