gnuradio/pybombs #505: grc global_blocks_path and pybombs

josephernst, 2018-06-18:
When using multiple prefixes with pybombs and grc, I have to manually edit the global_blocks_path in ~/.gnuradio/config.conf when switching between prefixes.  Otherwise, I have the wrong blocks in grc.

To replicate:

$ pybombs prefix init ~/prefix -a prefix -R gnuradio-default
$ pybombs prefix init ~/prefix2 -a prefix2 -R gnuradio-default
$ pybombs -p prefix2 install OUT_OF_TREE MODULE
$ source ~/prefix2/setup_env.sh
$ gnuradio-companion

Now the gnuradio-companion from prefix2 will load with the blocks list from prefix1

mbr0wn, 2018-06-18:
Wow, that is odd. How would it even know about prefix1? Bizarre.

mbr0wn, 2018-07-03:
OK, I see.  You shouldn't have to edit your `~/.gnuradio/config.conf` for the sake of accommodating PyBOMBS prefixes. If the path is unset, gnuradio-companion will usually find the blocks just fine by itself. @namccart changed this in 9c0e27ffa66e5c5f4dc9ee544de40885b444c7b7.

Maybe removing the line from `config.conf` is all it takes. I have a bunch of prefixes here and often switch back and forth (or even open multiple companions from different prefixes at the same time). After launching GRC from a prefix, this is the first thing it prints in the log window:
```
<<< Welcome to GNU Radio Companion 3.7.12git-351-ga098720f >>>

Block paths:
	/home/mbr0wn/.grc_gnuradio
	/home/mbr0wn/prefix/radar/share/gnuradio/grc/blocks
```
Notice how it picked up only blocks from my `radar` prefix (which is from where I launched it). Other prefixes have other paths, but they don't need encoding.

There is an actual environment variable (`GRC_BLOCKS_DIR`) which could help if you're not getting anywhere. Try running

    $ pybombs -p prefix2 run gnuradio-companion

...and see if that worked. The `run` command injects its own environment, which includes `GRC_BLOCKS_PATH` (although I'm not convinced that is necessary).