gnuradio/pybombs #510: Recipe 'gr-ieee-802154' missong

MountainLogic, 2018-07-27:
As I'm not sure who own this, I've also cross posted at https://github.com/bastibl/gr-ieee802-15-4/issues/37  

With the clean Ubuntu 18.04 LTS install and following the pybombs instructions at https://www.gnuradio.org/blog/pybombs-the-what-the-how-and-the-why,
then I run pybombs (version 2.3.2) does not include the recipe list.  gr-foo and gr-ieee-80211 are both present in the list.

I've tried

`pybombs install gr-ieee-802154
`
and pybombs then reports

`ERROR - Error fetch recipe 'gr-ieee-802154':
Package gr-ieee-802154 has no recipe file!
`

I notice that there is an old pybombs issue, #27 ,  related to 'gr-ieee-802154'

MountainLogic, 2018-07-27:
In my .pybombs dir I was able to clone off gr-ieee-80211.lwr and make the obvious changes to create my own gr-ieee0802154.lwr files with success.  So, I assume that '15.4 just got left out of the recipes list???

mbr0wn, 2018-08-17:
It's here: https://github.com/gnuradio/gr-etcetera/blob/master/gr-ieee-802154.lwr

closed 2018-08-17 by mbr0wn