gnuradio/pybombs #511: pygtk >= 2.10.0 - not found

rytse, 2018-08-06:
When I install using `pybombs prefix init ~/prefix -a myprefix -R gnuradio-default`, pybombs fails to find pygtk.

I have tried installing pygtk through conda using the following conda packages:
` conda install -c jochym pygtk `
` conda install -c ska pygtk `
`conda install -c ostrokach gtk`

Each time after installing through conda, `python -c "import pygtk"` works, but the pybombs installer says `Python checking for pygtk >= 2.10.0 - not found`.

I am on Ubuntu 16.04, Python 2.7.15 from Anaconda, and trying to build pybombs from commit f8a198a3aa8abb4c68aba3158c2c6f82013d05db.

Here's my full pybombs installation output:

```
(siglab) rytse@SEDRSSHYPRES:~/usrlibs/pybombs$ pybombs prefix init ~/prefix -a myprefix -R gnuradio-default
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.15
PyBOMBS - INFO - PyBOMBS Version 2.3.3a0
PyBOMBS.prefix - WARNING - There already is a prefix in `/home/rytse/prefix'.
Continue using this path Y/[N]? y
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.15
Alias `myprefix' already exists, overwrite Y/[N]? y
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.15
PyBOMBS.prefix - INFO - Installing default packages for prefix...
PyBOMBS.prefix - INFO - 
  - gnuradio
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gnuradio
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: gnuradio
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.Packager.source - WARNING - Build dir already exists: /home/rytse/prefix/src/gnuradio/build
Configuring: (100%) [==================================================================================================================]
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
-- Build type set to RelWithDebInfo.
-- Extracting version information from git describe...
-- Compiler Version: cc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O2 -g -DNDEBUG  -std=gnu99 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- Building Static Libraries: OFF
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   unit_test_framework
--   chrono
--   atomic
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable version "2.7.15", minimum required is "2.7") 
-- 
-- Checking for module SWIG
-- Found SWIG version 2.0.12.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Enabling python-support support.
[... 117 lines cut ...]
-- 
-- Python checking for numpy
-- Python checking for numpy - found
-- 
-- Configuring gnuradio-companion support...
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_PYTHON = ON
--   Dependency PYTHON_MIN_VER_FOUND = TRUE
--   Dependency CHEETAH_FOUND = TRUE
--   Dependency LXML_FOUND = TRUE
--   Dependency PYGTK_FOUND = FALSE
--   Dependency NUMPY_FOUND = TRUE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled gnuradio-companion but configuration checked failed
Call Stack (most recent call first):
  grc/CMakeLists.txt:45 (GR_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/home/rytse/prefix/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/home/rytse/prefix/src/gnuradio/build/CMakeFiles/CMakeError.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.
```

rfenouil, 2018-08-16:
Same problem here,

while trying to install gnuradio from source, I never managed to get pygtk recognized or installed on my computer (Ubuntu 14.04.5 LTS). 

I tried pip that told me:
```* Building PyGTK using distutils is only supported on windows. *
    * To build PyGTK in a supported way, read the INSTALL file.    *```

Tried using apt-get, no luck.
Tried to build from source using ```jhbuild``` which is a nightmare on itself (never worked).

Finally I ran into PyGObject documentation (and forums) which tells that pygtk is depreccated (?).

I also  tried to build gnuradio using pybombs but once I the build start it fails on the same error as OP's...

Any clue ?

rytse, 2018-08-16:
I have personally given up on running PyBOMBS at all, much less through conda, as it is so unstable. However, the apt package worked fine for me after installing a few other dependencies through apt (though I am on Ubuntu 16.04).

What is the output of your apt install?

Try

```
sudo apt-get install -y \
    cmake \
    autoconf \
    libtool \
    pkg-config \
    build-essential \
    python-docutils \
    libcppunit-dev \
    swig \
    doxygen \
    liblog4cpp5-dev \
    python-scipy \
    python-gtk2 \
    gnuradio-dev
```

rfenouil, 2018-08-16:
Hello @rytse 

thank you for your message, apt-get works for gnuradio (and gnuradio-dev) but only allows me to get an 'older' version.
I need a more recent one installed in order to install a fork of gr-osmosdr... Hence my attempts for building it, but no success so far.

When bhuilding gr-odsmosdr:
```
CMake Error at CMakeLists.txt:156 (MESSAGE):
  GnuRadio version required: >="3.7.8" found: "3.7.2.1"
```

Maybe it's time to upgrade my old ubuntu to a fresh 18.04 and simply get a newer version of gnuradio with apt-get...

rytse, 2018-08-16:
In that case, try `sudo apt install libgtk2.0-dev libgtk2.0-0` to resolve your PyGTK issue. [Here's](https://stackoverflow.com/questions/51731082/pygtk-issue-when-installing-gnuradio/51735438?noredirect=1#comment90542456_51735438) a related StackOverflow post. I never did get get it working in a virtualenv or a conda env, but maybe it will work in a global install.

rfenouil, 2018-08-16:
Thank you for the suggestion but 

```
libgtk2.0-0 is already the newest version.
libgtk2.0-dev is already the newest version.
```

no luck...

rytse, 2018-08-16:
It looks like you have gtk but not [gobject](https://pygobject.readthedocs.io/en/latest/getting_started.html). Once you have that, try following bohrax's instructions for adding symbolic symlinks to gtk, pygobject, and pygtk so that non-root python can see those packages.

rytse, 2018-08-16:
Did you create the simlinks that bohrax advises?

rfenouil, 2018-08-16:
I did, but in the wrong place... I did not realize that my user python depends on conda lib path...
Just tried again in conda lib folder... cleaned all my pybombs mess and started again... In progress.

rfenouil, 2018-08-16:
Was looking good... Until:

```
--   Dependency LXML_FOUND = TRUE
--   Dependency PYGTK_FOUND = FALSE
--   Dependency NUMPY_FOUND = TRUE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled gnuradio-companion but configuration checked failed
Call Stack (most recent call first):
  grc/CMakeLists.txt:45 (GR_REGISTER_COMPONENT)
...
```

rfenouil, 2018-08-16:
Which is weird because now the following works:

```
$ python
Python 2.7.14 |Anaconda, Inc.| (default, Dec  7 2017, 17:05:42) 
[GCC 7.2.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import pygtk
>>> 
```

rytse, 2018-08-16:
`import cairo`?

rfenouil, 2018-08-16:
after symlinking cairo:

```
$ python
Python 2.7.14 |Anaconda, Inc.| (default, Dec  7 2017, 17:05:42) 
[GCC 7.2.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import gtk
{myhomefolder}/miniconda2/lib/python2.7/gtk/__init__.py:40: RuntimeWarning: tp_compare didn't return -1 or -2 for exception
  from gtk import _gtk
ImportError: could not import pango
ImportError: could not import pango
GLib-GIO-Message: 16:24:42.776: Using the 'memory' GSettings backend.  Your settings will not be saved or shared with other applications.
>>> 
```

rfenouil, 2018-08-16:
```import cairo``` raised an error before I added the symlink.
Now loads ok.

Started another run of pybombs... Will let you know a bit later I have to go.
Thank you for your support btw !

rytse, 2018-08-16:
You're welcome! I would like to be able to run GNURadio through conda as well, so I'm happy to help.
If you figure out how to do it, please do post how you did it here so that I and others can recreate it :smiley:

mbr0wn, 2018-08-17:
I had issues installing GNU Radio (unrelated to PyBOMBS though) with respect to PyGTK (see also this discussion: https://github.com/gnuradio/gnuradio/issues/1859). I was missing the `gir` packages. This is our CI builder: https://github.com/gnuradio/gnuradio-buildbot/blob/public/worker/ubuntu-18.04.Dockerfile

rfenouil, 2018-08-17:
Just an update, after solving disk space problems and overheating crash during compilation, I managed to get through the full installation process. I have also been able to compile gr-osmosdr on top of that.

However gnuradio-companion crashes on startup with the following message:
$ ./gnuradio-companion 
{myhomefolder}/miniconda2/lib/python2.7/gtk/__init__.py:40: RuntimeWarning: tp_compare didn't return -1 or -2 for exception
  from gtk import _gtk
ImportError: could not import pango
ImportError: could not import pango
GLib-GIO-Message: 15:57:44.957: Using the 'memory' GSettings backend.  Your settings will not be saved or shared with other applications.
Traceback (most recent call last):
  File "./gnuradio-companion", line 96, in <module>
    check_gnuradio_import()
  File "./gnuradio-companion", line 72, in check_gnuradio_import
    die(err, GR_IMPORT_ERROR_MESSAGE)
  File "./gnuradio-companion", line 43, in die
    d = gtk.MessageDialog(
AttributeError: 'module' object has no attribute 'MessageDialog'

I guess pygtk is not happy even though it has been detected correctly during compilation.

rfenouil, 2018-08-20:
Hello, I finally took the easy path and reinstalled a newer ubuntu on my laptop...
Took me 2 hours in total, including configuration of gnuradio (apt-get) and compilation of gr-osmosdr ;)

Sorry, I know that does not really help people reading this thread, but all this took too much of my time already.
Thank you for your great support !

gmagno, 2019-07-20:
My experience with pybombs hasn't been great either I have to say.
Anyway, in case it helps someone, try and build both `py2cairo` and `pygtk` manually. Follow the instructions here: https://stackoverflow.com/a/17126095/1278365
The installation is done in a virtualenv which is quite convenient.