gnuradio/pybombs #512: AssertionError: assert isinstance(self.data, dict) on Debian/Ubuntu/Kali

velichkov, 2018-08-10:
Hi @mbr0wn,

On Debian testing, Ubuntu 18.04, Kali rolling  and Fedora 28,  using the latest version of pybombs from the pip repository, the `pybombs recipes add-defaults` fails with

```
$ pybombs recipes add-defaults
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
Traceback (most recent call last):
  File "/usr/local/bin/pybombs", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/pybombs/main.py", line 32, in main
    return dispatch() or 0
  File "/usr/local/lib/python2.7/dist-packages/pybombs/commands/base.py", line 194, in dispatch
    return get_cmd_dict(cmd_list)[args.command](cmd=args.command, args=args).run()
  File "/usr/local/lib/python2.7/dist-packages/pybombs/commands/base.py", line 120, in run
    return self.subcommands[self.args.sub_command]['run'](self)()
  File "/usr/local/lib/python2.7/dist-packages/pybombs/commands/recipes.py", line 180, in run_add_defaults
    if not self.add_recipe_dir(alias, uri):
  File "/usr/local/lib/python2.7/dist-packages/pybombs/commands/recipes.py", line 347, in add_recipe_dir
    Fetcher().fetch_url(uri, recipe_cache_top_level, alias, {}) # No args
  File "/usr/local/lib/python2.7/dist-packages/pybombs/fetcher.py", line 72, in fetch_url
    fetcher.assert_requirements()
  File "/usr/local/lib/python2.7/dist-packages/pybombs/requirer.py", line 80, in assert_requirements
    require_hostsys_dependencies(requirements)
  File "/usr/local/lib/python2.7/dist-packages/pybombs/requirer.py", line 50, in require_hostsys_dependencies
    install_type="binary", # Requirers may not request source packages
  File "/usr/local/lib/python2.7/dist-packages/pybombs/install_manager.py", line 112, in install
    lambda pkg: _cached_check_if_pkg_goes_into_tree(pkg, _check_if_pkg_goes_into_tree)
  File "/usr/local/lib/python2.7/dist-packages/pybombs/dep_manager.py", line 53, in make_dep_tree
    pkg_set = set([pkg for pkg in pkg_list if filter_callback(pkg)])
  File "/usr/local/lib/python2.7/dist-packages/pybombs/install_manager.py", line 112, in <lambda>
    lambda pkg: _cached_check_if_pkg_goes_into_tree(pkg, _check_if_pkg_goes_into_tree)
  File "/usr/local/lib/python2.7/dist-packages/pybombs/install_manager.py", line 97, in _cached_check_if_pkg_goes_into_tree
    ret_val = check_callback(pkg)
  File "/usr/local/lib/python2.7/dist-packages/pybombs/install_manager.py", line 63, in _check_if_pkg_goes_into_tree
    if not self.pm.exists(pkg):
  File "/usr/local/lib/python2.7/dist-packages/pybombs/package_manager.py", line 128, in exists
    if self.check_package_flag(name, 'forceinstalled'):
  File "/usr/local/lib/python2.7/dist-packages/pybombs/package_manager.py", line 89, in check_package_flag
    recipe.get_recipe(pkgname).category
  File "/usr/local/lib/python2.7/dist-packages/pybombs/recipe.py", line 383, in get_recipe
    r = Recipe(recipe_manager.recipe_manager.get_recipe_filename(pkgname))
  File "/usr/local/lib/python2.7/dist-packages/pybombs/recipe.py", line 219, in __init__
    self._data = self._load_recipe_from_file(filename)
  File "/usr/local/lib/python2.7/dist-packages/pybombs/recipe.py", line 256, in _load_recipe_from_file
    data = PBConfigFile(filename).get()
  File "/usr/local/lib/python2.7/dist-packages/pybombs/config_file.py", line 84, in __init__
    assert isinstance(self.data, dict)
AssertionError
```
The issue seems to be fixed in the latest master in d27fa9dd4cba809b99d65362cb02e6313cf37ddb
```
$ pip install git+git://github.com/gnuradio/pybombs.git
Collecting git+git://github.com/gnuradio/pybombs.git
  Cloning git://github.com/gnuradio/pybombs.git to /tmp/pip-sSJVR2-build
Requirement already satisfied: future in /usr/local/lib/python2.7/dist-packages (from PyBOMBS==2.3.3a0)
Requirement already satisfied: requests in /usr/local/lib/python2.7/dist-packages (from PyBOMBS==2.3.3a0)
Requirement already satisfied: ruamel.yaml in /usr/local/lib/python2.7/dist-packages (from PyBOMBS==2.3.3a0)
Requirement already satisfied: setuptools in /usr/lib/python2.7/dist-packages (from PyBOMBS==2.3.3a0)
Requirement already satisfied: six in /usr/lib/python2.7/dist-packages (from PyBOMBS==2.3.3a0)
Requirement already satisfied: idna<2.8,>=2.5 in /usr/lib/python2.7/dist-packages (from requests->PyBOMBS==2.3.3a0)
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in /usr/local/lib/python2.7/dist-packages (from requests->PyBOMBS==2.3.3a0)
Requirement already satisfied: urllib3<1.24,>=1.21.1 in /usr/local/lib/python2.7/dist-packages (from requests->PyBOMBS==2.3.3a0)
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python2.7/dist-packages (from requests->PyBOMBS==2.3.3a0)
Requirement already satisfied: ruamel.ordereddict; platform_python_implementation == "CPython" and python_version <= "2.7" in /usr/local/lib/python2.7/dist-packages (from ruamel.yaml->PyBOMBS==2.3.3a0)
Installing collected packages: PyBOMBS
  Found existing installation: PyBOMBS 2.3.2
    Uninstalling PyBOMBS-2.3.2:
      Successfully uninstalled PyBOMBS-2.3.2
  Running setup.py install for PyBOMBS ... done
Successfully installed PyBOMBS-2.3.3a0
root@f8347c011f94:/# pybombs recipes add-defaults
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.15
PyBOMBS - INFO - PyBOMBS Version 2.3.3a0
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.GitCacheManager - INFO - Creating new git cache in /root/.pybombs/gitcache
remote: Counting objects: 321, done.
remote: Total 321 (delta 0), reused 0 (delta 0), pack-reused 321
Receiving objects: 100% (321/321), 56.80 KiB | 223.00 KiB/s, done.
Resolving deltas: 100% (218/218), done.
From https://github.com/gnuradio/gr-etcetera
 * [new branch]      master     -> gr-etcetera/master
Cloning into 'gr-etcetera'...
warning: no common commits
remote: Counting objects: 1599, done.
remote: Total 1599 (delta 0), reused 0 (delta 0), pack-reused 1599
Receiving objects: 100% (1599/1599), 308.46 KiB | 789.00 KiB/s, done.
Resolving deltas: 100% (1331/1331), done.
From https://github.com/gnuradio/gr-recipes
 * [new branch]      homebrew   -> gr-recipes/homebrew
 * [new branch]      master     -> gr-recipes/master
Cloning into 'gr-recipes'...
```
Is it possible to release a new version as in a lot of documents and tutorials the default installation method is `[sudo] pip install pybombs`

mbr0wn, 2018-08-16:
#518 should take care of this. And yeah, it's time for a new version.

closed 2018-08-16 by mbr0wn

gareth8118, 2018-09-15:
Still broken on Ubuntu 18.04 using the PyPI version of pybombs.