gnuradio/pybombs #513: AttributeError: 'Recipe' object has no attribute 'target

velichkov, 2018-08-10:
Hi,

I'm trying to install the gnuradio using the latest master branch using the following docker file.
```
FROM ubuntu:18.04

RUN apt-get update && DEBIAN_FRONTEND=noninteractive apt-get install -y \
	git \
	python-pip \
	sudo \
	tzdata

RUN pip install git+https://github.com/gnuradio/pybombs.git

RUN pybombs auto-config
RUN pybombs recipes add-defaults
RUN pybombs -vv -y prefix init ~/prefix -a myprefix -R gnuradio-default
```
and it fails with 
```
PyBOMBS.install_manager - OBNOXIOUS - Checking if package `wxpython' goes into tree...
PyBOMBS.PackageManager - DEBUG - Checking if package wxpython is installable...
PyBOMBS.Recipe[wxpython] - OBNOXIOUS - Loading recipe file: /root/.pybombs/recipes/gr-recipes/wxpython.lwr
PyBOMBS.Recipe[wxpython] - OBNOXIOUS - Loading template file: /usr/local/lib/python2.7/dist-packages/pybombs/templates/autoconf.lwt
PyBOMBS.Recipe[wxpython] - OBNOXIOUS - Inheriting from file /usr/local/lib/python2.7/dist-packages/pybombs/templates/autoconf.lwt
PyBOMBS.Recipe[wxpython] - OBNOXIOUS - Loading template file: /usr/local/lib/python2.7/dist-packages/pybombs/templates/empty.lwt
PyBOMBS.Recipe[wxpython] - OBNOXIOUS - Inheriting from file /usr/local/lib/python2.7/dist-packages/pybombs/templates/empty.lwt
PyBOMBS.Recipe[wxpython] - OBNOXIOUS - Loaded recipe - wxpython
Traceback (most recent call last):
  File "/usr/local/bin/pybombs", line 11, in <module>
    load_entry_point('PyBOMBS==2.3.3a0', 'console_scripts', 'pybombs')()
  File "/usr/local/lib/python2.7/dist-packages/pybombs/main.py", line 32, in main
    return dispatch() or 0
  File "/usr/local/lib/python2.7/dist-packages/pybombs/commands/base.py", line 194, in dispatch
    return get_cmd_dict(cmd_list)[args.command](cmd=args.command, args=args).run()
  File "/usr/local/lib/python2.7/dist-packages/pybombs/commands/base.py", line 120, in run
    return self.subcommands[self.args.sub_command]['run'](self)()
  File "/usr/local/lib/python2.7/dist-packages/pybombs/commands/prefix.py", line 183, in run_init
    self.args.virtualenv
  File "/usr/local/lib/python2.7/dist-packages/pybombs/commands/prefix.py", line 337, in _init_prefix
    return install_dependencies(prefix_recipe.depends)
  File "/usr/local/lib/python2.7/dist-packages/pybombs/commands/prefix.py", line 318, in install_dependencies
    print_tree=True,
  File "/usr/local/lib/python2.7/dist-packages/pybombs/install_manager.py", line 112, in install
    lambda pkg: _cached_check_if_pkg_goes_into_tree(pkg, _check_if_pkg_goes_into_tree)
  File "/usr/local/lib/python2.7/dist-packages/pybombs/dep_manager.py", line 57, in make_dep_tree
    dep_trees[pkg] = self.make_tree_recursive(pkg, filter_callback)
  File "/usr/local/lib/python2.7/dist-packages/pybombs/dep_manager.py", line 77, in make_tree_recursive
    deps_to_install = set([dep for dep in all_deps if filter_callback(dep)])
  File "/usr/local/lib/python2.7/dist-packages/pybombs/install_manager.py", line 112, in <lambda>
    lambda pkg: _cached_check_if_pkg_goes_into_tree(pkg, _check_if_pkg_goes_into_tree)
  File "/usr/local/lib/python2.7/dist-packages/pybombs/install_manager.py", line 97, in _cached_check_if_pkg_goes_into_tree
    ret_val = check_callback(pkg)
  File "/usr/local/lib/python2.7/dist-packages/pybombs/install_manager.py", line 63, in _check_if_pkg_goes_into_tree
    if not self.pm.exists(pkg):
  File "/usr/local/lib/python2.7/dist-packages/pybombs/package_manager.py", line 127, in exists
    if self.check_package_flag(name, 'forceinstalled'):
  File "/usr/local/lib/python2.7/dist-packages/pybombs/package_manager.py", line 88, in check_package_flag
    recipe.get_recipe(pkgname).category
  File "/usr/local/lib/python2.7/dist-packages/pybombs/recipe.py", line 398, in get_recipe
    if target is not None and r.target != target:
AttributeError: 'Recipe' object has no attribute 'target'
```
It fails with the same error in ubuntu:16.04 image as well.

carlesfernandez, 2018-08-10:
Same problem here. It worked two days ago :-(

velichkov, 2018-08-11:
Installing `ruamel-yaml` from the apt/yum repository seems to fix the problem
```
apt-get install python-ruamel.yaml
dnf install python2-ruamel-yaml
```
This problem seems related to #498. Without this package the `pybombs prefix init` removes some fields from `~/.pybombs/config.yml` and it seems to breaks it.

Before running prefix init
```yaml
!!omap
- config:
    git-cache: /root/.pybombs/gitcache
    elevate_pre_args:
    - sudo
    - -H
    makewidth: 24
    packagers: apt,pymod,pip,pkgconfig,cmd
- recipes:
    gr-recipes: git+https://github.com/gnuradio/gr-recipes.git
- prefix_aliases:
    myprefix: /root/prefix
```
and after
```yaml
!!omap
- config:
    default_prefix: myprefix
- recipes:
    gr-recipes: git+https://github.com/gnuradio/gr-recipes.git
- prefix_aliases:
    myprefix: /root/prefix
```

rfenouil, 2018-08-16:
Hello,

I experience the same issue when trying to install gnuradio using pybombs on Ubuntu 14.04.5 LTS (never managed to compile ir from source because of pygtk dependency).

'''pip install ruamel.yaml''' did not save me...

metalhead777, 2018-08-16:
I have similar issues while installing gnuradio with pybombs on Ubuntu 16.04.5 LTS. 

The fix that worked for me also targets ```ruamel.yaml```, as described in #517. 

I followed the [readme](/gnuradio/pybombs/blob/master/README.md) until

```bash
[sudo] pip install PyBOMBS
```

After that, I reverted ruamel.yaml to 0.15.43 as described (@rfenouil this might also work for you):

```bash
pip uninstall ruamel.yaml
pip install ruamel.yaml==0.15.43
```

Afterwards, I continued with the Quickstart section in the readme:

```bash
pybombs auto-config
pybombs recipes add-defaults
pybombs prefix init ~/prefix -a myprefix -R gnuradio-default
pybombs config default_prefix myprefix
gnuradio-companion
```

rfenouil, 2018-08-16:
Just an update, after reading #515 I tried:
```pip install ruamel.yaml==0.15.43``` and that worked... until I ran into pygtk not found error again (but that's another story).

AvdN, 2018-08-16:
You should be able to do ``pip install ruamel.yaml==0.15.54``.

AvdN, 2018-08-16:
PR #518 should prevent this from happening. Although it will not fix a broken ``config.yml``

mbr0wn, 2018-08-16:
#518 is merged, closing.

closed 2018-08-16 by mbr0wn