gnuradio/pybombs #514: AttributeError: 'module' object has no attribute 'round_trip_load'

velichkov, 2018-08-11:
`./run-tests.sh --skip-pylint --container=ubuntu_1604` fails with

```
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.auto-config - DEBUG - apt is supported!
PyBOMBS.auto-config - DEBUG - Attempting to add binary package manager yumdnf
PyBOMBS.auto-config - DEBUG - Attempting to add binary package manager port
PyBOMBS.auto-config - DEBUG - Attempting to add binary package manager brew
PyBOMBS.auto-config - DEBUG - Attempting to add binary package manager zypper
PyBOMBS.auto-config - DEBUG - Attempting to add binary package manager pacman
PyBOMBS.auto-config - DEBUG - Attempting to add binary package manager portage
PyBOMBS.auto-config - DEBUG - Attempting to add binary package manager pymod
PyBOMBS.auto-config - DEBUG - pymod is supported!
PyBOMBS.auto-config - DEBUG - Attempting to add binary package manager pip
PyBOMBS.Packager.pip - DEBUG - Using pip executable: pip2
PyBOMBS.auto-config - DEBUG - pip is supported!
PyBOMBS.auto-config - DEBUG - Attempting to add binary package manager pkgconfig
PyBOMBS.auto-config - DEBUG - Attempting to add binary package manager cmd
PyBOMBS.auto-config - DEBUG - cmd is supported!
PyBOMBS.auto-config - DEBUG - Using binary packagers: ['apt', 'pymod', 'pip', 'cmd']
PyBOMBS.ConfigManager - INFO - Creating new config file /root/.pybombs/config.yml
Traceback (most recent call last):
  File "/usr/local/bin/pybombs", line 9, in <module>
    load_entry_point('PyBOMBS==2.3.3a0', 'console_scripts', 'pybombs')()
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/main.py", line 32, in main
    return dispatch() or 0
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/commands/base.py", line 194, in dispatch
    return get_cmd_dict(cmd_list)[args.command](cmd=args.command, args=args).run()
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/commands/autoconfig.py", line 121, in run
    cfg_file=self.cfg_file,
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/config_manager.py", line 638, in update_cfg_file
    return PBConfigFile(cfg_file).update(new_data)
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/config_file.py", line 76, in __init__
    self.yaml = AbstractYaml()
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.3.3a0-py2.7.egg/pybombs/config_file.py", line 52, in __init__
    self._load = self.yaml.round_trip_load
AttributeError: 'module' object has no attribute 'round_trip_load'
```
It fails on `RUN pybombs -v auto-config`
And here are the ruamel versions installed in the container
```bash
$ dpkg -l | grep ruamel
ii  python-ruamel.ordereddict  0.4.9-1                       amd64        dictionary that keeps keys in insertion resp. sorted order
ii  python-ruamel.yaml         0.10.23-1                     amd64        roundtrip YAML parser/emitter (Python 2 module)

$ pip list | grep ruamel
ruamel.ordereddict (0.4.9)
ruamel.yaml (0.10.23)
```

AvdN, 2018-08-16:
That is a pretty old version of ruamel.yaml. After setting up your ubuntu 16.04 you should upgrade ruamel.yaml (and pip and virtualenvutils). Preferably you should use a virtualenv in which to do so (i.e. intall virtualenv in your system Python, then create the virtualenv, update its pip, and then install other packages).

mbr0wn, 2018-08-17:
ruamel.yaml 0.10 should have the `round_trip_load()` call, though. Hmm.....

mbr0wn, 2019-09-19:
I've dropped < 0.15 support so I don't think this still applies.

closed 2019-09-19 by mbr0wn