gnuradio/pybombs #515: Multiple PyBOMBS recipe add seem to overwrite the config.yml, rather than add

richmaes, 2018-08-13:
Performing these two instructions back to back results in only the ettus alias being added.  The gr-recipes is deleted. 

$ pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git
$ pybombs recipes add ettus git+https://github.com/EttusResearch/ettus-pybombs.git

I have validated that If I just perform the gr-recipes command, then gr-recipes is added successfully, until I perform the second command.
I have attached logs in verbose mode.  

[pybombs_add.log](https://github.com/gnuradio/pybombs/files/2283908/pybombs_add.log)

trevora, 2018-08-15:
We discovered that ruamel.yaml version 15.57 was causing `pybombs prefix init` to fail. We reverted back to version 15.43 and it worked again.

We manually edited our config.yml file due to the issue described above and were able to successfully start installing recipes.

We didn't try reinstalling our recipes after reverting ruamel.yaml, so I don't know for certain whether changing the library would fix that issue as well. However, we suspect that the issues we saw with `pybombs prefix init` are related to the issues with config.yml.

We noticed that there have been multiple revisions of ruamel.yaml released today (8/15) to address various issues. We chose 15.43 because that was the version installed on an older system where everything worked as expected.

Hope this helps.

richmaes, 2018-08-15:
Rolling to ruamel.yaml==0.15.43 resolved the issue I was seeing.  Thank you for supplying that information.

AvdN, 2018-08-16:
This should no longer occur with PR #518 merged. However that will not revert already mangled config.yml files.

It would have  been nice if someone would have pinged me on this. (The error is not in the changes in ruamel.yaml, but in the assumptions in pybombs. I was able to relatively easily find where though).

mbr0wn, 2018-08-16:
@AvdN thanks for figuring this out.

closed 2018-08-16 by mbr0wn