gnuradio/pybombs #516: export Py3k paths in setup_env.sh

jdemel, 2018-08-15:
Hi all,
I try to use GR next with PyBOMBS. But it seems that `setup_env.sh` still exports all the Py2k paths to `PYTHONPATH`. At least that's the impression after trying `pybombs prefix write-env`. Is this a known issue? I couldn't find anything on the issue tracker.

From `PyBOMBS.ConfigManager - INFO - Prefix Python version is: 3.6.5` I assume PyBOMBS tries to make this a Py3k prefix.

PyBOMBS version: `PyBOMBS - INFO - PyBOMBS Version 2.3.3a0`

mbr0wn, 2018-08-16:
Yeah we currently hard-code the Python paths. It's not great. We can keep chaining Python paths, but I think we should have something cleaner. Any suggestions, any takers?

jdemel, 2018-08-17:
There are a few variables to consider.
Debian-based OS? The existence `dist-packages` seems to depend on it.
Python version? Py2k or Py3k? Obviously paths like `python3.6` or `python2.7` depend on that.
Do we need `lib` or `lib64` or just use both?

All in all, the necessary additions to `$PYTHONPATH` are
`$PYBOMBS_PREFIX/lib[64]/python[version]/[dist/site]-packages`

PyBOMBS knows which Python version to use. At least it prints a prefix python version. It wouldn't make sense to add any other python versions.

One way to handle this would be:
1. Add `lib` and `lib64` for all paths. At least as long as this does not cause problems
2. Add `dist` and `site` versions for all paths.
3. Differentiate between Py2k and Py3k and figure out which paths to add for which version. Probably this involves maintaining a list and some more code to determine specifics.

mbr0wn, 2018-09-18:
This will break 3.8. Needs fixing.

binlaveloos, 2019-08-25:
please fix the python version detection issues.
On my ubuntu x64 18.04 pybombs is pretty much useless.
without manual hacking it wont work python 2 and it wont work with python 3
A year has passed since this was labeled as a bug...

jdemel, 2019-08-26:
@binlaveloos there is a PR #541 that might fix this issue. But there are multiple things to consider and eventually, this seems to be a non-trivial issue. 
Since this open source project depends on volunteers, the best way forward is to contribute to issues yourself. So, try to propose a solution. Engage in a discussion to fix this issue. These are the things I can recommend to speed up the process.

mbr0wn, 2019-09-19:
I've started fixing this with 68685a4b53145d7e8937f820f18634622373c35a. Not great, but it's something.

closed 2019-09-19 by mbr0wn