gnuradio/pybombs #519: Store configuration data in XDG compliant directory

AvdN, 2018-08-16:
After testing some issues with pybombs and ruamel.yaml, I was left with yet another hidden file in my home-directory.
On Linux pybombs should follow the [XDG Base Directory Specification](http://standards.freedesktop.org/basedir-spec/basedir-spec-latest.html) and move the ``~/.pybombs`` directory to $XDG_CONFIG_HOME/pybombs (defaulting to ``~/.config/pybombs``.

Potentially the ``gitcache`` resp. the ``recipes`` should be under ``$XDG_CACHE_HOME`` resp. ``$XDG_DATA_HOME/pybombs``

Although easy to implement, you can install https://pypi.org/project/xdg/ and import from there. 

I suggest that in case you find an existing **flle** ``~/.pybombs/config.yml`` that you move it to the XDG compliant place ( usually that would be ``~/.config/pybombs/config.yml`` )  and then crate soft link back to `~/.pybombs`. That way new installations don't get clobbered, and  users of older ones still can find things. If the config.yml exists in both locations as **file**, that should throw an error.

mbr0wn, 2018-08-16:
Yeah this is a good idea. Any takers?

mbr0wn, 2018-08-17:
> I suggest that in case you find an existing flle ~/.pybombs/config.yml that you move it to the XDG compliant place ( usually that would be ~/.config/pybombs/config.yml ) and then crate soft link [...]

This is a good suggestion.