gnuradio/pybombs #520: PyBOMBS does not install gnuradio with error about uhd

armleo, 2018-08-30:
```
-- Configuring LibUHD support...
--   Dependency Boost_FOUND = 1
--   Dependency HAVE_PYTHON_PLAT_MIN_VERSION = TRUE
--   Dependency HAVE_PYTHON_MODULE_MAKO = FALSE
CMake Error at cmake/Modules/UHDComponent.cmake:59 (MESSAGE):
  Dependencies for required component LibUHD not met.
Call Stack (most recent call first):
  CMakeLists.txt:431 (LIBUHD_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/home/armleo/prefix/src/uhd/host/build/CMakeFiles/CMakeOutput.log".
See also "/home/armleo/prefix/src/uhd/host/build/CMakeFiles/CMakeError.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package uhd:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package uhd. Aborting.
```

jeremyrittenhouse, 2018-09-08:
The answer is right here `Dependency HAVE_PYTHON_MODULE_MAKO = FALSE`. You can verify this by running `python -c "import mako"` which should fail. Just run `pip install mako` to install

DavidLutton, 2018-09-10:
Same issue with mako and requests missing when installing UHD, I installed these with: `sudo apt install python-requests python-mako`

mbr0wn, 2018-09-18:
Mako and all the other stuff should get installed automatically though. How did you trigger the installation?

mbr0wn, 2018-09-18:
The Docker-based test for 18.04 is not failing. Please reopen if you can provide instructions on how to reproduce this (e.g., a Dockerfile).

closed 2018-09-18 by mbr0wn