gnuradio/pybombs #522: AssertionError following the Quickstart guide

DavidLutton, 2018-09-08:
On a fresh Ubuntu 18.04 installation with build-essential git python3-venv python3-pip installed
On a venv for PyBOMBS
  
I cannot get through the getting started guide, AssertionError when running `pybombs` recipes add-defaults`
Log and Backtrace [Pybombs_Assert.txt](https://gist.github.com/DavidLutton/ec590ffe54289cfdfa3bd80aa1c4c5eb) 

Last debug message shown is: "PyBOMBS.PackageManager - DEBUG - Checking if package git is installable..."

Checked [pybombs/debian.md at master · gnuradio/pybombs](https://github.com/gnuradio/pybombs/blob/master/debian.md) for clues by `pip freeze`, ruamel.yaml is installed 

Should I try to install PyBOMBS from source?

mbr0wn, 2018-09-10:
@DavidLutton Thanks for posting this. We recently fixed this bug on master, but we still need a released version of PyBOMBS with the fix. You can do a pip install from the git repo to avoid the issue.

closed 2018-09-10 by mbr0wn

SteveInBrussels, 2018-09-15:
I'm stuck too. Does not sound like this issue is really closed but pending "we still need a released version of PyBOMBS with the fix".
 - I do not understand "You can do a pip install from the git repo to avoid the issue." -- so what would be the equivalent few lines in the quick start? 
Or: 
- When will the quick start be working again? How will I know it has been fixed?

Thanks.

DavidLutton, 2018-09-17:
In order.
Yes

With pip:`pip install -e git+https://github.com/gnuradio/pybombs.git#egg=pybombs`
or if you use pipenv: `pipenv install -e git+https://github.com/gnuradio/pybombs.git#egg=pybombs`

Looking at the [TravisCI builds](https://travis-ci.org/gnuradio/pybombs) I cannot see any evidence of the Docker container based tests running.
Is this broken or are Docker tests executed elsewhere?