gnuradio/pybombs #523: Libfec problem installing

eugesd, 2018-09-11:
Hi all, I'm having some issues installing libfec, here is a print out of what I'm seeing:

 pybombs -p myprefix install libfec
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.15
PyBOMBS - INFO - PyBOMBS Version 2.3.3a0
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- libfec
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: libfec
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
Installing:  (100%) [=========================================================]
PyBOMBS.Packager.source - ERROR - Problem occurred while building package libfec:
Installation failed
PyBOMBS.install_manager - ERROR - Error installing package libfec. Aborting.

Thanks!
Eugenio

reiinakano, 2018-09-13:
@eugesd I managed to work around this with a hack. I was trying to install gr-satellites which required `libfec`.

Run
```
pybombs -v install libfec
```
This will fail, but somehow, the .a and .so files are still built and placed in prefix/lib. Unfortunately, the .h file does not get placed in prefix/include because of the failure.

So we copy fec.h from the downloaded source directly in include
`cp prefix/src/libfec/fec.h prefix/include/`

We then need to comment out the line in gr-satellites (or whichever package you're trying to install) recipe requiring libfec. 

Then finally `pybombs -v install gr-satellites(or whatever your package is)` should work

reiinakano, 2018-09-13:
@eugesd Never mind, this thread https://www.mail-archive.com/search?l=discuss-gnuradio@gnu.org&q=subject:%22%5C%5BDiscuss%5C-gnuradio%5C%5D+pybombs%22&o=newest&f=1 has the solution.

Just run `pybombs install libfec` as root aka sudo. Not sure why this is needed though, the whole point of a prefix was to prevent the need for root...

eugesd, 2018-09-13:
Awesome, I was actually trying to install gr-satellites as well. Either way, `sudo` worked! Thanks.

closed 2018-09-13 by eugesd