gnuradio/pybombs #525: Fresh Pybombs stops at UHD

cpbridges, 2018-09-25:
Hi Everyone,

I'm running at odroid xu4, which I understand isn't yet ported fully but hope someone may have got over some hurdles. My system is:

```
odroid@odroid:~$ uname -a
Linux odroid 4.14.69-148 #1 SMP PREEMPT Mon Sep 10 13:07:55 -03 2018 armv7l armv7l armv7l GNU/Linux
```
I get an error on running:
pybombs prefix init -a default prefix/default/ -R gnuradio-default

The error looks like a compiler flag issue but I'm unsure where to set this to enable/disable the right flags. Any pointers?

Best, Chris
```
Configuring: (100%) [=================================================================================================================================================================]
Building:    (100%) [=================================================================================================================================================================]
[  2%] Built target uhd_rpclib
[  2%] Building CXX object lib/CMakeFiles/uhd.dir/convert/convert_with_neon.cpp.o
In file included from /home/odroid/prefix/default/src/uhd/host/lib/convert/convert_with_neon.cpp:10:0:
/usr/lib/gcc/arm-linux-gnueabihf/7/include/arm_neon.h: In member function 'virtual void __convert_fc32_1_sc16_item32_le_1_PRIORITY_SIMD::operator()(const input_type&, const output_type&, size_t)':
/usr/lib/gcc/arm-linux-gnueabihf/7/include/arm_neon.h:6740:1: error: inlining failed in call to always_inline 'float32x4_t vdupq_n_f32(float32_t)': target specific option mismatch
 vdupq_n_f32 (float32_t __a)
 ^~~~~~~~~~~
/home/odroid/prefix/default/src/uhd/host/lib/convert/convert_with_neon.cpp:26:53: note: called from here
     float32x4_t Q0 = vdupq_n_f32(float(scale_factor));
                                                     ^
In file included from /home/odroid/prefix/default/src/uhd/host/lib/convert/convert_with_neon.cpp:10:0:
/usr/lib/gcc/arm-linux-gnueabihf/7/include/arm_neon.h:10844:1: error: inlining failed in call to always_inline 'void vst1_s16(int16_t*, int16x4_t)': target specific option mismatch
 vst1_s16 (int16_t * __a, int16x4_t __b)
 ^~~~~~~~
/home/odroid/prefix/default/src/uhd/host/lib/convert/convert_with_neon.cpp:55:17: note: called from here
         vst1_s16((reinterpret_cast<int16_t *>(&output[i+6])), D15);
         ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/odroid/prefix/default/src/uhd/host/lib/convert/convert_with_neon.cpp:10:0:
/usr/lib/gcc/arm-linux-gnueabihf/7/include/arm_neon.h:9027:1: error: inlining failed in call to always_inline 'int16x4_t vrev32_s16(int16x4_t)': target specific option mismatch
 vrev32_s16 (int16x4_t __a)
 ^~~~~~~~~~
/home/odroid/prefix/default/src/uhd/host/lib/convert/convert_with_neon.cpp:54:39: note: called from here
         int16x4_t D15 = vrev32_s16(D14);
                                       ^
In file included from /home/odroid/prefix/default/src/uhd/host/lib/convert/convert_with_neon.cpp:10:0:
/usr/lib/gcc/arm-linux-gnueabihf/7/include/arm_neon.h:7552:1: error: inlining failed in call to always_inline 'int16x4_t vmovn_s32(int32x4_t)': target specific option mismatch
 vmovn_s32 (int32x4_t __a)
 ^~~~~~~~~
/home/odroid/prefix/default/src/uhd/host/lib/convert/convert_with_neon.cpp:53:38: note: called from here
         int16x4_t D14 = vmovn_s32(Q13);
                                      ^
In file included from /home/odroid/prefix/default/src/uhd/host/lib/convert/convert_with_neon.cpp:10:0:
/usr/lib/gcc/arm-linux-gnueabihf/7/include/arm_neon.h:7440:1: error: inlining failed in call to always_inline 'int32x4_t vcvtq_s32_f32(float32x4_t)': target specific option mismatch
 vcvtq_s32_f32 (float32x4_t __a)
 ^~~~~~~~~~~~~
/home/odroid/prefix/default/src/uhd/host/lib/convert/convert_with_neon.cpp:52:42: note: called from here
         int32x4_t Q13 = vcvtq_s32_f32(Q11);
                                          ^
In file included from /home/odroid/prefix/default/src/uhd/host/lib/convert/convert_with_neon.cpp:10:0:
/usr/lib/gcc/arm-linux-gnueabihf/7/include/arm_neon.h:1172:1: error: inlining failed in call to always_inline 'float32x4_t vmulq_f32(float32x4_t, float32x4_t)': target specific option mismatch
 vmulq_f32 (float32x4_t __a, float32x4_t __b)
 ^~~~~~~~~
/home/odroid/prefix/default/src/uhd/host/lib/convert/convert_with_neon.cpp:51:44: note: called from here
         float32x4_t Q11 = vmulq_f32(Q10, Q0);
                                            ^
In file included from /home/odroid/prefix/default/src/uhd/host/lib/convert/convert_with_neon.cpp:10:0:
/usr/lib/gcc/arm-linux-gnueabihf/7/include/arm_neon.h:10844:1: error: inlining failed in call to always_inline 'void vst1_s16(int16_t*, int16x4_t)': target specific option mismatch
 vst1_s16 (int16_t * __a, int16x4_t __b)
 ^~~~~~~~
/home/odroid/prefix/default/src/uhd/host/lib/convert/convert_with_neon.cpp:49:17: note: called from here
         vst1_s16((reinterpret_cast<int16_t *>(&output[i+4])), D13);
         ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/odroid/prefix/default/src/uhd/host/lib/convert/convert_with_neon.cpp:10:0:
[... 294 lines cut ...]
/usr/lib/gcc/arm-linux-gnueabihf/7/include/arm_neon.h:10392:1: error: inlining failed in call to always_inline 'float32x4_t vld1q_f32(const float32_t*)': target specific option mismatch
 vld1q_f32 (const float32_t * __a)
 ^~~~~~~~~
/home/odroid/prefix/default/src/uhd/host/lib/convert/convert_with_neon.cpp:29:80: note: called from here
         float32x4_t Q4 = vld1q_f32(reinterpret_cast<const float *>(&input[i+2]));
                                                                                ^
In file included from /home/odroid/prefix/default/src/uhd/host/lib/convert/convert_with_neon.cpp:10:0:
/usr/lib/gcc/arm-linux-gnueabihf/7/include/arm_neon.h:10392:1: error: inlining failed in call to always_inline 'float32x4_t vld1q_f32(const float32_t*)': target specific option mismatch
 vld1q_f32 (const float32_t * __a)
 ^~~~~~~~~
/home/odroid/prefix/default/src/uhd/host/lib/convert/convert_with_neon.cpp:28:78: note: called from here
         float32x4_t Q1 = vld1q_f32(reinterpret_cast<const float *>(&input[i]));
                                                                              ^
lib/CMakeFiles/uhd.dir/build.make:670: recipe for target 'lib/CMakeFiles/uhd.dir/convert/convert_with_neon.cpp.o' failed
make[2]: *** [lib/CMakeFiles/uhd.dir/convert/convert_with_neon.cpp.o] Error 1
CMakeFiles/Makefile2:119: recipe for target 'lib/CMakeFiles/uhd.dir/all' failed
make[1]: *** [lib/CMakeFiles/uhd.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package uhd:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package uhd. Aborting.
odroid@odroid:~$
```

dkozel, 2018-10-17:
@cpbridges (Hi!) did you get this resolved?

kazazes, 2018-10-28:
@dkozel, @mbr0wn's fix works for me, Ubuntu 18.04 on an ODROID-XU4.

callen5914, 2019-02-10:
I am getting this on Mac OS Mojave. Is this fix relevant?

michaelld, 2019-02-11:
@callen5914 Nope; not relevant for macOS / MacOS X / OSX installs.

closed 2019-02-16 by mbr0wn

rdjondo, 2019-09-17:
I hit the same problem today when building on the Raspberry Pi 3 and I tried updating the .pybombs/config.yml as recommended above but that did not help.

Instead, here is what I did instead to solve the issue and it seemed to work (solution from path shared on https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=873608 ):

in bladeRF/host/lib/convert/CMakeLists.txt 
line 104, :
         ${CMAKE_CURRENT_SOURCE_DIR}/convert_with_neon.cpp
         ${CMAKE_CURRENT_SOURCE_DIR}/convert_neon.S
     )
after the parenthesis add :
    SET ( CMAKE_CXX_FLAGS "-mfpu=neon" )

f4grx, 2020-06-03:
Same problem today with a Raspberry Pi 4

Editing the CMakefile was successful, and is a better fix than disabling neon altogether.

So this minor CMake typo is 9 months old and still not fixed in upstream sources used by pybombs?

kaushiksv, 2020-12-11:
rdjondo's solution works for Raspberry PI 4 Model B:
```shell
pi@raspberrypi:~ $ lscpu
Architecture:        armv7l
Byte Order:          Little Endian
CPU(s):              4
On-line CPU(s) list: 0-3
Thread(s) per core:  1
Core(s) per socket:  4
Socket(s):           1
Vendor ID:           ARM
Model:               3
Model name:          Cortex-A72
Stepping:            r0p3
CPU max MHz:         1500.0000
CPU min MHz:         600.0000
BogoMIPS:            270.00
Flags:               half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
pi@raspberrypi:~ $ uname -r
4.19.66-v7l+
pi@raspberrypi:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

gigs94, 2021-07-29:
So pyBOMBS doesn't control this software build, it's just a toolchain/build environment.   The issue in this thread was with uhd and so I opened the above issue and submitted a build request.   Once this gets accepted into the master uhd, then it should build without issues.