gnuradio/pybombs #529: GnuRadio installation fails not being able to find pthread_create

kenichi-fukushima, 2018-10-19:
I'm new to SDR and tried to install GnuRadio using PyBOMBS because a book recommends to do so.

I got a following error log in `<prefix>/src/uhd/host/build/CMakeFiles/CMakeError.log` when I ran `pybombs install  gnuradio gr-osmosdr`.

I'm using Ubuntu 18.04 and I have `libpthread` in `/usr/lib/x86_64-linux-gnu`.

```
Determining if the pthread_create exist failed with the following output:
Change Dir: /home/keni/PyBOMBS/default/src/uhd/host/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_f8d81/fast"
/usr/bin/make -f CMakeFiles/cmTC_f8d81.dir/build.make CMakeFiles/cmTC_f8d81.dir/build
make[1]: Entering directory '/home/keni/PyBOMBS/default/src/uhd/host/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_f8d81.dir/CheckSymbolExists.c.o
/usr/bin/cc    -o CMakeFiles/cmTC_f8d81.dir/CheckSymbolExists.c.o   -c /home/keni/PyBOMBS/default/src/uhd/host/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_f8d81
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_f8d81.dir/link.txt --verbose=1
/usr/bin/cc      -rdynamic CMakeFiles/cmTC_f8d81.dir/CheckSymbolExists.c.o  -o cmTC_f8d81 
CMakeFiles/cmTC_f8d81.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x1b): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_f8d81.dir/build.make:97: recipe for target 'cmTC_f8d81' failed
make[1]: *** [cmTC_f8d81] Error 1
make[1]: Leaving directory '/home/keni/PyBOMBS/default/src/uhd/host/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_f8d81/fast' failed
make: *** [cmTC_f8d81/fast] Error 2

File /home/keni/PyBOMBS/default/src/uhd/host/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/keni/PyBOMBS/default/src/uhd/host/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_bd4c5/fast"
/usr/bin/make -f CMakeFiles/cmTC_bd4c5.dir/build.make CMakeFiles/cmTC_bd4c5.dir/build
make[1]: Entering directory '/home/keni/PyBOMBS/default/src/uhd/host/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_bd4c5.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_bd4c5.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.10/Modules/CheckFunctionExists.c
Linking C executable cmTC_bd4c5
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_bd4c5.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_bd4c5.dir/CheckFunctionExists.c.o  -o cmTC_bd4c5 -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_bd4c5.dir/build.make:97: recipe for target 'cmTC_bd4c5' failed
make[1]: *** [cmTC_bd4c5] Error 1
make[1]: Leaving directory '/home/keni/PyBOMBS/default/src/uhd/host/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_bd4c5/fast' failed
make: *** [cmTC_bd4c5/fast] Error 2
```

scateu, 2019-01-11:
same problem in

scateu, 2019-01-11:
same problem in 

```
root@Pybombs:~/gnuradio-prefix# uname -a
Linux Pybombs 4.15.0-1031-aws #33-Ubuntu SMP Fri Dec 7 09:32:27 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
root@Pybombs:~/gnuradio-prefix# cat /etc/issue
Ubuntu 18.04.1 LTS \n \l
```

scateu, 2019-01-11:
and the same in Ubuntu 14.04.5 LTS:

```
Linux vm 3.13.0-149-generic #199-Ubuntu SMP Thu May 17 10:12:50 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

py2sdr, 2019-10-27:
I had the same issue.

In my case, the issue got resolved by installing python3-mako and python3-numpy

texadactyl, 2019-11-09:
I had the exact same symptom on a Raspberry Pi 3B running Raspbian Buster up-to-date.
The packages python3-mako and python3-numpy are installed.

playaspec, 2020-03-26:
One more for the pile. Ubuntu 16.04. I completely purged all previous instances of additional python packages. 16.04 ships with python 2.7.x and 3.5.x, but newest gnuradio through pybombs requires 3.6.x. I added the deadsnakes PPA, and installed python 3.6 and pip from that archive. Set 3.6 to be the default though update-alternatives, then installed pybombs following the instructions at: https://github.com/gnuradio/pybombs, eliminating any remaining missing dependencies along the way. Current cmake (3.17.0) and latest volk (2.2.0) built from source. Numpy (1:1.11.0-1ubuntu1, both 2.7 and 3.6)  and mako (1.0.3+ds1-1ubuntu1 2.7 and 3.6) are default packages. System libpthreads is default (libpthread-stubs0-dev/xenial). GCC Ubuntu 5.4.0-6ubuntu1~16.04.12.

I've spent better than 24 hours of in the chair beating on this trying to get it to build, but it always craps out at this same spot. The error log file:

/home/playaspec/Dev/gnuradio-3.7-pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp reads:

```
Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: /home/playaspec/Dev/gnuradio-3.7-pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_312c5/fast && /usr/bin/make -f CMakeFiles/cmTC_312c5.dir/build.make CMakeFiles/cmTC_312c5.dir/build
make[1]: Entering directory '/home/playaspec/Dev/gnuradio-3.7-pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_312c5.dir/src.c.o
/usr/bin/cc   -std=gnu99 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -DCMAKE_HAVE_LIBC_PTHREAD   -o CMakeFiles/cmTC_312c5.dir/src.c.o   -c /home/playaspec/Dev/gnuradio-3.7-pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_312c5
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_312c5.dir/link.txt --verbose=1
/usr/bin/cc  -std=gnu99 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -DCMAKE_HAVE_LIBC_PTHREAD    -rdynamic CMakeFiles/cmTC_312c5.dir/src.c.o  -o cmTC_312c5
CMakeFiles/cmTC_312c5.dir/src.c.o: In function `main':
src.c:(.text+0x3c): undefined reference to `pthread_create'
src.c:(.text+0x48): undefined reference to `pthread_detach'
src.c:(.text+0x54): undefined reference to `pthread_cancel'
src.c:(.text+0x65): undefined reference to `pthread_join'
src.c:(.text+0x79): undefined reference to `pthread_atfork'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_312c5.dir/build.make:103: recipe for target 'cmTC_312c5' failed
make[1]: *** [cmTC_312c5] Error 1
make[1]: Leaving directory '/home/playaspec/Dev/gnuradio-3.7-pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp'
Makefile:138: recipe for target 'cmTC_312c5/fast' failed
make: *** [cmTC_312c5/fast] Error 2

Source file was:
#include <pthread.h>

void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_cancel(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/playaspec/Dev/gnuradio-3.7-pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_d6090/fast && /usr/bin/make -f CMakeFiles/cmTC_d6090.dir/build.make CMakeFiles/cmTC_d6090.dir/build
make[1]: Entering directory '/home/playaspec/Dev/gnuradio-3.7-pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_d6090.dir/CheckFunctionExists.c.o
/usr/bin/cc   -std=gnu99 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_d6090.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.17/Modules/CheckFunctionExists.c
Linking C executable cmTC_d6090
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_d6090.dir/link.txt --verbose=1
/usr/bin/cc  -std=gnu99 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_d6090.dir/CheckFunctionExists.c.o  -o cmTC_d6090  -lpthreads
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_d6090.dir/build.make:103: recipe for target 'cmTC_d6090' failed
make[1]: *** [cmTC_d6090] Error 1
make[1]: Leaving directory '/home/playaspec/Dev/gnuradio-3.7-pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp'
Makefile:138: recipe for target 'cmTC_d6090/fast' failed
make: *** [cmTC_d6090/fast] Error 2

Determining if the include file intrin.h exists failed with the following output:
[... 170 lines cut ...]
                         INVALID_HANDLE_VALUE,    // use paging file
                         NULL,                    // default security
                         PAGE_READWRITE,          // read/write access
                         0,                       // max. object size
                         size,                    // buffer size
                         lpName);                 // name of mapping object
        return 0;
    }
Determining if the include file windows.h exists failed with the following output:
Change Dir: /home/playaspec/Dev/gnuradio-3.7-pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_11a7a/fast && /usr/bin/make -f CMakeFiles/cmTC_11a7a.dir/build.make CMakeFiles/cmTC_11a7a.dir/build
[... cut ...]
make[1]: Leaving directory '/home/playaspec/Dev/gnuradio-3.7-pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp'
Makefile:138: recipe for target 'cmTC_11a7a/fast' failed
make: *** [cmTC_11a7a/fast] Error 2
```

While I'm not a professional developer, I am a sysadmin by day, and usually pretty adept at solving issues like this, but this one has got me. If I had to guess, it has something to do with gnuradio's cmake environment detection missing an edge case that should otherwise be valid. Any help would be appreciated.

velichkov, 2020-03-26:
Hi @playaspec,

> Ubuntu 16.04.
> I've spent better than 24 hours of in the chair beating on this trying to get it to build, but it always craps out at this same spot. The error log file

An honest question: why did you spent so much time trying to install the latest version of gnuradio on such an old Ubuntu version? Why don't you update your Ubuntu to a more recent version and then install gnuradio 3.8 from the [gnuradio's ppa](https://launchpad.net/~gnuradio/+archive/ubuntu/gnuradio-releases) repository? 

> /home/playaspec/Dev/gnuradio-3.7-pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp reads:

Don't look in `CMakeFiles/CMakeTmp ` or `CMakeFiles/CMakeError.log` files, there are a lot of error there and that's normal. Instead provide the full cmake output from the terminal (not the `CMakeOutput.log` file) and attach it as a file as pasting long outputs (as in your previous message) makes the discussion hard to follow.

playaspec, 2021-12-01:
> An honest question: why did you spent so much time trying to install the latest version of gnuradio on such an old Ubuntu version? Why don't you update your Ubuntu to a more recent version and then install gnuradio 3.8 from the gnuradio's ppa repository?

At the time of my post, Ubuntu 16.04, which is an LTS (Long Term Support) still had another 13 months of official support. Just because it's not the newest doesn't mean it's obsolete. That particular machine is used to support production data acquisition systems used for scientific research. It's neither trivial nor practical to blindly upgrade just 'because'.

willcode, 2021-12-01:
It sound like this issue was resolved one way or another, so closing.

closed 2021-12-01 by willcode