gnuradio/pybombs #532: Configuring GNU Radio cmake flags with pybombs config

kazazes, 2018-11-16:
Hey all, I've been trying to disable Qt-dependent features of GNU Radio for a docker install and have had no success. I've reverted to editing the recipes before installing.
```bash
sed -i -e "s/-DENABLE_GRC=ON/-DENABLE_GRC=OFF/g" -e "s/-DENABLE_GR_QTGUI=ON/-DENABLE_GR_QTGUI=OFF/g"
```

Is there a more graceful way of doing this with `pybombs config`? I can't quite get it to work.

tranzmatt, 2020-01-09:
Did you ever get an answer to this?  I need something similar for building uhd on a raspberrypi3

kazazes, 2020-01-10:
@tranzmatt I just teased this incantation out of the ARM Docker image I've been using since. You may just want to use [that](https://hub.docker.com/r/pckzs/pybombs)...

```
echo "vars:\n  config_opt: \"-DENABLE_OSMOSDR=ON -DENABLE_FMCOMMS2=ON -DENABLE_PLUTOSDR=ON -DENABLE_AD9361=ON -DENABLE_RAW_UDP=ON -DENABLE_PACKAGING=ON -DENABLE_UNIT_TESTING=OFF -DPYTHON_EXECUTABLE=/usr/bin/python3\"\n" >> /root/.pybombs/recipes/gr-recipes/gnss-sdr.lwr
echo "vars:\n  config_opt: \" -DINSTALL_LIB_DIR=\$prefix/lib -DENABLE_PYTHON3=ON\"\n" >> /root/.pybombs/recipes/gr-recipes/uhd.lwr
sed -i '/cppunit/d' /root/.pybombs/recipes/gr-recipes/gnuradio.lwr
sed -i '/gsl/d' /root/.pybombs/recipes/gr-recipes/gnuradio.lwr
sed -i '/alsa/d' /root/.pybombs/recipes/gr-recipes/gnuradio.lwr
sed -i '/wxpython/d' /root/.pybombs/recipes/gr-recipes/gnuradio.lwr
sed -i '/thrift/d' /root/.pybombs/recipes/gr-recipes/gnuradio.lwr
sed -i '/pygtk/d' /root/.pybombs/recipes/gr-recipes/gnuradio.lwr
sed -i '/pycairo/d' /root/.pybombs/recipes/gr-recipes/gnuradio.lwr
sed -i '/pyqt4/d' /root/.pybombs/recipes/gr-recipes/gnuradio.lwr
sed -i '/qwt/d' /root/.pybombs/recipes/gr-recipes/gnuradio.lwr
sed -i '/gitbranch/d' /root/.pybombs/recipes/gr-recipes/gnuradio.lwr
sed -i '/vars/d' /root/.pybombs/recipes/gr-recipes/gnuradio.lwr
sed -i '/config_opt/d' /root/.pybombs/recipes/gr-recipes/gnuradio.lwr
sed -i '/ssl/d' /root/.pybombs/recipes/gr-recipes/apache-thrift.lwr
echo "gitbranch: v3.7.13.4\n" >> /root/.pybombs/recipes/gr-recipes/gnuradio.lwr
echo "vars:\n  config_opt: \"-DENABLE_GR_AUDIO=OFF -DENABLE_GR_CHANNELS=OFF -DENABLE_GR_COMEDI=OFF -DENABLE_GR_DIGITAL=ON -DENABLE_DOXYGEN=OFF -DENABLE_GR_DTV=OFF -DENABLE_GR_FEC=OFF -DENABLE_GR_TRELLIS=OFF -DENABLE_GR_VIDEO_SDL=OFF -DENABLE_GR_VOCODER=ON -DENABLE_GR_WAVELET=OFF -DENABLE_GR_ZEROMQ=ON -DENABLE_GR_ANALOG=ON -DENABLE_GR_FFT=ON -DENABLE_GR_FILTER=ON -DENABLE_GRC=OFF\"\n" >> /root/.pybombs/recipes/gr-recipes/gnuradio.lwr
sed -i '/gitrev/d' /root/.pybombs/recipes/gr-recipes/gr-iio.lwr
sed -i 's/git+git:\/\/git.osmocom.org\/gr-osmosdr/git+https:\/\/github.com\/Nuand\/gr-osmosdr.git/' ~/.pybombs/recipes/gr-recipes/gr-osmosdr.lwr
echo "gitbranch: master\n" >> /root/.pybombs/recipes/gr-recipes/gr-iio.lwr
pybombs prefix init ${PyBOMBS_init} -a ${PyBOMBS_prefix}
```

tranzmatt, 2020-02-12:
Yeah, I've been having to do something similar in the Dockerfile, clobbering recipes after adding them :p