gnuradio/pybombs #533: Using python3 executable when Prefix is set to python2.7

zhudsonSM, 2019-01-08:
I am trying to install gnuradio but getting a "configuration failed" error. I believe it stems from the compiler trying to use python3 "User set python executable /user/bin/python3" when the prefix is set to python 2.7.12

> $ pybombs install gnuradio
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.12
PyBOMBS - INFO - PyBOMBS Version 2.3.3
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
/usr/local/lib/python2.7/dist-packages/pip/_vendor/requests/__init__.py:83: RequestsDependencyWarning: Old version of cryptography ([1, 2, 3]) may cause slowdown.
  warnings.warn(warning, RequestsDependencyWarning)
Install tree:
|
\- gnuradio
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: gnuradio
Cloning into 'gnuradio'...
remote: Enumerating objects: 1164, done.
remote: Counting objects: 100% (1164/1164), done.
remote: Total 2892 (delta 1164), reused 1164 (delta 1164), pack-reused 1728
Receiving objects: 100% (2892/2892), 1.28 MiB | 1.45 MiB/s, done.
Resolving deltas: 100% (2315/2315), completed with 322 local objects.
Checking connectivity... done.
Submodule 'volk' (https://github.com/gnuradio/volk.git) registered for path 'volk'
Cloning into 'volk'...
remote: Enumerating objects: 40, done.
remote: Counting objects: 100% (40/40), done.
remote: Compressing objects: 100% (24/24), done.
remote: Total 7792 (delta 15), reused 30 (delta 15), pack-reused 7752
Receiving objects: 100% (7792/7792), 2.13 MiB | 1.44 MiB/s, done.
Resolving deltas: 100% (5416/5416), done.
Checking connectivity... done.
Submodule path 'volk': checked out '3f3c91ee3ce51a31d8c23398124df74d3aa42955'
Configuring: (100%) [========================================================================================================================================================================================================================================================]
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
-- Build type set to RelWithDebInfo.
-- Extracting version information from git describe...
-- Compiler Version: cc (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- Building Static Libraries: OFF
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   unit_test_framework
--   chrono
--   atomic
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "2.7") 
-- Could NOT find PythonLibs: Found unsuitable version "2.7.12", but required is exact version "3.5" (found /usr/lib/x86_64-linux-gnu/libpython2.7.so)
-- Python checking for six - python 2 and 3 compatibility library - found
-- 
-- Checking for module SWIG
-- Found SWIG version 2.0.12.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = FALSE
--   Dependency SIX_FOUND = TRUE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled python-support but configuration checked failed
Call Stack (most recent call first):
  CMakeLists.txt:357 (GR_REGISTER_COMPONENT)
>
>-- Configuring incomplete, errors occurred!
See also "/home/zhudson/prefix/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/home/zhudson/prefix/src/gnuradio/build/CMakeFiles/CMakeError.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.

AlisterAmo, 2019-02-16:
Set your theory aside for a moment in order to focus on real error data:
`Dependency SWIG_VERSION_CHECK = FALSE`
It suggest that you have swig installed but version is not enough.
Share with us the result of `swig -version`

binlaveloos, 2019-08-25:
I had similar problems with pybombs, but for me where boost dependencies.
The reason why the dependencies where not installed correctly on my system seems to be that incorrect packages where being installed with apt in an earlier stage (i noticed a lot of python-xxx  packages being installed instead of python3-xxx)
After installing dependencies manualy (correct binary packages, and python modules it worked fine
I don't know if pip is used in pybombs, but if it is, it is probably using pip instead of pip3 for installing python packages

So i do think zhudsonSM has a valid point for suggesting incorrect python version is the cause.

reverbee, 2020-01-10:
Hi,I also met same problem when installing package gnuradio, can you share how you solved this problem, thanks!

zhudsonSM, 2020-01-10:
My issue was solved by correctly installing the latest version of Pybombs. Be sure to follow the directions in the ReadMe here and not any directions on any other sites/posts.

closed 2020-01-10 by zhudsonSM