gnuradio/pybombs #536: PyBOMBS Gr-GSM building error - gnuradio

sunil2985, 2019-02-26:
Before posting an issue, please check the following:

- [ ] Is this actually a bug in PyBOMBS, or is it a bug in the package you're
      installing? If it's the latter, this is the wrong place to post. It won't
      get fixed unless the author of the package is informed.

- [ ] How can this issue be reproduced? We have Docker-based tests for various
      Linux distributions, which are the reference for whether or not we
      consider a specific distribution as functional. Have you provided
      instructions to reproduce?

- [ ] Have you included all relevant logs to help developers find the issue?
      Did you bump the verbosity (`-v` or `-vv`)?

sunil2985, 2019-02-26:
Hi, @mbr0wn 

I am getting build error while installing GR-GSM 
"problem occurred while building package gnuradio:Build Failed"

Please help how to fix this
[image]

ckuethe, 2019-03-20:
What platform is this on? I'm Ubuntu 18.10 x86_64 and was able to build gr-gsm.