gnuradio/pybombs #540: Pybombs Failure to Install Gnuradio and UHD Mac. IOKIT NOT Found

UsersDoProgram, 2019-05-16:
Everytime I run pybombs install gnuradio on my Macbook Pro running Mojave I get an error saying, 
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 3.6.3
PyBOMBS - INFO - PyBOMBS Version 2.3.3
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gnuradio
   |
   \- uhd
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: uhd
PyBOMBS.Packager.source - WARNING - Build dir already exists: /Users/chris/Documents/src/uhd/host/build
Building:    (100%) [=========================================================]
[  2%] Built target uhd_rpclib
[  2%] Building CXX object lib/CMakeFiles/uhd.dir/usrp/x300/x300_impl.cpp.o
In file included from /Users/chris/Documents/src/uhd/host/lib/usrp/x300/x300_impl.cpp:8:
In file included from /Users/chris/Documents/src/uhd/host/lib/usrp/x300/x300_impl.hpp:20:
In file included from /Users/chris/Documents/src/uhd/host/include/uhd/transport/nirio/niusrprio_session.h:14:
In file included from /Users/chris/Documents/src/uhd/host/include/uhd/transport/nirio/niriok_proxy.h:16:
/Users/chris/Documents/src/uhd/host/include/uhd/transport/nirio/nirio_driver_iface.h:26:14: fatal error: 
      'IOKit/IOKitLib.h' file not found
    #include <IOKit/IOKitLib.h>
             ^~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [lib/CMakeFiles/uhd.dir/usrp/x300/x300_impl.cpp.o] Error 1
make[1]: *** [lib/CMakeFiles/uhd.dir/all] Error 2
make: *** [all] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package uhd:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package uhd. Aborting.
How can I solve this as I can't find a replica of my error anywhere.

michaelld, 2019-05-16:
Some background info would be useful:
* how are GR / UHD dependencies installed? (e.g., MacPorts, Fink, Homebrew, from source manually; some combination)
* please list out any changes to PyBOMBS you made, e.g., in the setup script?
* what are the commands you issued to get to this point of using PyBOMBS?
* please execute PyBOMBS with extra verbosity ... I think: "-vv" ... get "help" and look for "verbose" and use "extra verbosity", whatever flags that is. We need to see the link command.

My PyBOMBS install is currently failing; I'll work on getting it fixed for this & another related issue.

michaelld, 2019-05-16:
Looks like you're trying to build with/for Py36, yes? Did you make any special changes and/or follow some specific guide for getting Py3 working here?