gnuradio/pybombs #542: Failing to install recipe gnuradio38. It looks like pybombs tries to use the recipe name as the git branch...

gmagno, 2019-07-20:
Failing to install recipe gnuradio38.
It looks like pybombs tries to use the recipe name as the git branch name when raising the following error:

    ...
    PyBOMBS.Fetcher.git - DEBUG - We have git version 2.17.1
    PyBOMBS.Fetcher.git - DEBUG - Using url - https://github.com/gnuradio/gnuradio.git
    PyBOMBS.Fetcher.git - DEBUG - Adding remote into git ref
    PyBOMBS.GitCacheManager - DEBUG - Adding remote: gnuradio38 -> https://github.com/gnuradio/gnuradio.git
    PyBOMBS.GitCacheManager - DEBUG - Remote URL https://github.com/gnuradio/gnuradio.git already registered.
    fatal: No such remote or remote group: gnuradio38
    PyBOMBS.Fetcher - ERROR - Unexpected error while fetching git+https://github.com/gnuradio/gnuradio.git.
    PyBOMBS.Fetcher - ERROR - Command '['git', 'remote', 'update', 'gnuradio38']' returned non-zero exit status 1.
    PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio38:
    Unable to fetch recipe gnuradio38
    PyBOMBS.install_manager - ERROR - Error installing package gnuradio38. Aborting.

In order to reproduce this issue I suggest the following steps:

1. install `gnuradio-default` recipe;
2. install `gnuradio38` on a separate prefix, this should fail.

As a workaround rename the recipe file from `.../gnuradio38.lwr` to `gnuradio.lwr` (you might want to backup gnuradio.lwr beforehand)

mhostetter, 2019-07-24:
I'm having a similar issue. I installed GNU Radio 3.7 into prefix `~/gnuradio-3.7` and now am trying to install GNU Radio 3.8 into prefix `~/gnurdaio-3.8`. I'm running into the following error.

```bash
$ pybombs --version
2.3.4a0
```

### Install GNU Radio 3.7 successfully

```bash
$ pybombs prefix init ~/gnuradio-3.7 -a gr37
$ pybombs -p gr37 install gnuradio
```

### Install GNU Radio 3.8 with errors

```
$ pybombs prefix init ~/gnuradio-3.8 -a gr38
$ pybombs -p gr38 -vv install gnuradio38 
...
[DEBUG] Adding remote into git ref
[DEBUG] Adding remote: gnuradio38 -> https://github.com/gnuradio/gnuradio.git
[DEBUG] Remote URL https://github.com/gnuradio/gnuradio.git already registered.
fatal: No such remote or remote group: gnuradio38
...
```

It seems pybombs is trying to create a git remote called `gnuradio38` pointing to `https://github.com/gnuradio/gnuradio.git` but is failing because that URL is already registered (likely from the `gnuradio` recipe during the GNU Radio 3.7 install).

Full output below:

```bash
$ pybombs prefix init ~/gnuradio-3.8 -a gr38
$ pybombs -p gr38 -vv install gnuradio38 
[DEBUG] Reading config info from file: /etc/pybombs/config.yml
[DEBUG] Parsing config file failed (/etc/pybombs/config.yml).
[DEBUG] Reading config info from file: /home/matt/.pybombs/config.yml
[DEBUG] New config items: ordereddict([('git-cache', '/home/matt/.pybombs/gitcache'), ('elevate_pre_args', ['sudo', '-H']), ('makewidth', 8), ('packagers', 'apt,pymod,pip,pkgconfig,cmd'), ('default_prefix', 'gr37')])
[DEBUG] Template directory: /home/matt/.local/lib/python2.7/site-packages/pybombs/templates
[DEBUG] Inspecting config file: /home/matt/.pybombs/config.yml
[DEBUG] Resolving prefix alias gr38.
[DEBUG] Choosing prefix dir from command line: /home/matt/gnuradio-3.8
[DEBUG] Choosing default prefix config dir: /home/matt/gnuradio-3.8/.pybombs
[DEBUG] Inspecting config file: /home/matt/gnuradio-3.8/.pybombs/config.yml
[DEBUG] Prefix source dir is: /home/matt/gnuradio-3.8/src
[DEBUG] Trying to load inventory file /home/matt/gnuradio-3.8/.pybombs/inventory.yml...
[DEBUG] Prefix-local recipe dir is: /home/matt/gnuradio-3.8/.pybombs/recipes
[DEBUG] Reading config info from file: /home/matt/gnuradio-3.8/.pybombs/config.yml
[DEBUG] New config items: {}
[DEBUG] Full list of recipe locations: ['/home/matt/gnuradio-3.8/.pybombs/recipes', '/home/matt/.pybombs/recipes/gr-recipes', '/home/matt/.pybombs/recipes/gr-etcetera', '/home/matt/.local/lib/python2.7/site-packages/pybombs/recipes']
[DEBUG] Named recipe locations: {'gr-etcetera': 'git+https://github.com/gnuradio/gr-etcetera.git', 'gr-recipes': 'git+https://github.com/gnuradio/gr-recipes.git'}
[DEBUG] Python version derived from current interpreter.
[INFO] Prefix Python version is: 2.7.16
[INFO] PyBOMBS Version 2.3.4a0
[DEBUG] Adding recipe location: /home/matt/gnuradio-3.8/.pybombs/recipes
[DEBUG] Scanning directory '/home/matt/gnuradio-3.8/.pybombs/recipes' for recipes...
[DEBUG] Found 0 new recipes.
[DEBUG] Loading templates.
[TRACE] Adding template /home/matt/.local/lib/python2.7/site-packages/pybombs/templates/prefix.lwt
[TRACE] Adding template /home/matt/.local/lib/python2.7/site-packages/pybombs/templates/autoconf.lwt
[TRACE] Adding template /home/matt/.local/lib/python2.7/site-packages/pybombs/templates/sdk.lwt
[TRACE] Adding template /home/matt/.local/lib/python2.7/site-packages/pybombs/templates/empty.lwt
[TRACE] Adding template /home/matt/.local/lib/python2.7/site-packages/pybombs/templates/distutils.lwt
[TRACE] Adding template /home/matt/.local/lib/python2.7/site-packages/pybombs/templates/cmake.lwt
[TRACE] Adding template /home/matt/.local/lib/python2.7/site-packages/pybombs/templates/bjam.lwt
[TRACE] Adding template /home/matt/.local/lib/python2.7/site-packages/pybombs/templates/bootstrapautoconf.lwt
[TRACE] Adding template /home/matt/.local/lib/python2.7/site-packages/pybombs/templates/setuptools.lwt
[TRACE] Adding template /home/matt/.local/lib/python2.7/site-packages/pybombs/templates/boostflags.lwt
[DEBUG] Adding recipe location: /home/matt/.pybombs/recipes/gr-recipes
[DEBUG] Scanning directory '/home/matt/.pybombs/recipes/gr-recipes' for recipes...
[... 456 lines cut ...]
[DEBUG] Checking if package git is installed...
[TRACE] Woohoo, this one's already cached (git)
[TRACE] Woohoo, this one's already cached (git)
[TRACE] Woohoo, this one's already cached (git)
[TRACE] Checking if recipe git is installed
[TRACE] Parsing requirements string: git-core || git
[TRACE] Calling ev for recursive satisfier rule evaluation
[DEBUG] Package git has version 2.20.1 installed
[TRACE] Installed, but no update requested. Does not go into tree.
[DEBUG] Phase 1 complete: All binary dependencies installed.
[DEBUG] Phase 2: Recursively installing source packages to prefix:
[DEBUG] Phase 2 complete: All source packages installed.
[DEBUG] Requirements met.
[DEBUG] We have git version 2.20.1
[DEBUG] Using url - https://github.com/gnuradio/gnuradio.git
[DEBUG] Adding remote into git ref
[DEBUG] Adding remote: gnuradio38 -> https://github.com/gnuradio/gnuradio.git
[DEBUG] Remote URL https://github.com/gnuradio/gnuradio.git already registered.
fatal: No such remote or remote group: gnuradio38
[ERROR] Unexpected error while fetching git+https://github.com/gnuradio/gnuradio.git.
[ERROR] Command '['git', 'remote', 'update', 'gnuradio38']' returned non-zero exit status 1
[ERROR] Problem occurred while building package gnuradio38:
Unable to fetch recipe gnuradio38
[ERROR] Error installing package gnuradio38. Aborting.
```

draeath, 2019-08-08:
Getting something similar, but without the git related error.

STEP 17: RUN sudo apt-get update &&    pybombs recipes update &&    pybombs prefix init ~/.gnuradio -a gnuradio -R gnuradio38 --virtualenv &&    echo "source ~/.gnuradio/setup_env.sh" >> ~/.bashrc &&    sudo rm -rf /var/lib/apt/lists/*
Get:1 http://security-cdn.debian.org/debian-security buster/updates InRelease [39.1 kB]
Get:2 http://cdn-fastly.deb.debian.org/debian buster InRelease [118 kB]        
Get:4 http://security-cdn.debian.org/debian-security buster/updates/main amd64 Packages [55.1 kB]
Get:3 http://cdn-fastly.deb.debian.org/debian buster-updates InRelease [46.8 kB]
Get:5 http://cdn-fastly.deb.debian.org/debian buster/main amd64 Packages [7897 kB]
Fetched 8156 kB in 3s (3057 kB/s)   
Reading package lists... Done
[INFO] Prefix Python version is: 3.7.3
[INFO] PyBOMBS Version 2.3.4a0
[INFO] Updating recipe location `gr-etcetera'...
Already up to date.
Current branch master is up to date.
[INFO] Updating recipe location `gr-recipes'...
Already up to date.
Current branch master is up to date.
[INFO] Prefix Python version is: 3.7.3
[INFO] PyBOMBS Version 2.3.4a0
[ERROR] Could not find recipe for `gnuradio38'

Switching on verbose really didn't show anything useful, so I left that output out. I had to replace gnuradio.lwr with the contents of gnuradio38.lwr on disk (after the pybombs recipe update) and use 'gnuradio-default' to make it go.

Teque5, 2019-12-31:
If anyone else is having this issue, update pybombs and then delete your `~/.pybombs/gitcache` directory. That should solve it.