gnuradio/pybombs #543: GR3.8 install: missing libqwt-qt5-dev python3-click-plugins

jdemel, 2019-07-22:
The recipe `gnuradio38` does not care to install  `libqwt-qt5-dev` and `python3-click-plugins`. At least the click dependency is also missing from the manual wiki [Install GR3.8 Python3 dependencies on Ubuntu 18.04](https://wiki.gnuradio.org/index.php/UbuntuInstall#Bionic_Beaver_.2818.04.29) instructions.

jdemel, 2019-07-23:
I just want to add `python3-zmq` is missing as well. At least for the `qa_zeromq_*` tests.

mhostetter, 2019-07-26:
TL;DR: I had a similar issue as OP. I found `libqwt-qt5-dev`, `libqt5opengl5-dev`, `python3-click`, `python3-click-plugins` were missing from the `gnuradio38.lwr`recipe.

Initial error:

```
$ pybombs --version
2.3.4a0
$ pybombs -p gr38 -vv install gnuradio38
...
-- Configuring gr-channels support...
--   Dependency ENABLE_VOLK = ON
--   Dependency Boost_FOUND = 1
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_GR_BLOCKS = ON
--   Dependency ENABLE_GR_FFT = ON
--   Dependency ENABLE_GR_FILTER = ON
--   Dependency ENABLE_GR_ANALOG = ON
--   Enabling gr-channels support.
--   Override with -DENABLE_GR_CHANNELS=ON/OFF
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.7m.so (found suitable version "3.7.3", minimum required is "2") 
-- Python checking for PyQt5 - found
-- Checking for module 'Qt5Qwt6'
--   No package 'Qt5Qwt6' found
-- QWT Version: 6.1.4
-- Could NOT find Qwt (missing: QWT_LIBRARIES) 
-- 
-- Configuring gr-qtgui support...
--   Dependency Boost_FOUND = 1
--   Dependency QT_FOUND = 0
--   Dependency QWT_FOUND = FALSE
--   Dependency ENABLE_VOLK = ON
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_GR_FFT = ON
--   Dependency ENABLE_GR_FILTER = ON
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency PYQT5_FOUND = TRUE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled gr-qtgui but configuration checked failed
Call Stack (most recent call first):
  gr-qtgui/CMakeLists.txt:48 (GR_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/home/matt/gnuradio-3.8/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/home/matt/gnuradio-3.8/src/gnuradio/build/CMakeFiles/CMakeError.log".
[DEBUG] Thread signaled termination or returned
[DEBUG] Return value: 1
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package gnuradio:
Configuration failed
[ERROR] Error installing package gnuradio. Aborting.
```

After running, `$ sudo apt install libqwt-qt5-dev`, I then I received:

```
-- Configuring gr-channels support...
--   Dependency ENABLE_VOLK = ON
--   Dependency Boost_FOUND = 1
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_GR_BLOCKS = ON
--   Dependency ENABLE_GR_FFT = ON
--   Dependency ENABLE_GR_FILTER = ON
--   Dependency ENABLE_GR_ANALOG = ON
--   Enabling gr-channels support.
--   Override with -DENABLE_GR_CHANNELS=ON/OFF
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.7m.so (found suitable version "3.7.3", minimum required is "2") 
-- Python checking for PyQt5 - found
-- Checking for module 'Qt5Qwt6'
--   No package 'Qt5Qwt6' found
-- QWT Version: 6.1.4
-- 
-- Configuring gr-qtgui support...
--   Dependency Boost_FOUND = 1
--   Dependency QT_FOUND = 0
--   Dependency QWT_FOUND = TRUE
--   Dependency ENABLE_VOLK = ON
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_GR_FFT = ON
--   Dependency ENABLE_GR_FILTER = ON
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency PYQT5_FOUND = TRUE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled gr-qtgui but configuration checked failed
Call Stack (most recent call first):
  gr-qtgui/CMakeLists.txt:48 (GR_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/home/matt/gnuradio-3.8/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/home/matt/gnuradio-3.8/src/gnuradio/build/CMakeFiles/CMakeError.log".
[DEBUG] Thread signaled termination or returned
[DEBUG] Return value: 1
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package gnuradio:
Configuration failed
[ERROR] Error installing package gnuradio. Aborting.
```

After running `$ sudo apt install libqt5opengl5-dev`, I received:

```
-- Configuring gr-uhd support...
--   Dependency Boost_FOUND = 1
--   Dependency UHD_FOUND = TRUE
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_GR_FILTER = ON
--   Dependency ENABLE_GR_BLOCKS = ON
--   Dependency ENABLE_GR_ANALOG = ON
--   Enabling gr-uhd support.
--   Override with -DENABLE_GR_UHD=ON/OFF
--   UHD Version: 3.15.0.git-34-g3c8ed1b5
-- Python checking for Mako >= 0.4.2 - found
-- 
-- Configuring gr-utils support...
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_PYTHON = ON
--   Dependency MAKO_FOUND = TRUE
--   Enabling gr-utils support.
--   Override with -DENABLE_GR_UTILS=ON/OFF
-- Python checking for click - not found
-- Python checking for click-plugins - not found
CMake Error at gr-utils/python/modtool/CMakeLists.txt:36 (message):
  Python module click is required for gr-modtool

-- Configuring incomplete, errors occurred!
See also "/home/matt/gnuradio-3.8/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/home/matt/gnuradio-3.8/src/gnuradio/build/CMakeFiles/CMakeError.log".
[DEBUG] Thread signaled termination or returned
[DEBUG] Return value: 1
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package gnuradio:
Configuration failed
[ERROR] Error installing package gnuradio. Aborting.
```

After running `$ sudo apt install python3-click` and `$ sudo apt install python3-click-plugins`, the build succeeded.

jdemel, 2019-09-04:
Since GR 3.8 is released now and PyBOMBS is only a suitable package manager in case you want to install the latest stuff. I'd argue to update PyBOMBS to
1. Enforce Py3k usage (a new install should always use the newest Python version)
2. Support Gr3.8 installation only.

If you want to use older versions, you'd install GR via your distros package manager. If you want to install even older GR versions, you better know exactly why and how you do this. I'd argue community support for those GR versions ran out.

mormj, 2019-09-04:
There are valid reasons for continuing to use pybombs with 3.7 (maybe older?) in supporting legacy software/old OOTs, and being able to bounce back and forth between latest GR and older GR.  I don't think this is a fringe use case, but installing from package manager in this case makes things extremely difficult.

jdemel, 2019-09-05:
@mormj so you use PyBOMBS to switch between legacy versions?
Since PyBOMBS was released way after 3.7, I'd assume it doesn't support 3.6 or older.

What are PyBOMBS use cases? My understanding is:
- install bleeding edge software versions
- have multiple versions for different embedded targets around.

When NOT to use PyBOMBS:
A single GR install with the latest release. e.g. you want the latest GR 3.8 release. You'd just use Debian testing or you'd use the latest Fedora version, I heard they have frequent GR releases too. Basically, you rely on GR package maintainers and don't do a source install.

I'd argue that 3.7 community support is running out with the new release. If you want to use legacy software you either jump through the hoops to make things work for yourself or you pay someone for that service.

Either way, it is important to have a clear understanding of PyBOMBS use cases. Then we can all contribute accordingly.

mormj, 2019-09-05:
> have multiple versions for different embedded targets around.

I would expand this not just for embedded targets but just different versions of GR on the same machine.  For example, I may have a project that was developed a year ago under 3.7, that I need to get back up and running, but don't have the resources to update all the OOTs in the project to 3.8.  So on my dev server I have 3.7, 3.8, and master pybombs installs.  Sometimes there are pybombs installs at certain points gitrev in the recipes just because "it worked with this combination" - though with GNU radio itself, that is not as much an issue as UHD or other dependencies.  Sometimes the use case of pybombs is having different versions of UHD coexisting.

mbr0wn, 2019-09-19:
This was fixed on gr-recipes.

closed 2019-09-19 by mbr0wn