gnuradio/pybombs #545: Failure to configure Python 3.7

Ryandry1st, 2019-08-05:
I have freshly install pybombs and am attempting to install GNU Radio, however when I try to run 
`pybombs -v auto-config` however I get an assertion error. Below is part of the log.

[DEBUG] [1m[35mScanning directory 'c:\program files\python37\lib\site-packages\pybombs\recipes' for recipes...[0m
[DEBUG] [1m[35mFound 11 new recipes.[0m
[DEBUG] [1m[35mLoading templates.[0m
[DEBUG] [1m[35mInitializing command class for command auto-config[0m
[INFO] [1m[32mUsing config file: None[0m
[DEBUG] [1m[35mAttempting to add binary package manager apt[0m
[DEBUG] [1m[35mAttempting to add binary package manager yumdnf[0m
[DEBUG] [1m[35mAttempting to add binary package manager port[0m
[DEBUG] [1m[35mAttempting to add binary package manager brew[0m
[DEBUG] [1m[35mAttempting to add binary package manager zypper[0m
[DEBUG] [1m[35mAttempting to add binary package manager pacman[0m
[DEBUG] [1m[35mAttempting to add binary package manager portage[0m
[DEBUG] [1m[35mAttempting to add binary package manager pymod[0m
[DEBUG] [1m[35mpymod is supported![0m
[DEBUG] [1m[35mAttempting to add binary package manager pip[0m
Traceback (most recent call last):
  File "c:\program files\python37\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\program files\python37\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Program Files\Python37\Scripts\pybombs.exe\__main__.py", line 9, in <module>
  File "c:\program files\python37\lib\site-packages\pybombs\main.py", line 32, in main
    return dispatch() or 0
  File "c:\program files\python37\lib\site-packages\pybombs\commands\base.py", line 194, in dispatch
    return get_cmd_dict(cmd_list)[args.command](cmd=args.command, args=args).run()
  File "c:\program files\python37\lib\site-packages\pybombs\commands\autoconfig.py", line 118, in run
    auto_config_settings = AutoConfigurator(self.log).get_auto_config_settings(self.cfg.keys())
  File "c:\program files\python37\lib\site-packages\pybombs\commands\autoconfig.py", line 38, in get_auto_config_settings
    for k in cfg_keys
  File "c:\program files\python37\lib\site-packages\pybombs\commands\autoconfig.py", line 39, in <dictcomp>
    if hasattr(self, get_method_name(k))}
  File "c:\program files\python37\lib\site-packages\pybombs\commands\autoconfig.py", line 53, in _auto_config_packagers
    self.log
  File "c:\program files\python37\lib\site-packages\pybombs\packagers\base.py", line 129, in filter_available_packagers
    p = get_by_name(pkgr, pkgr_objects)
  File "c:\program files\python37\lib\site-packages\pybombs\packagers\base.py", line 108, in get_by_name
    return obj()
  File "c:\program files\python37\lib\site-packages\pybombs\packagers\pip.py", line 149, in __init__
    self.packager = ExternalPip(self.log)
  File "c:\program files\python37\lib\site-packages\pybombs\packagers\pip.py", line 56, in __init__
    assert self.cmd
AssertionError

Ryandry1st, 2019-08-06:
I have found that this is also the case for a seperate python 2.7.x environment. So something seems to be a problem with the referencing with windows perhaps?

mbr0wn, 2019-09-19:
Actually this is probably a pip.py bug. It isn't finding the pip executable on your system. Try running latest master, it includes a different check for that.

closed 2019-09-19 by mbr0wn